RfidResearchGroup/proxmark3 #832: NTAG215 old MFU format conversion not working?

shellster, 2020-07-05:
**Describe the bug**
If I use the very old (deprecated) iceman fork found here: https://github.com/iceman1001/proxmark3, I can use a tool like https://raw.githubusercontent.com/tomvanveen/samytools/master/mfubin2eml to clone an amiibo to a proxmark3 easy and successfully use it with a Nintendo switch.  However, if I use the latest version (proxmark3-4.9237) from this repo (and same amiibo file that works with the deprecated version) I get the following warnings:

```
proxmark3 /dev/ttyACM0 -c 'hf mf eload u amiibo'
[=] Session log <redacted>/log_20200705.txt
[=] Loading Preferences...
[+] loaded from JSON file <redacted>/preferences.json
[+] execute command from commandline: hf mf eload u amiibo

[=] Using UART port /dev/ttyACM0
[=] Communicating with PM3 over USB-CDC
[usb|script] pm3 --> hf mf eload u amiibo
[+] loaded 1020 bytes from text file amiibo.eml
[+] old mfu dump format, was converted on load to 243 pages
[=] Uploading to emulator memory
.................................................................................................................................................................................................................................................................

[-] ⛔ Warning, Ultralight/Ntag file content, Loaded 257 blocks into emulator memory

proxmark3 /dev/ttyACM0 -c 'hf 14a sim t 7 u 0418521a454b81'
[=] Session log <redacted>/log_20200705.txt
[=] Loading Preferences...
[+] loaded from JSON file <redacted>/preferences.json
[+] execute command from commandline: hf 14a sim t 7 u 0418521a454b81

[=] Using UART port /dev/ttyACM0
[=] Communicating with PM3 over USB-CDC
[usb|script] pm3 --> hf 14a sim t 7 u 0418521a454b81
[+] Emulating ISO/IEC 14443 type A tag with 7 byte UID (04 18 52 1A 45 4B 81 )
[=] Press pm3-button to abort simulation
```

When I try to use the amiibo the Nintendo switch does not recognize it as a valid amiibo.  I recognize that the tag file format has changed since the deprecated version (https://github.com/RfidResearchGroup/proxmark3/blob/8aebb8ef8ebf1f6301d8992c9054c87995cf4a01/doc/mfu_binary_format_notes.md), but I think there is some sort of bug in how the old format is being converted, since it is not resulting in a valid amiibo.  I'm happy to provide logs, or specific files if that is helpful.  If someone can point me to some documentation that can help me update my .eml files to work with the new version, I'd be grateful.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone https://github.com/iceman1001/proxmark3
2. cd into the directory.
3. make
4. cd client
5. ./flasher -b ../bootrom/obj/bootrom.elf ../armsrc/obj/fullimage.elf
6 Unplug/replug proxmark3.
7 (Referencing amiibo.eml that I can provide, in this case for Animal Crossing Lily)
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 - hw status
 - data tune

**Additional context**
Add any other context about the problem here.

shellster, 2020-07-05:
Accidentally submitted this before I was done typing up the bug issue.  So I closed the issue until I was done editing.  Reopening now.

iceman1001, 2020-07-12:
I pushed some minor changes,  one is the addition of a amiboo pack when used with a dumpfile which didn't have it.
Pull latest and see if it works better.

Side note:
Your Proxmark3 device has 256kb internal flash memory and the full image you flashed is really pushing the limits.
Try to remove some command support from the armsrc makefile,   to generate a smaller fullimage.  
```
  --= Nonvolatile Program Memory Size: 256K bytes, Used: 261435 bytes (100%) Free: 709 bytes ( 0%)
```

shellster, 2020-07-14:
I'll check it next week when I get home and have another computer.  Thanks for the response.  I tried building the latest from master, but I'm running into the same errors as described in https://github.com/RfidResearchGroup/proxmark3/issues/815 I'm on Devuan Linux.  I probably have an Ubuntu VM somewhere, at home.

mokus0, 2020-07-21:
I've seen this same issue on my system (rdv4, tested on multiple OSes) - the same .eml file works on an old build, but not on new one, including master as of today (commit 551bc6c71bf41d6d8b7f3a876ded720a81cbca73).  Happy to provide any info needed.

mokus0, 2020-07-21:
Looking at the header formats and poking around in the simulator memory between the two versions, I notice a couple things:

* In the header as loaded by the current master, the "version" field appears to be truncated to the lower 4 bytes (from the eml file, it should be `0004040201001103`, but the first 8 bytes of `hf mf eget 0` are `0000000001001103`.  Second, I'm not sure if it's actually wrong or not, but the `pages` field is loaded as 0xF2, whereas in the old version it prints a message after `eload` saying that it loaded 255 blocks.

I've tried manually fixing the version using `hf mf eset`, with and without also tweaking the `pages` field, but that wasn't enough to actually make it work.

Syfaro, 2020-07-21:
@cheeplusplus wrote about they got amiibo support working again here: https://blog.neocodenetworks.com/posts/2020-04-08-amiibo-emulation/

https://github.com/cheeplusplus/proxmark3/commit/52184833e5c0d6066db6e5009b48dd3c00629e01

mokus0, 2020-07-21:
Thanks, looks like the password part is probably the other piece I was missing.  The password value doesn't end up in the simulator memory with the new legacy load operation.  If my quick skim of the code is correct, it appears to be pulling it from address 1024.  I need to get to bed so I didn't dig any deeper to see whether that's a valid place to load from, but I can't access that via `hf mf eget` anyway.

shellster, 2020-07-22:
Thanks guys for all your testing as well as the previous link.

mokus0, 2020-07-22:
I believe I've tracked down the cause of the 0's in the first 4 bytes loaded.  They are actually the last 4 bytes - in the generated eml files from samyk's mfubin2eml tool, there are 1020 bytes including the old mfu header.  when read in and converted to the new format, the resulting data is now 1028 bytes due to the larger header.  The extra 4 bytes beyond 1024 overflow and end up written to bytes 0-3.  Manually deleting the last block from the eml file fixes the 'version' field.

mokus0, 2020-07-22:
Ok, and I've now figured out a bit more about the password part.
The Switch sends an auth request with some kind of a password to the card, and the firmware is pulling the expected password out of the second-to-last block of the dump.  If it matches the value sent by the switch, It then sends a response from the last block in the dump.  So with some adjustments to the eml files only, I'm able to get it working on the current master.

To summarize, I had to:

* shorten the eml file
* append the expected challenge code, which is different for each amiibo (I got it for a couple amiibos by just adding some debug printing to the firmware, but I'm guessing there's a way to calculate this) - it'd also be nice to have a firmware option to ignore it, like @cheeplusplus does in their branch.
* append the response code that the amiibo is supposed to send back (`80808080`)

It would also be good to change the header generated by mfubin2eml to the new format.
All of these changes should be feasible in mfubin2eml, assuming the challenge code can be calculated.

I'm beat for the night, and I don't know when I might have time to poke at this more, but that should at least be a good amount of info if someone else wants to run with it.

cheeplusplus, 2020-07-22:
@mokus0 If you're talking about calculating the PWD to check if you should send `pACK`, the code for that is already in proxmark:

https://github.com/RfidResearchGroup/proxmark3/blob/f830843442664eb70ad0361a4d7eee8e30044021/common/generator.c#L99-L109

If I recall correctly, part of the reason I hardcoded it was because I was using unencrypted amiibo dumps with `pm3_amii_bin2eml.pl`, which (if not in the dump) adds a header using a fixed UID and `00000000` for the PWD, and therefore a non-special-cased `MIFARE_ULEV1_AUTH` command would fail. You would have to either use a properly encrypted dump (DIY dump or re-encrypt with `amiitool`), or alternatively modify the bin2eml script to calculate the real PWD and put it in the header.

mokus0, 2020-07-22:
Yeah, I suspect that's exactly what I'd need - thanks! My plan if I have the time in the next day or so is to update the bin2eml script to generate the new header format, including only as much data as was in the original binary dump, and add the extra auth tokens.  Your hardcoded fix of ignoring the PWD works well in the interim.

kprocbert, 2020-07-29:
can confirm that it's working with the new mfu conversion script.

One question tho, when i try to write one of my chinese ntags with:

old version:

```
hf mf cload myfile
[+] loaded 1020 bytes from text file myfile.eml
[!!] 🚨 File content error. 
```

new version:

```
hf mf cload myfile
[+] loaded 596 bytes from text file myfile.eml
[!!] 🚨 File content error. 
```

any ideas? :/

iceman1001, 2020-07-29:
@mokus0   I think we have the same script from samy,  its just renamed,   so it would be nice if you did the same love to it as you did the upstream one :)

mokus0, 2020-07-30:
@iceman1001 sure, I can do that - it might be a couple days before I have the spare cycles though!

fbettag, 2020-08-04:
> can confirm that it's working with the new mfu conversion script.
>
> One question tho, when i try to write one of my chinese ntags with:
>
> old version:
>
> ```
> hf mf cload myfile
> [+] loaded 1020 bytes [...]

Same issue here

iceman1001, 2020-08-05:
until you compare the files there is nothing to be said.

iceman1001, 2022-02-05:
closing because of inactivity.  

if someone comes along and test different old style mfu binaries and can verify it still doesn't work we can reopen this one and take it from there

closed 2022-02-05 by iceman1001