RfidResearchGroup/proxmark3 #850: Great HF FPGA stuff to take from official repo

doegox, 2020-07-11:


iceman1001, 2020-07-13:
After having spend two weeks on this,  aka merge hell,  its is starting to look good.
@pwpiwi did some nice improvements to ISO15,  iCLASS and 14B.   I got most of it working, you find it in the branch FPGA.
Will do some more extensive testing this week regarding the iCLASS parts.
then 15,  and last 14B.

This change brought some cross over problems relating to the stack / bigbuff changes.  Where we found out that the trace log had issues with tosend array.   The solution was to move tosend to bigbuff as we do with emul memory.
Hopefully,  this solves also other issues in LF that is related to the legacy mixed up bigbuffer usage.  Its a dump pile.

When it comes to current iCLASS state:
Read/write/dump/chk/info  works ok
Simulation works ok.
Sniffing is ok but has some issues.
```

[usb] pm3 --> hf iclass sniff
[usb] pm3 --> [#] Sniff started. Press PM3 Button to stop.
[#] Sniff stopped.
[#] Sniff statistics:
[#]   ExpectTagAnswer: 1, TagIsActive: 0, ReaderIsActive: 0
[#]   DecodeTag State: 0
[#]   DecodeTag byteCnt: 0
[#]   DecodeTag posCount: 157555
[#]   DecodeReader State: 1
[#]   DecodeReader byteCnt: 1
[#]   DecodeReader posCount: 2
[#]   Trace length: 1418
[#]   Max behindBy: 255

[usb] pm3 --> hf iclass list
[=] downloading tracelog data from device
[+] Recorded activity (trace len = 1418 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO15693 / iCLASS - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |     360448 | Rdr |0a                                                                       |     | ACTALL
      53280 |     413728 | Rdr |0c                                                                       |     | IDENTIFY
     799456 |    1159904 | Rdr |0a                                                                       |     | ACTALL
     840224 |    1200672 | Rdr |0c                                                                       |     | IDENTIFY
     776448 |    1172736 | Rdr |00  00  00                                                               |     | HALT
    1296960 |    1919552 | Rdr |88  02                                                                   |     | READCHECK[Kd](2)
    1659808 |    1752992 | Rdr |88  08  00  00  00  00  00  00  00  00  00                               |     | READCHECK[Kd](8)
    1859648 |    1911872 | Rdr |05  63  00  00  00  00                                                   |     | CHECK
    1868352 |    2002496 | Rdr |00                                                                       |     | HALT
    2276928 |    2320960 | Rdr |05  00  00  00  00                                                       |     | CHECK
    2801216 |    3161664 | Rdr |87                                                                       |     | UPDATE(0)
    2982016 |    3342464 | Rdr |0c                                                                       |     | IDENTIFY
    2913408 |    3178624 | Rdr |4c  4c                                                                   |     | READ(76)
    2417600 |    2944960 | Rdr |4c  4c  4c  4c  4c  4c  4c  4c  4c  4c  4c  4c                           | !crc| READ(76)
    3988896 |    4090272 | Rdr |13! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00!                          | !crc| ?
    3941440 |    4337728 | Rdr |80! 80! 80!                                                              |     | HALT
    4653088 |    4697120 | Rdr |05  f0  00  00  00                                                       |     | CHECK
    5095104 |    5455552 | Rdr |05                                                                       |     | CHECK
    5313408 |    5936000 | Rdr |88  02                                                                   |     | READCHECK[Kd](2)
    5450080 |    5608800 | Rdr |00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00   |     |
            |            |     |00                                                                       |     | HALT
    5835776 |    5945344 | Rdr |58  00  00  00  00  00  00  00  00  00  00  00  00                       |     | READCHECK[Kc](0)
    5968352 |    6069728 | Rdr |87  00  00  00  00  00  00  00  00  1f  92  9b                           | !crc| UPDATE(0)
    6193120 |    7077856 | Rdr |0c  06  45                                                               |     | READ(6)
    6282432 |    6465728 | Rdr |06  06  50  00  80  b7  d8  d8  dd  d8  74  00  00  00  00  00  00  00   |     |
            |            |     |dd  a8  1d  b6                                                           |     | READ4(6)
    7067488 |    7427936 | Rdr |0a                                                                       |     | ACTALL
    7069280 |    7170656 | Rdr |00  00  00  00  00  00  00  40  00  00  00  00                           | !crc| HALT
    7232096 |    7546464 | Rdr |0c  05  80  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00   |     |
            |            |     |00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00   |     |
            |            |     |00  00                                                                   |     | READ(5)
[... cut ...]
    8806560 |    9104544 | Rdr |dd! 22! c2! a1! 74! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00!  |     |
            |            |     |00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 6d! 86!  |     | ?
    8952736 |    9267104 | Rdr |b0! 00! 00! 00! 00! 00! 00! 00! 00! c0! 00! 00! 00! 00! 00! 00! 00! 00!  |     |
            |            |     |00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00! 00!  |     |
            |            |     |00! 00!                                                                  |     | HALT

```

iceman1001, 2020-08-13:
_MERGEHELL_   **Current status:** 50% done.

We decided to merge the FPGA branch into master, since master is unstable anyway.
iCLASS support is rock solid but since merging of 15693 / 14443B isn't done we lost that support currently.

iceman1001, 2020-08-16:
iCLASS, ISO15693, ISO14443B  seems to be ok now.   Some individual commands might need some more love, but overall it looks good.

Examples of commands that may need some more love.
- hf iclass restore
- hf 15 restore
- hf 14b dump

Examples of bad commands since earlier,
- hf 15 sim
_the ISO15693 simulation doesn't do much. Not many protocol commands supported.  There are some forks of official repo which are trying to improve that._  @ceres-c   
- hf 14b sim
_same thing here,  would need a eload/esave  to simulate some different cards like SRIX / SRI512_

closed 2020-08-16 by iceman1001