RfidResearchGroup/proxmark3 #866: 14b raw truncating responses

doegox, 2020-07-11:
Example reading a Navigo with libnfc versus pm3: for all long responses, pm3 is losing the last bytes.
```
TX:(-4) 00 a4 04 00 08 31 54 49 43 2e 49 43 41 [2f 00]
RX:(-4)    6a 82 [fe 00]
[usb] pm3 --> hf 14b raw -s -p -c 0200a4040008315449432e494341
[LEN 5] 02 6A 82 [4B 4C] OK          

TX:(-4) 00 a4 00 00 02 3f 00 [06 00]
RX:(-4)     85 17 00 01 00 00 00 12 12 00 00 01 03 01 01 00 15 15 15 00 00 00 00 00 00 90 00 [54 00]
[usb] pm3 --> hf 14b raw -p -c 0300a40000023f00
[LEN 21] 03 85 17 00 01 00 00 00 12 12 00 00 01 03 01 01 00 15 15 [15 00] FAIL          

TX:(-4) 00 a4 00 00 02 00 02 [43 00]
RX:(-4)     85 17 02 04 02 1d 01 1f 00 00 00 01 01 01 01 00 00 00 00 00 00 00 00 00 00 90 00 [75 00]
[usb] pm3 --> hf 14b raw -p -c 0200a40000020002
[LEN 20] 02 85 17 02 04 02 1D 01 1F 00 00 00 01 01 01 01 00 00 [00 00] FAIL          

TX:(-4) 00 b2 01 04 1d [17 00]
RX:(-4)     00 00 00 00 00 00 00 04 00 71 00 42 00 00 00 00 c1 7a 16 45 02 50 14 00 01 29 98 43 00 90 00 [a2 00]
[usb] pm3 --> hf 14b raw -p -c 0300b201041d
[LEN 22] 03 00 00 00 00 00 00 00 04 00 71 00 42 00 00 00 00 C1 7A 16 [45 02] FAIL          
```

iceman1001, 2020-07-13:
The branch FPGA,  should have impact on 14B comms.

doegox, 2020-08-16:
I don't have the same Navigo with me but testing latest master with new 14b code, now I get complete tag comm, both with sniffing phone and sniffing scl3711

closed 2020-08-16 by doegox