RfidResearchGroup/proxmark3 #870: Testbed for features

slurdge, 2020-07-11:
It would be great IMO if we could create a set of cards that would allow to test for most common features (up to the "difficult" attacks).

We would list the hardware requirements, such as regular mifare1k card, regular lf tag, etc... and then flash appropriate data on them with a script.

From that point, another script could try all the commands on them and see if the results are valid.

It would help creating new features and testing optimizations.

iceman1001, 2021-04-02:
with the  /tools/pm3_tests.sh  ,  is this issue still relevant?   Or can we close?

iceman1001, 2021-04-02:
I don't see anyone remotely interested in writing all needed tests of the 600+ Proxmark3 client commands currently existing and all its use cases.

To answer myself,   no.  closing.

closed 2021-04-02 by iceman1001