RfidResearchGroup/proxmark3 #871: Add kind of machine-readable auto-completion mechanism

doegox, 2020-07-11:
To help graphical clients (and touchscreens ;) ), implement some machine-readable auto-completion, like:
```
token hf
=> {"one_of":["s14a","s14b","s15","sepa","sfelica","slegic",...]}
token hf 14a 
=> {"one_of":["slist","sinfo","sreader","scuids","ssim","ssniff",...]}
token hf 14a sim
=> {"many_of":["st":{"one_of":[1,2,3,4,5,6,7,8,9]},"su":{"one_of":["xUID4", "xUID7"]},"sx", "se", "sv"]}
```

So the graphical frontends can probe the client to discover the available tokens for a given command.

iceman1001, 2021-12-12:
And we get readline-based autocomplete for inside the Proxmark3 client.
It generates a large file with all commands ,  like we do the json file for the pm3 discord bot.   So the touch screen devices should be able to query it too without much changes.

iceman1001, 2022-02-05:
Closing because the needed command file for a graphical UI would need to query exists now in the repo.

closed 2022-02-05 by iceman1001