RfidResearchGroup/proxmark3 #876: Issue between MFC sim and MFC autopwn: wrong data in some of the ACLs

doegox, 2020-07-11:
Dumped a card

```
hf mf autopwn
# => hf-mf-2D1671AA-data.eml
hf mf eload hf-mf-2D1671AA-data
hf mf sim s 08
```
then with second reader
```
hf mf autopwn
# => hf-mf-2D1671AA-data-1.eml
```

```diff
--- hf-mf-2D1671AA-data.eml	2019-09-12 13:31:27.091223857 +0200
+++ hf-mf-2D1671AA-data-1.eml	2019-09-12 13:42:32.491149204 +0200
@@ -1,15 +1,15 @@
 2D1671AAE0880400488514574D503810
 0F010203030505050505050505050505
 3600000741B39B89D1B6B25500000051
-FFFFFFFFFFFFFF078000FFFFFFFFFFFF
+FFFFFFFFFFFF00000000FFFFFFFFFFFF
 65100E00899BB3412FCDB114AF6A503A
 00000000000000000000000000000000
 0000000000000000000000000000A62C
-FFFFFFFFFFFFFF078000FFFFFFFFFFFF
+FFFFFFFFFFFF00000000FFFFFFFFFFFF
 00000000000000000000000000000000
 00000000000000000000000000000000
 00000000000000000000000000000000
-FFFFFFFFFFFFFF078000FFFFFFFFFFFF
+FFFFFFFFFFFF00000000FFFFFFFFFFFF
 00000000000000000000000000000000
 00000000000000000000000000000000
 00000000000000000000000000000000
```
Other ACLs are ok:
```
A0A1A2A3A4A57877880051595C1BC141
```

iceman1001, 2020-09-02:
@doegox   I think I fixed this some days ago.

doegox, 2020-09-02:
nope
[image]

doegox, 2020-09-02:
So the emul mem on left seems ok before `hf mf sim` but emul mem (and file) dumped by second reader gets emptied ACLs

iceman1001, 2020-09-03:
aha,
@mwalker33  has debugged some,   inside the 
https://github.com/RfidResearchGroup/proxmark3/blob/master/armsrc/mifarecmd.c#L1923
the loop can be exit when bad read or bad access,  which leads to partial dump.

The client side doesn't understand if its been a partial dump, as seen here,  nor reports to user of it.
https://github.com/RfidResearchGroup/proxmark3/blob/master/client/src/cmdhfmf.c#L859

iceman1001, 2020-09-05:
After chatting with @mwalker33  and pushing some more fixes to understand the problem,  the flow should be fixed by now.
1. clear emulator memory before populating with keys and dump,  
2. if read Key A worked,  skip read with keyB.
3. if read key A and B fails,   report back to user that the dump is partial complete.

doegox, 2020-09-05:
Yes it's much better now.
It systematically says PARTIAL, most of the time it gets all ACLs but occasionally it gets ACLs of only first few sectors then zeroes.

doegox, 2020-09-05:
No matter which card with defaults, it always reports
```
[=] fast dump reported back failure w KEY A,  swapping to KEY B
[=] fast dump reported back failure w KEY B
[=] Dump file is PARTIAL complete
```

iceman1001, 2020-09-05:
your autopwn fails reading all blocks with keyA,   goes for keyB..  and fails there aswell.
So, is it bad reads?  or is it your access rights?
What does your trace list look like?

doegox, 2020-09-05:
Try autopwn on a blank MFC and you'll see

iceman1001, 2020-09-05:
.... try pull and at least the "always reports" is gone..
now its down the acl

closed 2020-09-05 by doegox