RfidResearchGroup/proxmark3 #879: printf -> PrintAndLogEx

doegox, 2020-07-11:
There are still quite a number of printf in the client code.
Maybe a few are legit, but most should be converted to make sure they get logged as well.
```
     28 cliparser/argtable3.c:
     27 emv/emv_pki.c:
     18 fido/cbortools.c:
     13 cmdhfmf.c:
     11 loclass/ikeys.c:
     10 hardnested/hardnested_bf_core.c:
      9 cliparser/cliparser.c:
      8 util.c:
      8 emv/test/cda_test.c:
      7 emv/test/dda_test.c:
      7 emv/crypto_polarssl.c:
      6 uart/uart_posix.c:
      6 crypto/libpcrypto.c:
      6 cmdhflegic.c:
      5 loclass/hash1_brute.c:
      5 emv/test/sda_test.c:
      5 cmdlf.c:
      4 uart/uart_win32.c:
      4 scripting.c:
      4 comms.c:
      4 cmdsmartcard.c:
      4 cmdhffido.c:
      3 mifare/mifarehost.c:
      3 loclass/elite_crack.c:
      3 emv/test/crypto_test.c:
      3 emv/emv_pk.c:
      3 cmdhfmfp.c:
      2 proxguiqt.cpp:
      2 emv/emv_roca.c:
      2 cmdlft55xx.c:
      2 cmdhficlass.c:
      2 cmddata.c:
      2 cmdanalyse.c:
      1 wiegand_formatutils.c:
      1 ui.c:
      1 tinycbor/cborpretty.c:
      1 jansson/strconv.c:
      1 hardnested/hardnested_bruteforce.c:
      1 guidummy.cpp:
      1 cmdhw.c:
      1 cmdhfmfu.c:
      1 cmdhffelica.c:
      1 cmdhf15.c:
      1 cmdhf14b.c:
```
(magic grep: `rg printf|egrep -v "(snprintf|sprintf|liblua|fprintf|:[[:space:]]*/|hardnested_tables.c|//.*printf|\.git)"|grep -o "[^:]*\.c:"|sort|uniq -c|sort -n -r`)

iceman1001, 2020-08-29:
```
     32 tools/mf_nonce_brute/mf_nonce_brute.c:
     18 client/src/fido/cbortools.c:
     17 tools/cryptorf/cm.c:
     13 tools/cryptorf/sm.c:
     10 client/deps/hardnested/hardnested_bf_core.c:
      9 client/src/cmdhfmf.c:
      8 common/cryptorf/cryptolib.c:
      8 client/src/util.c:
      8 client/src/cmdhfmfp.c:
      7 client/src/cmdhflegic.c:
      6 client/src/loclass/ikeys.c:
      6 client/src/crypto/libpcrypto.c:
      5 client/src/loclass/hash1_brute.c:
      4 client/src/ui.c:
      4 client/src/scripting.c:
      4 client/src/cmdlf.c:
      4 client/src/cmdhfmfdes.c:
      4 client/src/cmdhficlass.c:
      3 tools/cryptorf/util.c:
      3 client/src/mifare/mifarehost.c:
      3 client/src/loclass/elite_crack.c:
      3 client/src/comms.c:
      2 tools/fpga_compress/fpga_compress.c:
      2 client/src/uart/uart_posix.c:
      2 client/src/proxmark3.c:
      2 client/src/emv/emv_roca.c:
      2 client/src/cmdsmartcard.c:
      2 client/src/cmdlft55xx.c:
      2 client/src/cmdlfgallagher.c:
      1 common/crc.c:
      1 client/src/wiegand_formatutils.c:
      1 client/src/cmdlfcotag.c:
      1 client/src/cmdhw.c:
      1 client/src/cmdhfmfu.c:
      1 client/src/cmdhffelica.c:
      1 client/src/cmdhfcryptorf.c:
      1 client/src/cmdhf15.c:
      1 client/src/cmdhf14b.c:
      1 client/deps/tinycbor/cborpretty.c:
      1 client/deps/jansson/strconv.c:
```

closed 2020-08-29 by iceman1001

iceman1001, 2020-11-13:
So,  reducing the filter  (add cliparse, argtable)
```
rg printf|egrep -v "(snprintf|sprintf|liblua|fprintf|Dbprintf|armsrc|:[[:space:]]*/|hardnested_tables.c|zlib|mbedtls|cryptorf|mf_nonce_brute|hitag2crack|mfkey|nonce2key|//.*printf|\.git|Makefile|sh:|deprecated)"|grep -o "[^:]*\.c:"|sort|uniq -c|sort -n -r
```

```
     28 client/deps/cliparser/argtable3.c:
     19 client/src/pm3_pywrap.c:
     18 client/src/pm3_luawrap.c:
      6 client/deps/cliparser/cliparser.c:
      4 client/src/ui.c:                                          --> printing before "log" 
      2 tools/fpga_compress/fpga_compress.c:    --> the tools are ignored.
      2 client/src/proxmark3.c:                             --> is not included in the "lib" idea
      2 client/src/cmdlfem4x05.c:
      1 common/crc.c:                                          --> commented out
      1 client/deps/tinycbor/cborpretty.c:              --> commented out
      1 client/deps/jansson/strconv.c:                    --> commented out
```

This leads me the clean list,   where we have some new wrap files doing stuff.
``` 
     28 client/deps/cliparser/argtable3.c:
     19 client/src/pm3_pywrap.c:
     18 client/src/pm3_luawrap.c:
      6 client/deps/cliparser/cliparser.c:
      2 client/src/cmdlfem4x05.c:
```

doegox, 2020-11-13:
wrappers are generated by SWIG, I guess it is safe to ignore them for the moment.
but at some point we'll have to look into all I/O-related and SWIG.