RfidResearchGroup/proxmark3 #890: Rework Lua script names

doegox, 2020-07-11:
It would be nice to see immediately the category of a script by its name.
like getting all mifare script starting with ``mf_`` (if mfu + mfc)  or ``mfc_`` etc

msoose, 2020-08-13:
What type of naming convention should be used for this?

`%freqband%_%cardtype%_%function%`?:
- `14araw.lua` => `hf_14a_rxtxraw`? (rxtxraw = RX/receive TX/transmit raw)
- `dumptoemul-mfu.lua` => `hf_mfu_dumptoemulator`

What about scripts that don't directly interface with cards, such as `htmldump.lua`? Prefix 'sys' or 'local' or something similar?
- `sys_dumptohtml` or `local_dumptohtml`

doegox, 2020-08-13:
I'd suggest mimicking the pm3 commands tree so yes to `hf_14a_*`, etc.
So for local, maybe `data_*` ?

msoose, 2020-08-13:
Are fully descriptive commands preferred over abbreviated? Most people using pm3 should know what rx/tx refer to. Just wanna get some standards down for the future.

`hf_14a_rxtxraw`
`hf_14a_readwriteraw`
`hf_14a_rwraw` (I don't like this one)

iceman1001, 2020-08-13:
hf_14a_raw      keep it simple and similar,

iceman1001, 2020-08-13:
Well done @msoose !

I will just mention some things and bring it up,  so we can open up for discussion about it. It not in any way any negative feedback.  I just wanna find some optimal naming convention for us.

I think initially the new name suggestions still leaves a sour taste.   
I chatted with @doegox  ,  and the idea of archiving some old-out-date scripts,  since better implementations exists in the current client.  Things that comes to mind.  

- Some renaming looses info,   I think luxeo is easier to renamber since it refresh to a specific system,  rather than laundrymat.
- Test scripts for T55xx gives the wrong impression with new suggested names.  
- skeleton scripts might have that instead of cmd.

What do you all think?

msoose, 2020-08-13:
For sure, by no means exhaustive or even correct, more of a means to get things started and discussion going.

I was unsure about `t55xx`, some of the `hf` include `14a` while others have named cards, or the modifiers like you mentioned specifically naming luxeo vs generic laundromat.

iceman1001, 2020-08-13:
Besides my spelling sucks,    yes,  
hf_mf*  
vs
mf*

maybe we don't need the hf_  and lf_  ?    well,  for  lf_hid  I thought that was good name since it direct you to the client commands.   hf_mfm  and hf_mfc  doesn't,  which might confuse more.

msoose, 2020-08-13:
Yeah, I was unsure of how to do Mifare vs Mifare Mini vs Mifare Classic vs Mifare UL, etc...aside from `hf_mf_function-mini` and `-classic` and `-ultralight` but that's getting too convoluted.

iceman1001, 2020-08-13:
yes, that would be too convoluted, I agree.
How about
mfc_   for classic
mf_     for mini ( mfm_ is confusing)
mfp_   for plus
mfdes_ for desfire
14a_
14b_
15_

so,   14a_magic_  etc ?

doegox, 2020-08-13:
I added support for subdirs in "script list", this might help sorting things differently than prefixes.
At the moment if one wants to call a script in a subdir he still needs to type e.g. `script run mifare/formatMifare` but we could also adapt searchFinalFile to recurse

closed 2021-01-27 by iceman1001