RfidResearchGroup/proxmark3 #896: LogTrace usages

doegox, 2020-07-11:
I fixed some usages where duration was wrongly used as timestamp_end.
Still, some usages seem weird...

client/src/cmdtrace.c:
* ISO14443A - All times are in carrier periods (1/13.56MHz)
* Thinfilm - All times are in carrier periods (1/13.56MHz)
* iClass: ?
* LEGIC - Reader Mode: Timings are in ticks (1us == 1.5ticks), Tag mode: (1/212 kHz == 4.7us)
* ISO14443B: ?
* ISO15693: ?
* ISO7816-4 / Smartcard : ?
* Hitag1 / Hitag2 / HitagS - Timings in ETU (8us)
* FeliCa : ?

ARM:
* ISO14443A: 
  * hf 14a reader: timings seem ok, but with offsets of 64 or 128 units compared to e.g. libnfc/spec. 
  * Internally using `(GetCountSspClk() + 8) & 0xfffffff8` then `*16 or <<4` and with corrective offsets
* iClass
  *  SniffIClass: GetCountSspClk()
  * doIClassSimulation: GetCountSspClk() << 4
  * ReaderTransmitIClass_ext/ReaderReceiveIClass : no duration, no start ? g_rsamples==0 unused?
* LEGIC
  * reader mode: using GET_TICKS
  * tag mode = ??
* ISO14443B
  * simulate: 0, 0
  * GetTagSamplesFor14443bDemod / CodeAndTransmit14443bAsReader / SniffIso14443b : GetCountSspClk();
* ISO15693
  * GetCountSspClk() << 4
* ISO7816-4
  * 0, 0
* Hitag1 / Hitag2 / HitagS
  * SniffHiitag2: 0, 0
  * SimulateHitag2: lf_count_edge_periods(128)/0, 0
  * ReaderHitag: idem
* FeliCa
  * TransmitFor18092_AsReader: `GetCountSspClk()) & 0xfffffff8)` then `<< 4`, duration = felica_lasttime_prox2air_start + felica_lasttime_prox2air_start ??
  * WaitForFelicaReply / felicasniff: ((GetCountSspClk() & 0xfffffff8) << 4) - DELAY_AIR2ARM_AS_READER - timeout,  ((GetCountSspClk() & 0xfffffff8) << 4) - DELAY_AIR2ARM_AS_READER, ??

iceman1001, 2020-07-13:
In the branch FPGA,  the timings has been improved for  iCLASS; ISO15, and 14b.

iceman1001, 2026-09-10:
Hitag ,  sc/sam ,  felica  should print times nicely and accurate now

closed 2026-09-10 by iceman1001