RfidResearchGroup/proxmark3 #897: SWIG revolución

doegox, 2020-07-11:
Issue to track ideas, progress and tasks to
- define a unified user API, starting with sth like open/close/sendcmd
- get a C libpm3 (used also by proxmark3 client via that user API to not duplicate bins)
- expose the unified API internally to embedded Python and Lua (one day, that will replace current Lua API)
- expose the unified API externally to Python scripts

First PoC tests on https://github.com/RfidResearchGroup/proxmark3/tree/libpm3_experiments
Current compilation paths to keep an eye on:
- make
- cmake
- cmake /lib              (test of C libpm3)
- cmake /libpy            (test of py wrapper around C libpm3)
- cmake /client_with_swig (test of "libpm3" API accessible to embedded py & lua)

Next steps:
- [x] make client cmake independent of build location (${PM3_ROOT}...)
- [x] start with decent minimal API
- [x] embedded py & lua
- [x] extend API to context to avoid global vars in shared lib ?
- [ ] Make Lua and Python looking for the real libpm3rrg_rdv4.so name and make sure the lib is in their respective path when installed / when in repo. If possible.
- [ ] convert Lua scripts to use the unified API ? provide a `sed` script to automate script conversion?
- [ ] extend API to be able to divide client => light client + static libpm3
- [ ] move to dynamic libpm3, ! make install)
- [ ] bring same cmake logic to Makefile
- [ ] once we got dynamic libpm3 separated from client, get python module "published" (make install & for local dev) for external python
- [ ] find ways to return output instead of direct print, #dbg msgs will probably need a callback
- [ ] get libpm3 and scripts running on PS/WSL/OSX

snowsign, 2023-07-29:
Any updates on this? It's been a few years and there still isn't a consistent Python/Lua API. Python scripts run through the client currently need to rely on the serial port, which isn't possible when running through the client due to locking.

iceman1001, 2023-07-29:
No active work is going on.  Feel free to pick up and complete it!