RfidResearchGroup/proxmark3 #898: Replace BigBuf with a small & efficient allocator

slurdge, 2020-07-11:
It shouldn't be hard to replace BigBuf with an embeded allocator:
https://github.com/dimonomid/umm_malloc

The only missing bit is requesting the biggest chunk of free memory, which is easy to implement (for traces etc.)

We can then gradually remove static memory taken & put it back to this allocator, taking care of not creating memory leaks.

End result: more memory & easier tracking (basically removing each little function taking a bunch of .data for itself...)

doegox, 2020-08-31:
I started adding umm code here : https://github.com/RfidResearchGroup/proxmark3/tree/umm_wip
Status: it should be usable but it still needs a space and size, currently it's pointing to BigBuff space.

doegox, 2020-08-31:
oops, @slurdge had already his own test branch, see also https://github.com/slurdge/proxmark3/tree/umm