RfidResearchGroup/proxmark3 #900: Finish switch to NG frames

doegox, 2020-07-11:
One way to ease the remaining old/mix frames is to enlarge max capacity of NG frame so we can directly accommodate an OLD frame (with args + data) into a NG.

iceman1001, 2020-08-17:
Not that many  *SendCommandOLD*   or *reply_old* left in the repo.
Of course a bunch of SendCommandMIX...

We can actually increase a NG frame to any multiple of 64bytes.  There is no direct reason for not increasing it to lets say 1024 bytes,  Total frame size.

1024 will not make too large impact on memory on device side for sending / receiving.
We would need to look into the BTADDON, usart implementation if that is effected.

iceman1001, 2020-10-06:
Just for cross ref.
https://github.com/RfidResearchGroup/proxmark3/issues/992

closed 2024-01-14 by iceman1001