RfidResearchGroup/proxmark3 #904: iclass functions getting '[!] ⚠️  failed authenticating with debit key' error on vanilla legacy iclass tag

thesle3p, 2020-07-14:
Unable to run chk,dump or rdbl in legacy iclass tags

**To Reproduce**
Steps to reproduce the behavior:
````
[usb] pm3 --> hf iclass dump k  <redacted>
[-] ⛔ no tag found
[usb] pm3 --> hf iclass dump k <redacted>
[!] ⚠️  failed authenticating with debit keyime in iclass checkkeys: 4 seconds

````
**Expected behavior**
tag would dump

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Debian linux
 - inside proxmark3 client run the following commands and paste the output here.
```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-593-g8934fd52 2020-07-10 19:50:51
  compiled with GCC 9.3.0 OS:Linux ARCH:x86_64

 [ PROXMARK3 RDV4 ]
  external flash:                  present
  smartcard reader:                present

 [ PROXMARK3 RDV4 Extras ]
  FPC USART for BT add-on support: absent

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-593-g8934fd52 2020-07-10 19:51:13
       os: RRG/Iceman/master/v4.9237-593-g8934fd52 2020-07-10 19:51:26
  compiled with GCC 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-02-22 at 12:51:14
  HF image built for 2s30vq100 on 2020-01-12 at 15:31:16

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 258952 bytes (49%) Free: 265336 bytes (51%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

```
```
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size.............42056
[#]   Available memory........41288
[#] Tracing
[#]   tracing ................1
[#]   traceLen ...............322
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-01-12 at 15:31:16
[#] Flash memory
[#]   Baudrate................24 MHz
[#]   Init....................OK
[#]   Memory size.............2 mbits / 256 kb
[#]   Unique ID...............0xD5690C23DFB79F2A
[#] Smart card module (ISO 7816)
[#]   version.................v3.10
[#] LF Sampling config
[#]   [q] divisor.............95 ( 125.00 kHz )
[#]   [b] bits per sample.....8
[#]   [d] decimation..........1
[#]   [a] averaging...........Yes
[#]   [t] trigger threshold...0
[#]   [s] samples to skip.....0 
[#] LF Sampling Stack
[#]   Max stack usage.........4008 / 8480 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#]               leading zero |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#]    1 of 4 coding reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#] 
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed............500ms
[#]   Bytes transferred.......253952
[#]   Transfer Speed PM3 -> Client = 507904 bytes/s
[#] Various
[#]   Max stack usage.........4096 / 8480 bytes
[#]   DBGLEVEL................1
[#]   ToSendMax...............24
[#]   ToSendBit...............8
[#]   ToSend BUFFERSIZE.......2308
[#]   Slow clock..............29906 Hz
[#] Installed StandAlone Mode
[#]   HF - Reading Visa cards & Emulating a Visa MSD Transaction(ISO14443) - (Salvador Mendoza)
[#] Flash memory dictionary loaded
[usb] pm3 --> 
```
```
[usb] pm3 --> data tune
[=] Measuring antenna characteristics, please wait...
 🕕  10
[=] ---------- LF Antenna ----------
[+] LF antenna: 71.65 V - 125.00 kHz
[+] LF antenna: 38.27 V - 134.83 kHz
[+] LF optimal: 74.53 V - 126.32 kHz
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 46.23 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 88 is 134.83 kHz, 95 is 125.00 kHz.

 ```

```

iceman1001, 2020-07-14:
Try the release ?   The current state of the repo is unstable.

thesle3p, 2020-07-14:
It's present in the 4.9237 release as well

iceman1001, 2020-07-14:
hm..  try the FPGA branch?

thesle3p, 2020-07-17:
fpga branch works

iceman1001, 2020-07-17:
I suspected that.

conside master to be unstable until FPGA branch gets merged.  Still the 14B functions to fix up.

Sp0ge, 2020-07-19:
```
$make clean && make all

===================================================================
Platform name:     Proxmark3 rdv4
PLATFORM:          PM3RDV4
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH -DRDV4 LF HITAG EM4x50 ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   HF_MSDSAL
===================================================================
[*] MAKE client/clean
===================================================================
Client platform:   Linux
GUI support:       QT not found, disabled
native BT support: Bluez found, enabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 v3.8 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library
compiler version:  gcc (Debian 9.3.0-13) 9.3.0
===================================================================
[*] MAKE bootrom/clean
[*] MAKE fpga_compress/clean
[*] MAKE armsrc/clean
[*] MAKE recovery/clean
[*] MAKE mfkey/clean
[*] MAKE nonce2key/clean
[*] MAKE mf_nonce_brute/clean

===================================================================
Platform name:     Proxmark3 rdv4
PLATFORM:          PM3RDV4
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH -DRDV4 LF HITAG EM4x50 ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   HF_MSDSAL
===================================================================
[*] MAKE client/all
===================================================================
Client platform:   Linux
GUI support:       QT not found, disabled
native BT support: Bluez found, enabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 v3.8 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library
compiler version:  gcc (Debian 9.3.0-13) 9.3.0
===================================================================
[-] CC src/aidsearch.c
[-] CC src/cmdanalyse.c
[-] CC src/cmdcrc.c
[-] CC src/cmddata.c
[-] CC src/cmdflashmem.c
[-] CC src/cmdflashmemspiffs.c
[-] CC src/cmdhf.c
[-] CC src/cmdhf14a.c
[-] CC src/cmdhf14b.c
[-] CC src/cmdhf15.c
[-] CC src/cmdhfcryptorf.c
[-] CC src/cmdhfepa.c
[-] CC src/cmdhffelica.c
[-] CC src/cmdhffido.c
[-] CC src/cmdhficlass.c
[-] CC src/cmdhflegic.c
src/cmdhflegic.c:16:10: fatal error: readline/readline.h: No such file or directory
   16 | #include <readline/readline.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:706: obj/cmdhflegic.o] Error 1
make: *** [Makefile:137: client/all] Error 2

```
How i can fix that?
Pls help

thesle3p, 2020-07-19:
Did you run sudo apt-get install --no-install-recommends git ca-certificates build-essential pkg-config \
libreadline-dev gcc-arm-none-eabi libnewlib-dev qtbase5-dev libbz2-dev libbluetooth-dev like the wiki tells you to?

Sp0ge, 2020-07-19:
yes , i did it

Sp0ge, 2020-07-19:
i install and update everything and it don't work. i work in parrot os (debain)

thesle3p, 2020-07-19:
Then you should file your own ticket to fix that issue

iceman1001, 2020-07-19:
@Sp0ge  Please don't hijack issues.

iceman1001, 2020-08-13:
this issue should be solved with merging of FPGA branch into master.

closed 2020-08-13 by iceman1001