RfidResearchGroup/proxmark3 #910: Error Compiling crapto1 on ARM64

msoose, 2020-07-30:
***Compilation problems***
[-] CC src/cmdsmartcard.c
[-] CC ../common/crapto1/crapto1.c../common/crapto1/crapto1.c:358:59: error: result of      '10 ^ 24' is 18; did you mean '1e24'?      [-Werror,-Wxor-used-as-pow]  ...BEBIT(in, 10), fb) << (10 ^ 24); ~~~^~~~ 1e24
../common/crapto1/crapto1.c:358:59: note: replace expression with '0xA ^ 24' to silence this warning
../common/crapto1/crapto1.c:367:57: error: result of '2 ^ 24' is 26; did you mean '1 << 24' (16777216)?      [-Werror,-Wxor-used-as-pow]  ...lfsr_rollback_bit(s, BEBIT(in, 2), fb) << (2 ^ 24);  ~~^~~~   1 << 24
../common/crapto1/crapto1.c:367:57: note: replace expression with '0x2 ^ 24' to silence this warning2 errors generated.
make[1]: *** [Makefile:706: obj/crapto1/crapto1.o] Error 1make: *** [Makefile:137: client/all] Error 2

**To Reproduce**
Steps to reproduce the behavior:
Build on ARM64 / Android device in Termux

**Expected behavior**
Clean compile of client

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Android
 - master / 9659764
 - PM3 OTHER

doegox, 2020-07-30:
Sigh, what a weird warning.
@msoose does it help if we simply swap all (x ^ 24) into (24 ^ x) ? as apparently it spots only the 10^ and 2^ and not the other ones.

msoose, 2020-07-30:
I guess? I'm not too familiar with why this is being thrown for the operator.

doegox, 2020-07-31:
Well, could you try? I don't have the same compilation setup as you to test it by myself...

anonymix007, 2020-07-31:
I have tried and it definitely looks like that this is false-positive trigger of clang. Also it looks like I will need to compile gcc (which is not present in repos, only clang):

LD proxmark3
[*] MAKE bootrom/all
/data/data/com.termux/files/usr/bin/sh: 1: arm-none-eabi-gcc: not found
compiler version:
[=] GEN version.c                               [-] CC version.c
make[1]: arm-none-eabi-gcc: No such file or directory
make[1]: *** [../common_arm/Makefile.common:124: obj/version.o] Error 127
make: *** [Makefile:131: bootrom/all] Error 2

msoose, 2020-07-31:
@doegox Swapping just the entries throwing warnings works and it links correctly.

Let me change them and I'll submit a PR?

iceman1001, 2020-07-31:
Funny,  the solution is given in the output.  It wants hex values instead of decimal.   No need to swap order.
```
./common/crapto1/crapto1.c:367:57: note: replace expression with '0x2 ^ 24' to silence this warning2 errors generated.
```

doegox, 2020-07-31:
@iceman1001 it's not like we've just these 2 lines, we've 32 lines and converting all of them to 0xNN ^24 looks really ugly.
maybe 0xN ^0x18  or what I proposed 24 ^ N
https://github.com/RfidResearchGroup/proxmark3/blob/master/common/crapto1/crapto1.c#L335-L369

doegox, 2020-07-31:
> @doegox Swapping just the entries throwing warnings works and it links correctly.
> 
> Let me change them and I'll submit a PR?

Yes thanks!

PR #912 (2020-07-31): Build with clang on aarch64 / Termux on Android
Fix #910
closed 2020-07-31 by doegox