RfidResearchGroup/proxmark3 #911: ERROR: cannot communicate with the Proxmark

GregorBiswanger, 2020-07-30:
Hello Community, 

I tried my new Proxmark3 on Windows 10 to get it working with ProxSpace. 

In the end I finally managed to install a new bootrom and fullimage with `pm3-flash-all`. Then I would like to connect with the command `pm3`. It finds the port automatically and connects. Then I get an error message that communication is not working.

Then the device manager only shows my device under USB controller and that it would be an unknown device (error when requesting a device description). Windows only recognizes the device when I disconnect it from the USB and plug it in again with the button pressed (the blue and green flash light is on). I then installed the bootrom and fullimage again. It didn't help.

My terminal output:

```
$ runme64.bat
pm3 ~$ ./pm3-flash-all
[=] Session log C:\Users\Gregor\Documents\ProxSpace\pm3/.proxmark3/logs/log_20200730.txt
[+] About to use the following files:
[+]    C:\Users\Gregor\Documents\ProxSpace\pm3\client\../bootrom/obj/bootrom.elf
[+]    C:\Users\Gregor\Documents\ProxSpace\pm3\client\../armsrc/obj/fullimage.elf
[+] Waiting for Proxmark3 to appear on COM7
[|] 59 found
[=] Available memory on this board: 512K bytes

[=] Permitted flash range: 0x00100000-0x00180000
[+] Loading ELF file C:\Users\Gregor\Documents\ProxSpace\pm3\client\../bootrom/obj/bootrom.elf
[+] Loading usable ELF segments:
[+]    0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
[+]    1: V 0x00200000 P 0x00100200 (0x00000d50->0x00000d50) [R X] @0x298

[+] Loading ELF file C:\Users\Gregor\Documents\ProxSpace\pm3\client\../armsrc/obj/fullimage.elf
[+] Loading usable ELF segments:
[+]    0: V 0x00102000 P 0x00102000 (0x0003d788->0x0003d788) [R X] @0x94
[+]    1: V 0x00200000 P 0x0013f788 (0x00001964->0x00001964) [RW ] @0x3d81c
[=] Note: Extending previous segment from 0x3d788 to 0x3f0ec bytes

[+] Flashing...
[+] Writing segments for file: C:\Users\Gregor\Documents\ProxSpace\pm3\client\../bootrom/obj/bootrom.elf
[+]  0x00100000..0x001001ff [0x200 / 1 blocks]
 OK
[+]  0x00100200..0x00100f4f [0xd50 / 7 blocks]
 OK

[+] Writing segments for file: C:\Users\Gregor\Documents\ProxSpace\pm3\client\../armsrc/obj/fullimage.elf
[+]  0x00102000..0x001410eb [0x3f0ec / 505 blocks]
mm OK

[+] All done

Have a nice day!
pm3 ~$ ./pm3
[=] Session log C:\Users\Gregor\Documents\ProxSpace\pm3/.proxmark3/logs/log_20200730.txt
[=] Using UART port COM7
[!!] ERROR: cannot communicate with the Proxmark

pm3 ~$
```

GregorBiswanger, 2020-08-04:
I execute `make PLATFORM=PM3OTHER clean && make PLATFORM=PM3OTHER all` and `./pm3-flash-all` - Now it´s work without problems.

closed 2020-08-04 by GregorBiswanger

iceman1001, 2021-11-29:
Good thing it works now,

Don't know where you read PM3OTHER in the documents on this repo.  It was renamed and it is PM3GENERIC.

The makefiles still allow the old name.

PieBru, 2023-02-17:
Same problem here, solved with:
```
$ make PLATFORM=PM3GENERIC clean && make PLATFORM=PM3GENERIC all
$ ./pm3-flash-all
$ ./pm3
```

Thank you.

iceman1001, 2023-02-17:
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/4_Advanced-compilation-parameters.md#firmware

jb4inkss, 2023-02-18:
> I execute `make PLATFORM=PM3OTHER clean && make PLATFORM=PM3OTHER all` and `./pm3-flash-all` - Now it´s work without problems.

Hello i think i have the same problem. Where can i execute this commande ? `make PLATFORM=PM3OTHER clean && make PLATFORM=PM3OTHER all` 

 thnkxx
[log_20230217.txt](https://github.com/RfidResearchGroup/proxmark3/files/10773020/log_20230217.txt)

iceman1001, 2023-02-18:
If you read the above post you would not use PM3OTHER.   In order to understand,  you would need to post the commands you actually used.

PieBru, 2023-02-19:
Yes, thank you IceMan for pointing me to the doc.
I posted here because I found this issue before knowing your excellent documentation already explained my pm3 easy noob problem.
Thank you again.
Piero

polprog, 2023-09-11:
I'd like to confirm this works on a "Proxmark3 Easy 512MB" version.

     make PLATFORM=PM3OTHER clean && make PLATFORM=PM3OTHER all

Then `pm3-flash-fullimage`

The symptoms before that, on regular firmware was that the PM3 disconnected USB device after booting and the ACM device was gone.

Anaphylaxis, 2024-10-22:
Even after following the PM3GENERIC clean make command, I still receive "PM3_RDV4 firmware on a generic device" from `hw tune`