RfidResearchGroup/proxmark3 #917: LTO-6 past 255

modemmike, 2020-08-10:
Any pointers on how to read past line 255? Basically, anyway to get all the blocks from 16kB tapes?

iceman1001, 2020-08-10:
not without any kind of datasheet....

modemmike, 2020-08-10:
I was afraid that was going to be the answer, ty for clarification. So, we all need an LTO-6/7 datasheet. Not that I could but IF I had a contact somewhere, what specifically would I ask for?

iceman1001, 2020-08-10:
IF you had a contact somewhere,  I would have asked if I could get a copy of a full datasheet for LTO-6  / specs.  And if there was an issue with that,  I would say image copies of the  rfid protocol pages would do just fine

iceman1001, 2020-11-04:
Some progress in getting your hands on a datasheet?

modemmike, 2020-11-04:
Possibly but not as of yet. It's on the radar but the client has been busy.

iceman1001, 2020-11-04:
Feel free to join the discord server and discuss it even more...

iceman1001, 2021-05-06:
closing because of lack of needed information

closed 2021-05-06 by iceman1001