RfidResearchGroup/proxmark3 #921: READ ME: Current MASTER status

doegox, 2020-08-13:
Hi all
Currently, the master is "bleeding edge", maybe a bit more than usual ;)
You've basically two options depending which techs you want to address:
* Use the latest release https://github.com/RfidResearchGroup/proxmark3/releases/tag/v4.9237
  * HF: ISO14443A, ISO14443B, ISO15693
* Use the master branch
  * HF: ISO14443A, iCLASS

Of course, once @iceman1001 has done with the MERGEHELL, all HF techs will be available in a single firmware.

iceman1001, 2020-08-14:
**_Even more bleeding edge than expected._**

As of right now,  don't compile with  BTADDON option on current master.

msoose, 2021-01-08:
Curiously, are there any current "outstanding" issues or features pending before a release can be branched?

tcprst, 2021-01-08:
We're finishing up with the cliparser for all commands before a release.

iceman1001, 2021-01-08:
and having a check on what made BTADDON compiled fw unstable on device.
Then its ready for a new release

iceman1001, 2021-03-22:
We did much bug fixes from coverity scan & cppchecker since last summer.   @mwalker33  et all has tested BTADDON last week and don't find any issues with it.   Maybe not a conclusive testing but atleast I am getting comfortable enough for recommend we close this issue about BTADDON.

doegox, 2021-03-22:
> I am getting comfortable enough for recommend we close this issue about BTADDON.

enough to allow for a new release once cliparser is done?

iceman1001, 2021-03-22:
yes

Fl0-0, 2021-03-28:
In my tests BTADDON and/or FPC_USART_DEV still have the issues (unusable antennas and UART:: write time-out) on Kali LInux:

```
[=] Using UART port /dev/ttyACM0
[=] Communicating with PM3 over USB-CDC

  ██████╗ ███╗   ███╗█████╗ 
  ██╔══██╗████╗ ████║╚═══██╗
  ██████╔╝██╔████╔██║ ████╔╝
  ██╔═══╝ ██║╚██╔╝██║ ╚══██╗
  ██║     ██║ ╚═╝ ██║█████╔╝       Iceman ☕
  ╚═╝     ╚═╝     ╚═╝╚════╝    ❄️ bleeding edge

  https://github.com/rfidresearchgroup/proxmark3/

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-3405-ga34bf3264 2021-03-28 17:05:39
  compiled with GCC 10.2.1 20210110 OS:Linux ARCH:x86_64

 [ PROXMARK3 ]
  device.................... RDV4
  firmware.................. RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... present

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-3377-gdb3719792 2021-03-25 09:13:23
       os: RRG/Iceman/master/v4.9237-3405-ga34bf3264 2021-03-28 17:05:38
  compiled with GCC 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 [ FPGA ] 
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Internal SRAM size: 64K bytes
  --= Architecture identifier: AT91SAM7Sxx Series
  --= Embedded flash memory 512K bytes ( 62% used )

[usb] pm3 --> hw tune
[=] ---------- Reminder ------------------------
[=] `hw tune` doesn't actively tune your antennas,
[=] it's only informative.
[=] Measuring antenna characteristics, please wait...
 🕘  12
[=] ---------- LF Antenna ----------
[+] LF optimal: 1832072.03 V -   0.00 kHz
[+] Approx. Q factor (*): 319594.8 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[!] ⚠️  HF antenna is UNUSABLE

(*) Q factor must be measured without tag on the antenna

[-] ⛔ Not showing LF tuning graph since all values is zero.

[usb] pm3 --> hf search
 🕚  Searching for ThinFilm tag...[!] ⚠️  timeout while waiting for reply.
 🕒  Searching for ISO14443-A tag...[!!] 🚨 UART:: write time-out
 🕔  Searching for ISO15693 tag...[!!] 🚨 UART:: write time-out
[!!] 🚨 UART:: write time-out
 🕖  Searching for iCLASS / PicoPass tag...^C 
```

iceman1001, 2021-04-15:
Cliparser is now implemented everywhere.
Next stop,  14B timings...
After that we can prep for a new release.  

I am sure there are some minor bugs related to the cliparse changes hiding plain sight.

iceman1001, 2021-04-23:
I pushed a fix for 14b timings,   so time to weed out show stopper bugs

iceman1001, 2021-07-15:
Closing,   since we now have a new release and not much feedback about broken functionality.

closed 2021-07-15 by iceman1001