RfidResearchGroup/proxmark3 #924: HF Antenna not working properly

theguy147, 2020-08-13:
**Describe the bug**

All HF commands do not work after updating e.g. `hf search` with a regular MF Classic 1K (S50). 

In order for any LF or HF command to work at all I had to follow #819 and decrease the stack size in `common_arm/ldscript.common` to `8K` first (but I did try other values from 6K to 8488 as well). After applying this change the `hw tune`/`data tune` commands work fine and also the `hf tune` command works and reacts to HF tags close to the proxmark3 as it should.

**To Reproduce**

1. `git checkout master && git pull`
2. `sed -i 's/stacksize : 8488/stacksize : 8K/' common_arm/ldscript.common` 
3. `make clean && make all`
4.  attach proxmark3 via USB cable
5. `./pm3-flash-all && ./pm3`
6.  Place Mifare Classic 1K tag on proxmark3
7. pm3 --> `hf search`

result => `[-]No known/supported 13.56 MHz tags found`

**Expected behavior**

I was expecting to see the UID and other Information about the MF Classic tag instead.

**Desktop:**

 - OS: Arch Linux (Linux 5.7.9-1 x86_64)

<details>
<summary>[usb] pm3 --> hw version</summary>

```
 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-783-g6bd0138f-dirty-unclean 2020-08-13 19:34:49
  compiled with GCC 10.1.0 OS:Linux ARCH:x86_64

 [ PROXMARK3 RDV4 ]
  external flash:                  present
  smartcard reader:                present

 [ PROXMARK3 RDV4 Extras ]
  FPC USART for BT add-on support: absent

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-783-g6bd0138f-dirty-unclean 2020-08-13 19:35:15
       os: RRG/Iceman/master/v4.9237-783-g6bd0138f-dirty-unclean 2020-08-13 19:35:32
  compiled with GCC 10.1.0

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 292344 bytes (56%) Free: 231944 bytes (44%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
```
</details>
<details>
<summary>[usb] pm3 --> hw status</summary>

```
[#] Memory
[#]   BigBuf_size.............41636
[#]   Available memory........41636
[#] Tracing
[#]   tracing ................0
[#]   traceLen ...............13
[#]   dma8 memory.............-2112828
[#]   dma16 memory............-2112828
[#]   toSend memory...........-2112828
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
[#] Flash memory
[#]   Baudrate................24 MHz
[#]   Init....................OK
[#]   Memory size.............2 mbits / 256 kb
[#]   Unique ID...............0xD567A882A7369225
[#] Smart card module (ISO 7816)
[#]   version.................v3.11
[#] LF Sampling config
[#]   [q] divisor.............95 ( 125.00 kHz )
[#]   [b] bits per sample.....8
[#]   [d] decimation..........1
[#]   [a] averaging...........Yes
[#]   [t] trigger threshold...0
[#]   [s] samples to skip.....0 
[#] LF Sampling Stack
[#]   Max stack usage.........3984 / 8184 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A | 
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 | 
[#] 
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed............500ms
[#]   Bytes transferred.......284672
[#]   Transfer Speed PM3 -> Client = 569344 bytes/s
[#] Various
[#]   Max stack usage.........4080 / 8184 bytes
[#]   DBGLEVEL................1 ( ERROR )
[#]   ToSendMax...............9
[#]   ToSend BUFFERSIZE.......2308
[#]   Slow clock..............31651 Hz
[#] Installed StandAlone Mode
[#]   HF - Reading Visa cards & Emulating a Visa MSD Transaction(ISO14443) - (Salvador Mendoza)
[#] Flash memory dictionary loaded
[#]   Mifare..................820 keys
[#]   T55x7...................109 keys
[#]   iClass..................7 keys
```
</details>
<details>
<summary>[usb] pm3 --> data tune</summary>

```
[=] Measuring antenna characteristics, please wait...
[-]  9
[=] ---------- LF Antenna ----------
[+] LF antenna: 27,10 V - 125,00 kHz
[+] LF antenna: 32,41 V - 134,83 kHz
[+] LF optimal: 32,30 V - 133,33 kHz
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 24,81 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 88 is 134,83 kHz, 95 is 125,00 kHz.
```
</details>
<details>
<summary>git stuff</summary>

`git status`

```
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   common_arm/ldscript.common

no changes added to commit (use "git add" and/or "git commit -a")
```

`git rev-parse --short master`

```
6bd0138f
```

`git diff`

```
diff --git a/common_arm/ldscript.common b/common_arm/ldscript.common
index 610e5c0b..0472b8fd 100644
--- a/common_arm/ldscript.common
+++ b/common_arm/ldscript.common
@@ -9,7 +9,7 @@ ms of the GNU GPL, version 2 or,
 -----------------------------------------------------------------------------
 */
 
-stacksize = DEFINED(stacksize) ? stacksize : 8488;
+stacksize = DEFINED(stacksize) ? stacksize : 8K;
 commonareasize = 0x20;
 
 /* AT91SAM7S256 has 256k Flash and 64k RAM */
```
</details>

**Additional context**

All LF commands seem to work perfectly and are not affected by this issue (at least I didnt encounter any issues here)

I do have two HF antennas (the one for the original RDV 4.0 and the improved one for LF from RDV 4.01) but the issue occurs with both of them. Obviously I also checked if any connections for the antenna are loose.

Also I did try to vary the distance between reader and tags and I tried several MF Classic tags (that all work with different readers/ChameleonTiny etc)

Additionally I tried recompiling and reflashing several times with different commits going back until (and including) release v4.9237. I wasn't using the proxmark3 now for some months but this morning (before updating) everything was working perfectly fine with the firmware and client from around August 2019. The problem is that I cant seem to get the old commits to compile, probably because of changes in version 10 of GCC.

iceman1001, 2020-08-13:
Yeah,  the master branch is unstable.  
Try using the release version and see if you have better luck
https://github.com/RfidResearchGroup/proxmark3/releases

theguy147, 2020-08-13:
wow, that was a fast reply. unfortunately as I wrote in the last paragraph above even release v4.9237 does not work for me...

iceman1001, 2020-08-13:
Now,  what is your output from  **hw status** and **hw tune** ?

Just for reference: https://github.com/RfidResearchGroup/proxmark3/issues/921

theguy147, 2020-08-13:
`hw status` and `hw tune` are included in my initial report (They are collapsed to save space so you have to click on the little arrows next to them). Also I have read #921 before posting the issue.

theguy147, 2020-08-13:
Here are all the outputs for release v4.2937 now (just to be complete):

<details>
<summary>hw tune</summary>

```
[=] Measuring antenna characteristics, please wait...
[/] 10
[+] LF antenna: 27,64 V - 125,00 kHz
[+] LF antenna: 32,45 V - 134,83 kHz
[+] LF optimal: 32,45 V - 133,33 kHz
[+] LF antenna is OK 

[+] HF antenna: 27,50 V - 13.56 MHz
[+] HF antenna is OK 

[+] Displaying LF tuning graph. Divisor 88 is 134,83 kHz, 95 is 125,00 kHz.
```
</details>

<details>
<summary>hw status</summary>

```
#db# Memory
#db#   BIGBUF_SIZE.............40000
#db#   Available memory........40000
#db# Tracing
#db#   tracing ................1
#db#   traceLen ...............586
#db# Currently loaded FPGA image
#db#   mode.................... HF image built for 2s30vq100 on 2020-01-12 at 15:31:16
#db# Flash memory
#db#   Baudrate................24 MHz
#db#   Init....................OK
#db#   Memory size.............2 mbits / 256 kb
#db#   Unique ID...............0xD567A882A7369225
#db# Smart card module (ISO 7816)
#db#   version.................v3.11
#db# LF Sampling config
#db#   [q] divisor.............95 ( 125.00 kHz)
#db#   [b] bits per sample.....8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db#   [s] samples to skip.....0 
#db# LF T55XX config
#db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
#db#            mode            |start|write|write|write| read|write|write
#db#                            | gap | gap |  0  |  1  | gap |  2  |  3
#db# ---------------------------+-----+-----+-----+-----+-----+-----+------
#db# fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
#db#     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A | 
#db#               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A | 
#db#    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 | 
#db# 
#db# Transfer Speed
#db#   Sending packets to client...
#db#   Time elapsed............500ms
#db#   Bytes transferred.......282112
#db#   Transfer Speed PM3 -> Client = 564224 bytes/s
#db# Various
#db#   DBGLEVEL................1
#db#   ToSendMax...............24
#db#   ToSendBit...............8
#db#   ToSend BUFFERSIZE.......2308
#db#   Slow clock..............31651 Hz
#db# Installed StandAlone Mode
#db#   HF - Reading Visa cards & Emulating a Visa MSD Transaction(ISO14443) - (Salvador Mendoza)
#db# Flash memory dictionary loaded
#db#   Mifare..................820 keys
#db#   T55x7...................109 keys
#db#   iClass..................7 keys
```
</details>

<details>
<summary>hw version</summary>

```
 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/HEAD/v4.9237 2020-08-13 20:29:22
  compiled with GCC 10.1.0 OS:Linux ARCH:x86_64

 [ PROXMARK3 RDV4 ]
  external flash:                  present
  smartcard reader:                present

 [ PROXMARK3 RDV4 Extras ]
  FPC USART for BT add-on support: absent

 [ ARM ]
  bootrom: RRG/Iceman/HEAD/v4.9237 2020-08-13 20:30:18
       os: RRG/Iceman/HEAD/v4.9237 2020-08-13 20:30:30
  compiled with GCC 10.1.0

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-02-22 at 12:51:14
  HF image built for 2s30vq100 on 2020-01-12 at 15:31:16

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 288560 bytes (55%) Free: 235728 bytes (45%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
```
</details>

iceman1001, 2020-08-13:
ooo,  collapsable areas.  I never seen that on GH before.   There is a first for everything!  

Interesting, My 14a worked nice like a week ago.  Today my 14a doesn't work.    I will have a look at it.
_Been too focused on iCLASS_

theguy147, 2020-08-13:
Ok, cool. BTW iClass works like a charm ;)

iceman1001, 2020-08-13:
Hm,  it works well.  I was testing on a bad magic gen1 tag before.

<details>
<summary>hf 14a info</summary>
```
[usb] pm3 --> hf 14a info

[+]  UID: 04 9D BA 42 A2 3E 80
[+] ATQA: 00 44
[+]  SAK: 08 [2]
[+] MANUFACTURER:    NXP Semiconductors Germany
[+] Possible types:
[+]    MIFARE Classic 1K / Classic 1K CL2
[+]    MIFARE Plus 2K / Plus EV1 2K
[+]    MIFARE Plus CL2 2K / Plus CL2 EV1 2K
[=] proprietary non iso14443-4 card found, RATS not supported
[+] Prng detection: hard
[?] Hint: try `hf mfp info`
[usb] pm3 --> hf 14a info

[+]  UID: 04 9D BA 42 A2 3E 80
[+] ATQA: 00 44
[+]  SAK: 08 [2]
[+] MANUFACTURER:    NXP Semiconductors Germany
[+] Possible types:
[+]    MIFARE Classic 1K / Classic 1K CL2
[+]    MIFARE Plus 2K / Plus EV1 2K
[+]    MIFARE Plus CL2 2K / Plus CL2 EV1 2K
[=] proprietary non iso14443-4 card found, RATS not supported
[+] Prng detection: hard
[?] Hint: try `hf mfp info`
[usb] pm3 --> hw tune
[=] Measuring antenna characteristics, please wait...
 🕛   9
[=] ---------- LF Antenna ----------
[+] LF antenna: 41.75 V - 125.00 kHz
[+] LF antenna: 28.30 V - 134.83 kHz
[+] LF optimal: 41.75 V - 125.00 kHz
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 38.80 V - 13.56 MHz
[+] HF antenna is OK
[+] Displaying LF tuning graph. Divisor 88 is 134.83 kHz, 95 is 125.00 kHz.
[usb] pm3 -->
```
</details>

theguy147, 2020-08-13:
Using v4.9.2937 I just tested again the following tags (which all work with other devices):

- MF 1K S50 Gen2
- MF 1K S50 Gen1a
- MF Ultralight-C Gen2
- MF 4K Gen2

All of them give me:

`[usb] pm3 --> hf 14a info -v`
```
[!] iso14443a card select failed
```

theguy147, 2020-08-13:
I just switched the **Q** on the RDV4.01 Antenna from 7 to 14 and now it works most of the times. I thought that setting doesn't influence the HF antenna?! Also why doesnt it work with the original antenna then?

Does that mean my antennas are faulty or could that still be a software issue?

iceman1001, 2020-08-13:
Sounds more like your antenna or device is acting up.   I hope you connected in the right positions if you removed it ;)

theguy147, 2020-08-13:
yes, I did. And also retried it several times. wow, i'm very confused! I guess I gotta get myself a new antenna then...

Thanks anyway for the good and very very fast support ;)

closed 2020-08-13 by theguy147

iceman1001, 2020-08-13:
lets try something.

revert your "lowering" of the stack.
make clean; make -j; flash all
try `hf 14a info` only on your tags one by one.

theguy147, 2020-08-13:
wow, now I'm even more confused!! somehow it works now (at least the 3 times i tried). Even when turning the **Q** switch back. I guess it was stuck in the middle before or something and that caused an issue with the antennas...

iceman1001, 2020-08-13:
Well,  good thing it works for you.   
However  **hf search**  calls  14b, felica and all those protocols hasn't been fixed yet.   So that command isn't a good one to use.

theguy147, 2020-08-15:
not sure if this is the right place or if I should post this in a new issue or the related issue I referenced above but here it is:

I **do** need to "lower the stack" if I compile the firmware with bluetooth support (even when using the proxmark3 with the usb cable afterwards). Otherwise `hw tune`, `data tune` and all `lf` and `hf` commands don't work. Without the bluetooth support it works fine without this change...

If you want me to open a new issue for this I can do that of course.