RfidResearchGroup/proxmark3 #925: Antennas problem

nico0481, 2020-08-17:
Hello,

As I try to read or search any tag (LF or HF),  I get time-ou error.

Example:
[usb] pm3 --> hf search
🕔  Searching for ThinFilm tag...[!] ⚠️  timeout while waiting for reply.
 🕘  Searching for ISO14443-A tag...UART:: write time-out
 🕚  Searching for ISO15693 tag...UART:: write time-out
UART:: write time-out
 🕐  Searching for iCLASS / PicoPass tag...UART:: write time-out
UART:: write time-out
UART:: write time-out
UART:: write time-out
UART:: write time-out
UART:: write time-out
UART:: write time-out
UART:: write time-out
UART:: write time-out
UART:: write time-out
UART:: write time-out
 

Here is HW Version:
 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-817-gb941e6aa-dirty-unclean 2020-08-17 12:48:29
  compiled with GCC 8.3.0 OS:Linux ARCH:x86_64

 [ PROXMARK3 RDV4 ]
  external flash:                  present
  smartcard reader:                present

 [ PROXMARK3 RDV4 Extras ]
  FPC USART for BT add-on support: present

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-817-gb941e6aa-dirty-unclean 2020-08-17 12:49:10
       os: RRG/Iceman/master/v4.9237-817-gb941e6aa-dirty-unclean 2020-08-17 12:49:36
  compiled with GCC 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 304440 bytes (58%) Free: 219848 bytes (42%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

Here is HW Tune:
[usb] pm3 --> hw tune
[=] Measuring antenna characteristics, please wait...
 🕒  12
[=] ---------- LF Antenna ----------
[+] LF optimal: 542862.64 V -   0.00 kHz
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[!] ⚠️  HF antenna is UNUSABLE

[-] ⛔ Not showing LF tuning graph since all values is zero.

Same issue with previous matching versions of client/fw
Antennas have not been removed from previous working use, screws are correctly fastened.
I have not idea what has triggered this issue.

Thank you

Regards

iceman1001, 2020-08-17:
I will refer to the pinned issue here  https://github.com/RfidResearchGroup/proxmark3/issues/921

Since you compiled with BTADDON...

```
FPC USART for BT add-on support: present
```

nico0481, 2020-08-17:
Solved!!

Thank you

closed 2020-08-17 by nico0481