RfidResearchGroup/proxmark3 #928: [Idea] Proxmark AUR

Windslab, 2020-08-25:
How about if you will create Proxmark repo with dependence in the Arch Linux AUR? If code has changed AUR will tell you - you should upgrade your package[repo].

It shall looks like a one button click to work with repo, also you can add shutouts for run/flash/compile. You even can do an UI.

Pamac - It's a new way to work on Linux with Proxmark.

No need thanks, better fix a forum login. Scine SSL certificate it's not working ;)

iceman1001, 2020-08-25:
We don't do packages nor binary distributions.   We leave that for others to implement.
Like the package for Arch,   https://www.archlinux.org/packages/community/x86_64/proxmark3/   

Feel free to implement and support such support in Arch.

closed 2020-08-25 by iceman1001

Windslab, 2020-08-25:
> Feel free to implement and support such support in Arch.

I'll explain to you why I don't and not recommend to any bodey
Try to setup Arch Linux to Vmware through Cryptsetup, you will be awesome to see this picture, after the Disk Defragmentation and Compact procedure in vmware settings:

[image]

What about forum? I am too lazy to create a new account :(

Windslab, 2020-08-25:
@iceman1001 what enviroment you are use in your videos? How can I get it?

As I see you have lounch from WT some WSL?

[image]

iceman1001, 2020-08-25:
I use Window Terminal with WSL-1   

Take note, WSL2 doesn't work with usb-serial.