RfidResearchGroup/proxmark3 #929: [idea]PM3 shell scripts

Windslab, 2020-08-28:
Hi to all. Can you please corect the sysntacs of the .sh scripts for the do some of commands? It's does't work. Problems with cd/r and git not see the .git

Arch Linux the scripts for i3blocks:

```
#!/bin/sh

cd ~/proxmark3
git fetch
git pull
make clean && make all
pm3-flash-all
```
and

```
#!/bin/sh

cd ~/proxmark3
cd client
proxmark3 /dev/ttyACM0
```

doegox, 2020-08-28:
* Your scripts assume the repo is already cloned in ~/proxmark3, right ?
* `git fetch` + `git pull` is redundant. Actually `git pull` == `git fetch` + `git merge`
* if you chain make clean && make all, better to chain also the flashing, so you don't attempt to flash if the compilation failed.
* when you call a command from the local directory, you need to write it with "./" : `./pm3-flash-all`

So my guess is that you need something like this
```
#!/bin/sh

[ -d ~/proxmark3 ] || git clone https://github.com/RfidResearchGroup/proxmark3.git
cd ~/proxmark3
git pull
make clean && make -j && ./pm3-flash-all
```

and 

```
cd ~/proxmark3/client
./proxmark3 /dev/ttyACM0
```

Or just
```
cd ~/proxmark3
./pm3
```

Windslab, 2020-08-28:
> * Your scripts assume the repo is already cloned in ~/proxmark3, right ?

Yes

You can take a look to the pic and see that's problem in in \r and other WTF moments. Some thimes Linux make me cry...

See on i3blocks two links, now it's more clear for you, why me asking and .sh scripts. Mayme for you that's will be cool also :)

[image]

Why it's not working?

doegox, 2020-08-28:
It seems you're using a windows editor, try convert your scripts with `dos2unix`

Windslab, 2020-08-28:
> using a windows editor

Exactly! Thx to encoding...

closed 2020-08-28 by Windslab