RfidResearchGroup/proxmark3 #931: [BUG] Full screen mode freeze

Windslab, 2020-08-29:
Working excelent in the urxvt terminal mode:
[image]

Not working at all in fullscreen mode by i3 etc:
[image]

It looks like totally freezes, even a cursor. After you have backed to non full screen mode, all of you has been did will displayed in the command =>

The clear Arch system so that's your bug 👍 

PS: We need more colors in syntax highlight. Will work on this way?

iceman1001, 2020-08-29:
Closed by lacking of relevant information to debug.

closed 2020-08-29 by iceman1001

Windslab, 2020-08-29:
Wery radically!
I`ll give you it, could you please tell with one you need? Command? Grep? Logs?

Sorry I am not professional user of linux...

Windslab, 2020-08-30:
@iceman1001 How are you doing