RfidResearchGroup/proxmark3 #933: Mifare Magic card UID

mwalker33, 2020-09-02:
**Describe the bug**
When I run 'hf mf auto' on a Mifare Magic Card UID, it did not exist correctly.
When run on a real mifare classic, it worked as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Place UID Majic Card on proxmark
2. run "hf mf auto"

**Expected behavior**
Regardless of if the card can support a test or not, exit cleanly.

sample output
```
[usb] pm3 --> hf mf auto
[!] no known key was supplied, key recovery might fail
[+] loaded 23 keys from hardcoded default array
[=] running strategy 1

[=] Chunk: 0.9s | found 30/32 keys (23)

[=] running strategy 2

[=] Chunk: 0.9s | found 30/32 keys (23)

[+] target sector:  0 key type: A -- found valid key [FF FF FF FF FF FF ] (used for nested / hardnested attack)
[+] target sector:  0 key type: B -- found valid key [FF FF FF FF FF FF ]
[+] target sector:  2 key type: A -- found valid key [FF FF FF FF FF FF ]
[+] target sector:  2 key type: B -- found valid key [FF FF FF FF FF FF ]
[+] target sector:  3 key type: A -- found valid key [FF FF FF FF FF FF ]
[+] target sector:  3 key type: B -- found valid key [FF FF FF FF FF FF ]
[+] target sector:  4 key type: A -- found valid key [FF FF FF FF FF FF ]
[+] target sector:  4 key type: B -- found valid key [FF FF FF FF FF FF ]
[+] target sector:  5 key type: A -- found valid key [FF FF FF FF FF FF ]
[+] target sector:  5 key type: B -- found valid key [FF FF FF FF FF FF ]
[+] target sector:  6 key type: A -- found valid key [FF FF FF FF FF FF ]
[+] target sector:  6 key type: B -- found valid key [FF FF FF FF FF FF ]
[+] target sector:  7 key type: A -- found valid key [FF FF FF FF FF FF ]
[+] target sector:  7 key type: B -- found valid key [FF FF FF FF FF FF ]
[+] target sector:  8 key type: A -- found valid key [FF FF FF FF FF FF ]
[+] target sector:  8 key type: B -- found valid key [FF FF FF FF FF FF ]
[+] target sector:  9 key type: A -- found valid key [00 00 00 00 00 00 ]
[+] target sector:  9 key type: B -- found valid key [00 00 00 00 00 00 ]
[+] target sector: 10 key type: A -- found valid key [FF FF FF FF FF FF ]
[+] target sector: 10 key type: B -- found valid key [FF FF FF FF FF FF ]
[+] target sector: 11 key type: A -- found valid key [FF FF FF FF FF FF ]
[+] target sector: 11 key type: B -- found valid key [FF FF FF FF FF FF ]
[+] target sector: 12 key type: A -- found valid key [FF FF FF FF FF FF ]
[+] target sector: 12 key type: B -- found valid key [FF FF FF FF FF FF ]
[+] target sector: 13 key type: A -- found valid key [FF FF FF FF FF FF ]
[+] target sector: 13 key type: B -- found valid key [FF FF FF FF FF FF ]
[+] target sector: 14 key type: A -- found valid key [FF FF FF FF FF FF ]
[+] target sector: 14 key type: B -- found valid key [FF FF FF FF FF FF ]
[+] target sector: 15 key type: A -- found valid key [FF FF FF FF FF FF ]
[+] target sector: 15 key type: B -- found valid key [FF FF FF FF FF FF ]
[-] Nested attack failed, trying again (1/10)

[!] Button pressed. Aborted.

[usb] pm3 -->  
```
Notes:
All keys Except Sector "1" where found.
Button was not pressed.
LEDs A,C and D left on. - Pressing button does clear the LEDs back to all off.

**Desktop (please complete the following information):**
 - OS: Windows
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/psk-test/v4.9237-861-g20d55ec9-dirt...-unclean 2020-08-31 18:19:40
  compiled with MinGW-w64 8.2.0 OS:Windows (64b) ARCH:x86_64

 [ PROXMARK3 RDV4 ]
  external flash:                  present
  smartcard reader:                present

 [ PROXMARK3 RDV4 Extras ]
  FPC USART for BT add-on support: absent

 [ ARM ]
  bootrom: RRG/Iceman/master/c49bfa08 2020-05-06 17:25:38
       os: RRG/Iceman/psk-test/v4.9237-861-g20d55ec9-dirt...-unclean 2020-08-30 19:40:13
  compiled with GCC 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 288552 bytes (55%) Free: 235736 bytes (45%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
```
 - hw status
```
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size.............42260
[#]   Available memory........39952
[#] Tracing
[#]   tracing ................0
[#]   traceLen ...............221
[#]   dma8 memory.............-2111908
[#]   dma16 memory............-2111908
[#]   toSend memory...........39952
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
[#] Flash memory
[#]   Baudrate................24 MHz
[#]   Init....................OK
[#]   Memory size.............2 mbits / 256 kb
[#]   Unique ID...............0xD567A882A734BD26
[#] Smart card module (ISO 7816)
[#]   version.................v3.10
[#] LF Sampling config
[#]   [q] divisor.............95 ( 125.00 kHz )
[#]   [b] bits per sample.....8
[#]   [d] decimation..........1
[#]   [a] averaging...........Yes
[#]   [t] trigger threshold...0
[#]   [s] samples to skip.....0
[#] LF Sampling Stack
[#]   Max stack usage.........4872 / 8480 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
[#]    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |
[#]
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed............500ms
[#]   Bytes transferred.......245248
[#]   Transfer Speed PM3 -> Client = 490496 bytes/s
[#] Various
[#]   Max stack usage.........4872 / 8480 bytes
[#]   DBGLEVEL................1 ( ERROR )
[#]   ToSendMax...............39
[#]   ToSend BUFFERSIZE.......2308
[#]   Slow clock..............29636 Hz
[#] Installed StandAlone Mode
[#]   HF - Reading Visa cards & Emulating a Visa MSD Transaction(ISO14443) - (Salvador Mendoza)
[#] Flash memory dictionary loaded
[#]   Mifare..................859 keys
[#]   T55x7...................79 keys
```
 - data tune
```
[usb] pm3 --> data tune
[=] Measuring antenna characteristics, please wait...
[|]  9
[=] ---------- LF Antenna ----------
[+] LF antenna: 36.12 V - 125.00 kHz
[+] LF antenna: 26.12 V - 134.83 kHz
[+] LF optimal: 36.09 V - 123.71 kHz
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 47.03 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 88 is 134.83 kHz, 95 is 125.00 kHz.
```

iceman1001, 2020-09-02:
Try pulling and test now

mwalker33, 2020-09-02:
That works. All good.

Thanks

iceman1001, 2020-09-02:
some more fixes,   staticnested should be working better inside autopwn,   samething with generating uid filename for keyfile.
I am closing this one now.

closed 2020-09-02 by iceman1001