RfidResearchGroup/proxmark3 #934: [BUG] To do LF/CRLF

Windslab, 2020-09-02:
Whats poppin

The all `*.eml` files, obtained from Windows in `Windows (CRLF)` does't read by `*nix` at all. Beacouse `*nix` wish love only Unix (LF) encoding.

You move your dumps from one OS to other and you should change encoding by hands. That's not good.

iceman1001, 2020-09-02:
or you learn to use a tool for it...

closed 2020-09-02 by iceman1001

Windslab, 2020-09-02:
@iceman1001 Can you explain, why I need always use some kind of tool if you can once add a library for automaticaly converting?

You loosing your time to answer on issues like this, you can fix and that's end :)

doegox, 2020-09-02:
Submit us a PR if that's so easy to do

Windslab, 2020-09-02:
You have bash at linux and powershell at Windows...

mwalker33, 2020-09-02:
A few comments.
- I really dont think we should be looking at calling external commands that are OS dependent from within proxmark client.  So the use of bash (which may or may not be the shell on linux, or powershell on windows, which may or may not be functional (e.g. powershell is starting to get blocked by clients to prevent malware that uses it); is most likly going to end up with more issues then less.
BUT as an end user who works with both, should all ready be aware of the CRLF challenge and can then use what ever tools the choose.

- Most dumps now have 3 formats, josn, bin and eml/text the bin (for example) should be OS independent as is.

- Developers can choose where they want to spend their time.  Most will spend it on where they want to see improvements.   Everyone works differently, but for me, I like to focus on one thing at a time; so to spend time on what may be a short task/quick win, while working on more complex things leads to lots of lost time on the complex challenge.    When you comeback to a task, you have to revisit things and work out where you got to.   "get back into the zone"
As such, I personally will leave quick wins until I want a break from the complex tasks.

Windslab, 2020-09-02:
No problem you can use Python to convert.

Example:

LF to CRLF converting:

`with open('input.txt') as inp, open('output.txt', 'w') as out:
txt = inp.read()
txt = txt.replace('\n', '\r\n')
out.write(txt)`

Windslab, 2020-09-02:
@doegox, @iceman1001 But you can do this natively in C.

There you go! Now you doing best there!