RfidResearchGroup/proxmark3 #935: [idea] Dracula color cheme

Windslab, 2020-09-02:
As I see you added a colors or Windows in terminal! That's grait!

Could/can you add an https://draculatheme.com/ cheme in to Proxmark UI? That's one of best color chemes ever and it free to use.

https://github.com/dracula/dracula-theme

I promese you will apreciate.

Windslab, 2020-09-02:
Also we need rework the typography, I mean all symbols with standartization.

For example items must looks not like `--=`, it shall looks natively `—`
May be [usb] will change to $port number?
Will be better to see:
[COM3] pm3 -->
[TTY0] pm3 -->

Hope you understand a way! A lot of work!

doegox, 2020-09-02:
Colors: by default to remain compatible with all supported OSes, we need to use only the basic set of ANSI colors.
But there could be a feature to be able to personalize colors via config file, why not.
For the `[usb]`, this is not the port as seen by the computer but the port as seen by the proxmark: USB <> FPC
The port as seen by the computer is written in the first lines of the communication:
```
[=] Using UART port /dev/ttyACM0
[=] Communicating with PM3 over USB-CDC
```

iceman1001, 2020-09-02:
If you make a color scheme plugin to be used with the preference.json file,   go for it,  like @doegox  said.
By the nature of the requester,  I would suggest adding this to the _wishlist_  in the wiki pages instead as a _issue_

Windslab, 2020-09-02:
> For the `[usb]`, this is not the port as seen by the computer but the port as seen by the proxmark: USB <> FPC
Yes that's static label whitch have no profit leads. Better to see [COM3] at the it place. The green/red leads and connection status both.

Why I need to know any kind of the `USB <> FPC` some information about it? To the end user that's no need!

Example:
- In the Notepad++(Mousepad especialy for you @doegox), you have numbers of the lines 1,2,3,4,5,6 etc, imaginate that's you have [usb],[usb],[usb],[usb],[usb],[usb],[usb],[usb] on each line! You not happy. Yes? You better to add `_NO_` and then you will see where from till to has lost connection, or a port number both. If my port number has changed from 3 to 5 and then I had reconnect?

Eventually you can do this:
```
[21:54:01][COM3] --> hw search
[21:54:03][COM3] --> hf mf
[21:54:06][COM3] --> hw conncect com5
[21:54:21][COM5] --> hw conncect com5
[21:54:03][COM3] --> hf mf autopwn
```
etc...

Looks good yeah? If no:
```
[COM3] --> hw search
[COM3] --> hf mf
[COM3] --> hw conncect com5
[COM5] --> hw conncect com5
[COM3] --> hf mf autopwn
```
To the `preferences.json` add:
`"client.overlay.timeline": "off"`

> If you make a color scheme plugin to be used with the preference.json file, go for it, like @doegox said.
> By the nature of the requester, I would suggest adding this to the _wishlist_ in the wiki pages instead as a _issue_

That's will be grait.
Create an templeate please in `preferences.json` also I wish to see more colors, what do I mean:
Reduce not colored text, each item of the information shall have it color.

Example:

- client:  RRG/Iceman/master/v4.9237-822-g7af9fd3f-dirty-unclean 2020-08-18 10:23:06

That's is singe color. Each item must have it color. Example:

- client:(green) RRG/Iceman/master/v4.9237-822-g7af9fd3f-dirty-unclean(yellow) 2020-08-18(blue) 10:23:06(aqua)

[image]

`UI` that's not programming, that's since and you are trying to argue.

:)

doegox, 2020-09-02:
> Why I need to know any kind of the USB <> FPC some information about it? To the end user that's no need!

To you that's no need. But you're not the only end user.

For the color scheme, that's something requiring a lot of effort which is not in our plans, so, as @iceman1001 suggested, you can add it to the wishlist and maybe someone else interested and competent enough will do the job for you, or not.

closed 2020-09-02 by doegox

Windslab, 2020-09-02:
> To you that's no need. But you're not the only end user.

You exactly has answer for yourself, Philip!

I am an end user, not developer like you! In extended debug mode you can have all of you with! So that is the true way to add it is to it.

> For the color scheme, that's something requiring a lot of effort which is not in our plans, so, as @iceman1001 suggested, you can add it to the wishlist and maybe someone else interested and competent enough will do the job for you, or not.

All of I trying to add you are closing, that's why I am not contributing. You even not won't to make this topic readable to others, you just closing it fast as can.

You really can't make an auto encoding change? Common! :) Even I can that :)

doegox, 2020-09-02:
> Even I can that :)

Then submit your PR or add the idea to the wiki wishlist.
You're free to suggest ideas, we're free to spend our free time on it or not, that's how open source works.
Changing hardcoded colors by other hardcoded ones is easy, sure. But we won't do that because interoperability requires we stick to the basic ANSI colors. Implementing colors handling from a preferences file is a *huge* effort because today colors are set by macros and are textual strings, not functions.
We've far more issues to solve which are impacting basic usability of the device than spending time on colors.