RfidResearchGroup/proxmark3 #937: Compilation error armsrc/mifarecmd.c:1968 after commit c0c580bc6663253a01daa797e6573fc3becdc23a

Fl0-0, 2020-09-03:
***Compilation problem***

```
compiler version:  arm-none-eabi-gcc (15:8-2019-q3-1) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
[-] CC mifarecmd.c
mifarecmd.c: In function 'MifareECardLoad':
mifarecmd.c:1968:27: error: pointer targets in passing argument 1 of 'emlSetMem' differ in signedness [-Werror=pointer-sign]
                 emlSetMem("\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00", FirstBlockOfSector(sectorNo) + blockNo, 1);
                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from mifarecmd.c:21:
mifareutil.h:102:6: note: expected 'uint8_t *' {aka 'unsigned char *'} but argument is of type 'char *'
 void emlSetMem(uint8_t *data, int blockNum, int blocksCount);
      ^~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [../common_arm/Makefile.common:110: obj/mifarecmd.o] Error 1
make: *** [Makefile:134: armsrc/all] Error 2

```

**Describe the bug**
Compilation error after commit c0c580bc6663253a01daa797e6573fc3becdc23a

**To Reproduce**
Steps to reproduce the behavior:

`make clean && make` on Kali Linux

**Desktop (please complete the following information):**
 - OS: Kali Linux

iceman1001, 2020-09-03:
try pulling and see if it went away

Fl0-0, 2020-09-03:
It is gone. Thx @iceman1001

closed 2020-09-03 by Fl0-0