RfidResearchGroup/proxmark3 #943: `lf search` gets pm3 unresponsive with FDX-B tag when polling for EM4x50

doegox, 2020-09-04:
Bug is quite tricky to reproduce.
It happens only when a FDX tag is at some distance from the reader.
`lf search` execution timeouts at some point and the device is unresponsive and must be unplugged.
I tracked it down to 2 spots in lf search routine (i.e. when first spot is commented, problem appears on second spot) :
* `detect_4x50_block()`
* `read_em4x50_uid()`

Both are leading to `em4x50_read()` => `CMD_LF_EM4X50_READ` => ARM: `em4x50_read()` => `find_em4x50_tag()` => `find_single_listen_window()`

The `while` loop of `find_single_listen_window()` if we get `check_pulse_length(get_pulse_length(), 3 * EM4X50_T_TAG_FULL_PERIOD)` true and `check_pulse_length(get_pulse_length(), 2 * EM4X50_T_TAG_FULL_PERIOD)` false repeteadly.
But it seems that it's rather `get_pulse_length()` which never ends.

To reproduce:
* put a FDX on the antenna (I first used a glass implant, then I could reproduce the bug with a T5577: `lf fdx clone 992 001000358463`)
* run `lf em 4x50_info`
* move the tag slowly away from antenna and repeat until `lf em 4x50_info` timeouts

mwalker33, 2020-09-05:
Looking at the arm src, that loop under those conditions may never end.
I could not make it repeatably have true on the *3 and false on the *2, but if the card was such that it was true, it would loop forever.

```
 static bool find_single_listen_window(void) {

    // find single listen window

    int cnt_pulses = 0;

    while (cnt_pulses < EM4X50_T_WAITING_FOR_SNGLLIW) {

        // identification of listen window is done via evaluation of
        // pulse lengths
        if (check_pulse_length(get_pulse_length(), 3 * EM4X50_T_TAG_FULL_PERIOD)) {
            if (check_pulse_length(get_pulse_length(), 2 * EM4X50_T_TAG_FULL_PERIOD)) {
                // listen window found
                return true;
            }
// Need something here to ensure it does not loop for ever.
// What happens if we increment cnt_pulses here as well ?
        } else {
            cnt_pulses++;
        }
    }
    return false;
}
```

mwalker33, 2020-09-05:
"... But it seems that it's rather get_pulse_length() which never ends...."

That has 3 while loops with potential no exit either.
Since that is looking for the low-high-low, if the card stopped responding mid check, the next transition may never happen.

iceman1001, 2020-09-05:
it should have timeout added,    as in param,   and say if not meet the search critieria within the first 5000samples,   in each of those three loops,  then return false.

iceman1001, 2020-09-05:
I push a test fix,  with timeouts,
https://github.com/RfidResearchGroup/proxmark3/commit/6aa65b735b3767e4d824f12477f0c93efaed8976

The timelimit was arbitrary selected and should be considered as testvalue.  Maybe 10000, 12000, 16000, is better?

doegox, 2020-09-05:
I still get pm3 unresponsive after the timeout @iceman1001

doegox, 2020-09-05:
Testing the latest fix, it still becomes unresponsive and it seems even worse to me: it's also unresponsive when the FDX is in the middle of the antenna and readable with `lf fdx read`.

iceman1001, 2020-09-05:
and?

doegox, 2020-09-06:
so, much better now :)

closed 2020-09-06 by doegox