RfidResearchGroup/proxmark3 #944: Error getting the keys of a Mifare Classic Static nonce

Knocks83, 2020-09-05:
**Describe the bug**
When I try finding the keys of a static nonce card, it keeps saying he found x key candidates but doesn't find anything.
Command: `hf mf staticnested 1 15 A FFFFFFFFFFFF d`
Behavior:
```
[+] Testing known keys. Sector count 16
[=] Chunk: 0.4s | found 0/32 keys (24)
[+] Time to check 23 known keys: 0 seconds

[+] enter static nested key recovery
[+] Found 56516 key candidates
[\] 56448/56516 keys | 129.2 keys/sec | worst case    0.5 seconds
[+] target block:  0 key type: A

[+] Found 56516 key candidates
[|] 56448/56516 keys | 129.3 keys/sec | worst case    0.5 seconds
[+] target block:  4 key type: A

[+] Found 111521 key candidates
[|]111468/111521 keys | 128.7 keys/sec | worst case    0.4 seconds
[+] target block:  8 key type: A

[+] Found 111521 key candidates
[|]111468/111521 keys | 131.4 keys/sec | worst case    0.4 seconds
[+] target block: 16 key type: A

[+] Found 111521 key candidates
[|]111468/111521 keys | 131.4 keys/sec | worst case    0.4 seconds
[+] target block: 20 key type: A
```
and it goes on endlessly. 

**To Reproduce**
Steps to reproduce the behavior:
Try finding the keys of a static nonced Mifare Classic 1k tag

**Expected behavior**
It should give me the tag keys

**Desktop**
 - OS: Windows 10  Pro
 - UNOFFICIAL PROXMARK
 - hw version
```
 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-966-g7904d962-dirty-unclean 2020-09-05 18:46:29
  compiled with MinGW-w64 10.2.0 OS:Windows (64b) ARCH:x86_64

 [ PROXMARK3 ]

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-966-g7904d962-dirty-unclean 2020-09-05 18:48:12
       os: RRG/Iceman/master/v4.9237-966-g7904d962-dirty-unclean 2020-09-05 18:48:42
  compiled with GCC 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 255904 bytes (49%) Free: 268384 bytes (51%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
```

 - hw status
```
[#] Memory
[#]   BigBuf_size.............44136
[#]   Available memory........44136
[#] Tracing
[#]   tracing ................0
[#]   traceLen ...............339
[#]   dma8 memory.............-2110032
[#]   dma16 memory............-2110032
[#]   toSend memory...........-2110032
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
[#] LF Sampling config
[#]   [q] divisor.............95 ( 125.00 kHz )
[#]   [b] bits per sample.....8
[#]   [d] decimation..........1
[#]   [a] averaging...........Yes
[#]   [t] trigger threshold...0
[#]   [s] samples to skip.....0
[#] LF Sampling Stack
[#]   Max stack usage.........3952 / 8480 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |
[#]
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed............500ms
[#]   Bytes transferred.......248320
[#]   Transfer Speed PM3 -> Client = 496640 bytes/s
[#] Various
[#]   Max stack usage.........4112 / 8480 bytes
[#]   DBGLEVEL................1 ( ERROR )
[#]   ToSendMax...............12
[#]   ToSend BUFFERSIZE.......2308
[#]   Slow clock..............31593 Hz
[#] Installed StandAlone Mode
[#]   HF - Reading Visa cards & Emulating a Visa MSD Transaction(ISO14443) - (Salvador Mendoza)
```
 - data tune
```
[=] Measuring antenna characteristics, please wait...
[/] 10
[=] ---------- LF Antenna ----------
[+] LF antenna: 41.70 V - 125.00 kHz
[+] LF antenna: 35.90 V - 134.83 kHz
[+] LF optimal: 47.37 V - 129.03 kHz
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 25.01 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 88 is 134.83 kHz, 95 is 125.00 kHz.
```

iceman1001, 2020-09-05:
Does your card have a static nonce?

Knocks83, 2020-09-05:
When I run hf mf nested it says I do

iceman1001, 2020-09-05:
And how would we know that if you don't provide enough information ?

Knocks83, 2020-09-05:
> 
> 
> And how would we know that if you don't provide enough information ?

I said "When I try finding the keys of a static nonce card" at the beginning of the issue

Knocks83, 2020-09-05:
Edited it a little to make it more comprehensible

iceman1001, 2020-09-05:
I am afraid I will not be able to help you.  Maybe someone else can.

closed 2020-09-13 by iceman1001