RfidResearchGroup/proxmark3 #946: iCLASS 2K wrbl not working

jumpycalm, 2020-09-06:
**Describe the bug**
Someone reported this issue on the Proxmark forum, I was never able to post anything on the forum, so I'm bring up the issue here. Basically I am seeing this issue: http://www.proxmark.org/forum/viewtopic.php?id=8089
The issue is not repro on the 16K iCLASS. The issue is only repro on the 2K iCLASS. I tried the 2K iCLASS from 3 sellers and I got the same problem. Please note, the iCLASS fobs I purchased are just regular fobs (Not SE, SEO etc.) with default key. I am able to read no problem and I am able to program the fob with the iclassfield + HID Omnikey reader no problem.

**To Reproduce**
hf ic wrbl b 6 d BBBBBBBBBBBBBBBB k 0

**Expected behavior**
Write to block 6 success

**Screenshots**
[usb] pm3 --> hf ic rdbl b 6 k 0

[+]  block 06 : AA AA AA AA AA AA AA AA

[usb] pm3 --> hf ic wrbl b 6 d BBBBBBBBBBBBBBBB k 0
[+] Using key[0] AE A6 84 A6 DA B2 32 78
[-] Writing failed
[usb] pm3 --> hf ic in

[=] --- Tag Information --------------------------
[=] -------------------------------------------------------------
[+]     CSN: A3 D7 BC 10 FE FF 12 E0   (uid)
[+]  Config: 12 FF FF FF 7F 1F FF 3C   (Card configuration)
[+] E-purse: FC FF FF FF FF FF FF FF   (Card challenge, CC)
[+]      Kd: 00 00 00 00 00 00 00 00   (Debit key, hidden)
[+]      Kc: 00 00 00 00 00 00 00 00   (Credit key, hidden)
[+]     AIA: FF FF FF FF FF FF FF FF   (Application Issuer area)
[=] ------ card configuration ------
[+]   Mode: Application (locked)
[+] Coding: ISO 14443-2 B / 15693
[+]  Crypt: Secured page, keys not locked
[=]     RA: Read access not enabled
[=] App limit 0x12, OTP 0xFFFF, Block write lock 0xFF
[=]      Chip 0x7F, Mem 0x1F, EAS 0xFF, Fuses 0x3C
[=] ------ Memory ------
[=]     2 KBits/2 App Areas (256 bytes)
[=]     AA1 blocks 13 { 0x06 - 0x12 (06 - 18) }
[=]     AA2 blocks 18 { 0x13 - 0x1F (19 - 31) }
[=] ------ KeyAccess ------
[=]  Kd = Debit key (AA1),  Kc = Credit key (AA2)
[=]      Read A - Kd or Kc
[=]      Read B - Kd or Kc
[=]     Write A - Kc
[=]     Write B - Kc
[=]       Debit - Kd or Kc
[=]      Credit - Kc
[=] ------ Fingerprint ------
[+] CSN is in HID range
[+] Credential : iCLASS legacy
[+]  Card type : PicoPass 2K

**Desktop (please complete the following information):**
 - OS: WSL Debian
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-852-g6d5b12a7 2020-08-23 11:42:08
  compiled with GCC 8.3.0 OS:Linux ARCH:x86_64

 [ PROXMARK3 ]

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-608-g1f3e6714 2020-07-18 15:41:38
       os: RRG/Iceman/master/v4.9237-964-g643a53c5 2020-09-05 04:21:22
  compiled with GCC 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 255424 bytes (49%) Free: 268864 bytes (51%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

 - hw status
[#] Memory
[#]   BigBuf_size.............44128
[#]   Available memory........41308
[#] Tracing
[#]   tracing ................0
[#]   traceLen ...............182
[#]   dma8 memory.............-2110040
[#]   dma16 memory............41308
[#]   toSend memory...........41820
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
[#] LF Sampling config
[#]   [q] divisor.............95 ( 125.00 kHz )
[#]   [b] bits per sample.....8
[#]   [d] decimation..........1
[#]   [a] averaging...........Yes
[#]   [t] trigger threshold...0
[#]   [s] samples to skip.....0
[#] LF Sampling Stack
[#]   Max stack usage.........3960 / 8480 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |
[#]               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |
[#]    1 of 4 coding reference |  31 |  20 |  18 |  34 |  15 |  50 |  66 |
[#]
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed............500ms
[#]   Bytes transferred.......97280
[#]   Transfer Speed PM3 -> Client = 194560 bytes/s
[#] Various
[#]   Max stack usage.........4120 / 8480 bytes
[#]   DBGLEVEL................1 ( ERROR )
[#]   ToSendMax...............10
[#]   ToSend BUFFERSIZE.......2308
[#]   Slow clock..............31659 Hz
[#] Installed StandAlone Mode
[#]   HF - Reading Visa cards & Emulating a Visa MSD Transaction(ISO14443) - (Salvador Mendoza)

 - data tune
[usb] pm3 --> data tune
[=] Measuring antenna characteristics, please wait...
[-]  9
[=] ---------- LF Antenna ----------
[+] LF antenna: 34.77 V - 125.00 kHz
[+] LF antenna: 39.02 V - 134.83 kHz
[+] LF optimal: 42.02 V - 130.43 kHz
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 31.90 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 88 is 134.83 kHz, 95 is 125.00 kHz.

**Additional context**
Additional tests performed:
1. Tested on multiple Proxmark3 Easy as well as Proxmark3 RDV4, same result;
2. Tested the iCLASS fobs from multiple vendors, some of them are even order directly from HID dealer;
3. The iCLASS fobs I have are all programmed fobs with default key. But block 6 is not locked. I have no problem changing block 6 with iclassfiled + HID Omnikey reader.
4. I tried to debug this issue myself, but I went to a dead end. I put some debug output in my code, I found that the authentication passed. i.e. I was using the correct key. But the write operation failed at iclass_send_cmd_with_retries(). I tried to increase the retry counts as well as play around with the delays, but no success.

iceman1001, 2020-09-06:
Works like a charm for me.
I can see in your commands you don't use  *b 06*   ,  where you need to have the blocknumber as 2 hex symbols, like stated in the helptext.

Maybe you don't have the correct key? 

```
[usb] pm3 --> hf icl in

[=] --- Tag Information --------------------------
[=] -------------------------------------------------------------
[+]     CSN: 32 B3 D5 00 F7 FF 12 E0   (uid)
[+]  Config: 12 FF FF FF 7F 1F FF 3C   (Card configuration)
[+] E-purse: 8A FE FF FF FF FF FF FF   (Card challenge, CC)
[+]      Kd: 00 00 00 00 00 00 00 00   (Debit key, hidden)
[+]      Kc: 00 00 00 00 00 00 00 00   (Credit key, hidden)
[+]     AIA: FF FF FF FF FF FF FF FF   (Application Issuer area)
[=] ------ card configuration ------
[+]   Mode: Application (locked)
[+] Coding: ISO 14443-2 B / 15693
[+]  Crypt: Secured page, keys not locked
[=]     RA: Read access not enabled
[=] App limit 0x12, OTP 0xFFFF, Block write lock 0xFF
[=]      Chip 0x7F, Mem 0x1F, EAS 0xFF, Fuses 0x3C
[=] ------ Memory ------
[=]     2 KBits/2 App Areas (256 bytes)
[=]     AA1 blocks 13 { 0x06 - 0x12 (06 - 18) }
[=]     AA2 blocks 18 { 0x13 - 0x1F (19 - 31) }
[=] ------ KeyAccess ------
[=]  Kd = Debit key (AA1),  Kc = Credit key (AA2)
[=]      Read A - Kd or Kc
[=]      Read B - Kd or Kc
[=]     Write A - Kc
[=]     Write B - Kc
[=]       Debit - Kd or Kc
[=]      Credit - Kc
[=] ------ Fingerprint ------
[+] CSN is in HID range
[+] Credential : iCLASS legacy
[+]  Card type : PicoPass 2K
```

```
[usb] pm3 --> hf iclass wr
Write data to a iCLASS tag

Usage:  hf iclass wrbl b <block> d <data> k <key> [c|e|r|v]

Options:
  h         : Show this help
  b <block> : The block number as 2 hex symbols
  d <data>  : set the Data to write as 16 hex symbols
  k <key>   : access Key as 16 hex symbols or 1 hex to select key from memory
  c         : credit key assumed

  e         : elite computations applied to key
  r         : raw, no computations applied to key (raw)
  v         : verbose output
Examples:
        hf iclass wrbl b 0A d AAAAAAAAAAAAAAAA k 001122334455667B
        hf iclass wrbl b 1B d AAAAAAAAAAAAAAAA k 001122334455667B c
        hf iclass wrbl b 1B d AAAAAAAAAAAAAAAA k 0

[usb] pm3 --> hf iclass wr b 0c d 0102030405060708 k 0
[+] Using key[0] AE A6 84 A6 DA B2 32 78

[+] Wrote block 0C successful
[usb] pm3 -->
[usb] pm3 -->
[usb] pm3 --> hf iclas wr b 06 d bbbbbbbbbbbbbbbb k 0
[+] Using key[0] AE A6 84 A6 DA B2 32 78
[+] Wrote block 06 successful
[usb] pm3 --> hf iclass rd b 06 k 0

[+]  block 06 : BB BB BB BB BB BB BB BB

[usb] pm3 --> hf iclas list
```

*Run the check command against your tag*
`hf iclass chk f iclass_default_keys`

jumpycalm, 2020-09-06:
Thank you Iceman!
Same result for using 2 hex symbol and I do have the correct key:
```
[usb] pm3 --> hf ic wrbl b 06 d BBBBBBBBBBBBBBBB k 0
[+] Using key[0] AE A6 84 A6 DA B2 32 78
[-] Writing failed
[usb] pm3 --> hf iclass chk f iclass_default_keys
[+] loaded  7 keys from dictionary file /home/m9/rrg/client/dictionaries/iclass_default_keys.dic
[+] Reading tag CSN / CCNR...
[+]     CSN: D9 E7 BC 10 FE FF 12 E0
[+]    CCNR: FC FF FF FF FF FF FF FF 00 00 00 00
[+] Generating diversified keys
[+] Searching for DEBIT key...

[+] Found valid key AE A6 84 A6 DA B2 32 78

[+] time in iclass chk 1 seconds
[+] Key already at keyslot 0
```
Please note, the issue only reproducible with 2K iCLASS but not reproducible with 16K iCLASS. It works like a charm for me for any of my 16K iCLASS. I think jramb0 who reported this issue on the forum also using a 2K iCLASS. Do you have any 2K iCLASS you can try? `[+]  Card type : PicoPass 2K`

iceman1001, 2020-09-06:
My test was with a 
```
[=] ------ Fingerprint ------
[+] CSN is in HID range
[+] Credential : iCLASS legacy
[+]  Card type : PicoPass 2K
```

Did you have some distance between reader and tag?

jumpycalm, 2020-09-06:
Yes, I always put a sticky notes about 80mm thick between the reader and the tag for optimal reading and writing performance. Maybe I can send you some iCLASS tags I have to you to see if you can repro the issue? I can find your email address on the forum. I believe I am not the only one who is seeing this issue based on someone's post on the forum.

iceman1001, 2020-09-06:
80mm?   8cm  is quite far.    I guess you meant 8mm.  

Feel free to send some iCLASS tag,

jumpycalm, 2020-09-06:
Sorry, it's a typo. I will email you regarding the address to send the tags to. Hope you can get a repro. I tried to fix the issue from source code myself, I spent a day on it, no success.

iceman1001, 2020-10-23:
I may get a HID reader/writer,  and will be able to verify if the timings for writing is long enough.

iceman1001, 2020-11-09:
Thanks! 
I found the problem and pushed a fix.  Try it out

commit a41e71d0 (2020-11-09): fix #946,  unsecured card use crc16, secured MAC
closed 2020-11-09 by iceman1001