RfidResearchGroup/proxmark3 #950: make error  in raspberry 4b

dark8960, 2020-09-08:
CC src/emv/test/crypto_test.c
[-] CC src/emv/test/cryptotest.c
[-] CC src/emv/test/cda_test.c
[-] CC src/emv/test/dda_test.c
[-] CC src/emv/test/sda_test.c
[-] CC src/fido/additional_ca.c
[-] CC src/fido/cose.c
[-] CC src/fido/cbortools.c
[-] CC src/fido/fidocore.c
[-] CC src/fileutils.c
In function 'filelist',
    inlined from 'filelist.constprop.isra' at src/fileutils.c:1497:13:
src/fileutils.c:1484:9: error: 'strncat' output may be truncated copying 1023 bytes from a string of length 1023 [-Werror=stringop-truncation]
 1484 |         strncat(tmp_fullpath, path, sizeof(tmp_fullpath) - 1);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:706: obj/fileutils.o] Error 1
make: *** [Makefile:101: client/all] Error 2

doegox, 2020-09-08:
Can you tell us the info appearing at the beginning of the compilation?
Sth like
```
===================================================================
Client platform:   Linux
GUI support:       QT5 found, enabled
native BT support: Bluez not found, disabled
Jansson library:   system library found
Lua library:       system library found
Python3 library:   Python3 v3.8 found, enabled
Readline library:  enabled
Whereami library:  system library found
compiler version:  gcc (Debian 9.3.0-14) 9.3.0
===================================================================
```

I assume you're compiling the current RRG master branch?

dark8960, 2020-09-08:
root@kali:~/proxmark3# make

===================================================================
Platform name:     Proxmark3 rdv4
PLATFORM:          PM3RDV4
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH -DRDV4 LF HITAG EM4x50 ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   HF_MSDSAL
===================================================================
[*] MAKE client/all
===================================================================
Client platform:   Linux
GUI support:       QT5 found, enabled
native BT support: Bluez found, enabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 v3.8 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library                   compiler version:  gcc (Debian 10.2.0-5) 10.2.0
===================================================================
[-] CC src/aidsearch.c
[-] CC src/cmdanalyse.c
[-] CC src/cmdcrc.c
[-] CC src/cmddata.c
[-] CC src/cmdflashmem.c
[-] CC src/cmdflashmemspiffs.c
[-] CC src/cmdhf.c
[-] CC src/cmdhf14a.c
[-] CC src/cmdhf14b.c
[-] CC src/cmdhf15.c
[-] CC src/cmdhfcryptorf.c
[-] CC src/cmdhfepa.c
[-] CC src/cmdhffelica.c
[-] CC src/cmdhffido.c
[-] CC src/cmdhficlass.c
[-] CC src/cmdhflegic.c
[-] CC src/cmdhflist.c
[-] CC src/cmdhflto.c
[-] CC src/cmdhfmf.c
[-] CC src/cmdhfmfdes.c
[-] CC src/cmdhfmfhard.c
[-] CC src/cmdhfmfu.c
[-] CC src/cmdhfmfp.c
[-] CC src/cmdhfthinfilm.c
[-] CC src/cmdhftopaz.c
[-] CC src/cmdhw.c
[-] CC src/cmdlf.c
[-] CC src/cmdlfawid.c
[-] CC src/cmdlfcotag.c
[-] CC src/cmdlfem4x.c
[-] CC src/cmdlfem4x50.c
[-] CC src/cmdlffdx.c
[-] CC src/cmdlfguard.c
[-] CC src/cmdlfgallagher.c
[-] CC src/cmdlfhid.c
[-] CC src/cmdlfhitag.c
[-] CC src/cmdlfindala.c
[-] CC src/cmdlfio.c
[-] CC src/cmdlfjablotron.c
[-] CC src/cmdlfkeri.c
[-] CC src/cmdlfmotorola.c
[-] CC src/cmdlfnedap.c
[-] CC src/cmdlfnexwatch.c
[-] CC src/cmdlfnoralsy.c
[-] CC src/cmdlfpac.c
[-] CC src/cmdlfparadox.c
[-] CC src/cmdlfpcf7931.c
[-] CC src/cmdlfpresco.c
[-] CC src/cmdlfpyramid.c
[-] CC src/cmdlfsecurakey.c
[-] CC src/cmdlft55xx.c
[-] CC src/cmdlfti.c
[-] CC src/cmdlfviking.c
[-] CC src/cmdlfvisa2000.c
[-] CC src/cmdmain.c
[-] CC src/cmdparser.c
[-] CC src/cmdscript.c
[-] CC src/cmdsmartcard.c
[-] CC src/cmdtrace.c
[-] CC src/cmdusart.c
[-] CC src/cmdwiegand.c
[-] CC src/comms.c
[-] CC src/crypto/asn1dump.c
[-] CC src/crypto/asn1utils.c
[-] CC src/crypto/libpcrypto.c
[-] CC src/emv/apduinfo.c
[-] CC src/emv/cmdemv.c
[-] CC src/emv/crypto.c
[-] CC src/emv/crypto_polarssl.c
[-] CC src/emv/dol.c
[-] CC src/emv/dump.c
[-] CC src/emv/emv_pk.c
[-] CC src/emv/emv_pki.c
[-] CC src/emv/emv_pki_priv.c
[-] CC src/emv/emv_roca.c
[-] CC src/emv/emv_tags.c
[-] CC src/emv/emvcore.c
[-] CC src/emv/emvjson.c
[-] CC src/emv/tlv.c
[-] CC src/emv/test/crypto_test.c
[-] CC src/emv/test/cryptotest.c
[-] CC src/emv/test/cda_test.c
[-] CC src/emv/test/dda_test.c
[-] CC src/emv/test/sda_test.c
[-] CC src/fido/additional_ca.c
[-] CC src/fido/cose.c
[-] CC src/fido/cbortools.c
[-] CC src/fido/fidocore.c
[-] CC src/fileutils.c
In function 'filelist',
    inlined from 'filelist.constprop.isra' at src/fileutils.c:1497:13:
src/fileutils.c:1484:9: error: 'strncat' output may be truncated copying 1023 bytes from a string of length 1023 [-Werror=stringop-truncation]
 1484 |         strncat(tmp_fullpath, path, sizeof(tmp_fullpath) - 1);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:706: obj/fileutils.o] Error 1
make: *** [Makefile:101: client/all] Error 2
root@kali:~/proxmark3#

Is it the problem caused by the version of GCC?

dark8960, 2020-09-08:
Isroot@kali:~/proxmark3# make

===================================================================
Platform name:     Proxmark3 rdv4
PLATFORM:          PM3RDV4
Platform extras:   No extra selected
Included options:  SMARTCARD FLASH -DRDV4 LF HITAG EM4x50 ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF HFPLOT
Standalone mode:   HF_MSDSAL
===================================================================
[*] MAKE client/all
===================================================================
Client platform:   Linux
GUI support:       QT5 found, enabled
native BT support: Bluez found, enabled
Jansson library:   system library not found, using local library
Lua library:       system library not found, using local library
Python3 library:   Python3 v3.8 found, enabled
Readline library:  enabled
Whereami library:  system library not found, using local library                   compiler version:  gcc (Debian 10.2.0-5) 10.2.0
===================================================================
[-] CC src/aidsearch.c
[-] CC src/cmdanalyse.c
[-] CC src/cmdcrc.c
[-] CC src/cmddata.c
[-] CC src/cmdflashmem.c
[-] CC src/cmdflashmemspiffs.c
[-] CC src/cmdhf.c
[-] CC src/cmdhf14a.c
[-] CC src/cmdhf14b.c
[-] CC src/cmdhf15.c
[-] CC src/cmdhfcryptorf.c
[-] CC src/cmdhfepa.c
[-] CC src/cmdhffelica.c
[-] CC src/cmdhffido.c
[-] CC src/cmdhficlass.c
[-] CC src/cmdhflegic.c
[-] CC src/cmdhflist.c
[-] CC src/cmdhflto.c
[-] CC src/cmdhfmf.c
[-] CC src/cmdhfmfdes.c
[-] CC src/cmdhfmfhard.c
[-] CC src/cmdhfmfu.c
[-] CC src/cmdhfmfp.c
[-] CC src/cmdhfthinfilm.c
[-] CC src/cmdhftopaz.c
[-] CC src/cmdhw.c
[-] CC src/cmdlf.c
[-] CC src/cmdlfawid.c
[-] CC src/cmdlfcotag.c
[-] CC src/cmdlfem4x.c
[-] CC src/cmdlfem4x50.c
[-] CC src/cmdlffdx.c
[-] CC src/cmdlfguard.c
[-] CC src/cmdlfgallagher.c
[-] CC src/cmdlfhid.c
[-] CC src/cmdlfhitag.c
[-] CC src/cmdlfindala.c
[-] CC src/cmdlfio.c
[-] CC src/cmdlfjablotron.c
[-] CC src/cmdlfkeri.c
[-] CC src/cmdlfmotorola.c
[-] CC src/cmdlfnedap.c
[-] CC src/cmdlfnexwatch.c
[-] CC src/cmdlfnoralsy.c
[-] CC src/cmdlfpac.c
[-] CC src/cmdlfparadox.c
[-] CC src/cmdlfpcf7931.c
[-] CC src/cmdlfpresco.c
[-] CC src/cmdlfpyramid.c
[-] CC src/cmdlfsecurakey.c
[-] CC src/cmdlft55xx.c
[-] CC src/cmdlfti.c
[-] CC src/cmdlfviking.c
[-] CC src/cmdlfvisa2000.c
[-] CC src/cmdmain.c
[-] CC src/cmdparser.c
[-] CC src/cmdscript.c
[-] CC src/cmdsmartcard.c
[-] CC src/cmdtrace.c
[-] CC src/cmdusart.c
[-] CC src/cmdwiegand.c
[-] CC src/comms.c
[-] CC src/crypto/asn1dump.c
[-] CC src/crypto/asn1utils.c
[-] CC src/crypto/libpcrypto.c
[-] CC src/emv/apduinfo.c
[-] CC src/emv/cmdemv.c
[-] CC src/emv/crypto.c
[-] CC src/emv/crypto_polarssl.c
[-] CC src/emv/dol.c
[-] CC src/emv/dump.c
[-] CC src/emv/emv_pk.c
[-] CC src/emv/emv_pki.c
[-] CC src/emv/emv_pki_priv.c
[-] CC src/emv/emv_roca.c
[-] CC src/emv/emv_tags.c
[-] CC src/emv/emvcore.c
[-] CC src/emv/emvjson.c
[-] CC src/emv/tlv.c
[-] CC src/emv/test/crypto_test.c
[-] CC src/emv/test/cryptotest.c
[-] CC src/emv/test/cda_test.c
[-] CC src/emv/test/dda_test.c
[-] CC src/emv/test/sda_test.c
[-] CC src/fido/additional_ca.c
[-] CC src/fido/cose.c
[-] CC src/fido/cbortools.c
[-] CC src/fido/fidocore.c
[-] CC src/fileutils.c
In function 'filelist',
    inlined from 'filelist.constprop.isra' at src/fileutils.c:1497:13:
src/fileutils.c:1484:9: error: 'strncat' output may be truncated copying 1023 bytes from a string of length 1023 [-Werror=stringop-truncation]
 1484 |         strncat(tmp_fullpath, path, sizeof(tmp_fullpath) - 1);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:706: obj/fileutils.o] Error 1
make: *** [Makefile:101: client/all] Error 2
root@kali:~/proxmark3#

 it the problem caused by the version of GCC?
im sure that use  the current RRG master branch

dark8960, 2020-09-08:
[image]

doegox, 2020-09-08:
My first thought was that your recent GCC is flagging a potential issue ignored by previous versions, but I updated my gcc 10 to 10.2.0-6 and I can't reproduce the issue :( I need to dig more...

mwalker33, 2020-09-09:
I updated my kali to the latest (same gcc as op) and a pi 3b.
Prior to the update, no error.
Post the update, same error.

i.e. from a make clean, it will scroll though every file (as expected) until it hits fileutils.c
Making tmp_fullpath bigger seems to work
 char tmp_fullpath[1025] = {0}; // was 1024, changed to 1025

iceman1001, 2020-09-09:
or maybe have parentheses around   tmp_fullpath - 1 in that case?

mwalker33, 2020-09-09:
The warning is saying that the source might be truncated as is knows it could be 1024 bytes.
I believe it is tell us as that could mean it may NOT be null terminated.
To test, I set it back to 1024, then ensured the target was terminated (0x00)
```
        char tmp_fullpath[1024] = {0};

        strncat(tmp_fullpath, path, sizeof(tmp_fullpath)-1 );
        tmp_fullpath[1023] = 0x00;
```
this resulted in no warning/error and compiled.

dark8960, 2020-09-09:
Thanks for your replies, it seems to work, it compiles perfectly, thank you very much

closed 2020-09-09 by dark8960

mwalker33, 2020-09-09:
Not sure if we wish to close just yet.  
I was working on ideas to help with the homework as I could see and repeat the issue. 
@doegox  and @iceman1001  can work out the best way for the permanent fix.

iceman1001, 2020-09-09:
@mwalker33   I pushed your fix.

doegox, 2020-09-09:
yeah I would have loved to be able to reproduce the setup as well but, well, it'll comme sooner enough I guess...