RfidResearchGroup/proxmark3 #956: mfdes changekey doesn't support changing keys on EVx cards

mdlam92, 2020-09-16:
I have a few EV2 and EV3 cards that are fresh, default 16 byte 3DES keys for the PMK, and I can't seem to get the mfdes changekey commands to work. Both when I target the PICC master key and an application's master key.

Sometimes it reports:
`[usb] pm3 --> hf mfdes changekey -n 0 -t 2 -k 00000000000000000000000000000000 -u 2 -j 00000000000000000000000000000000
[!] ⚠️     Can't change key -> Current configuration/status does not allow the requested command`

or
`
[usb] pm3 --> hf mfdes changekey -n 1 -t 2 -k 00000000000000000000000000000000 -u 2 -j 00000000000000000000000000000000
[!] ⚠️     Can't change key -> Length of command string invalid`

or
`[usb] pm3 --> hf mfdes changekey -n 1 -t 2 -k 00000000000000000000000000000000 -u 2 -j 00000000000000000000000000000000
[!] ⚠️     Can't change key -> Value of the parameter(s) invalid`

depending on whether I select the PICC AID then auth then try changekey, or auth with the PMK then try changekey.

I think this is because in the mifare_desfire_change_key() method there is no keyset argument taken which is required for EVx change key commands. I might try and make a version of this method for EVx cards today, but I think the cryptogram is different for EVx cards also.

iceman1001, 2020-11-04:
You still have this issue?

georgethejanitor, 2020-11-04:
Yes I have followed all the steps on the repo and the YouTube video I can’t firgure out what I’m doing wrong.

mwalker33, 2021-05-17:
I have been testing on the EV1 cards.
I still have on my list look at changing the "same key" i.e. Auth with key X and change Key X 
Changing a different key should be OK now. (e.g. Auth with key 0 change Key 1) for des, 3des and AES.
If you could re-test and let me know (a copy of the command used and the hf mfdes list would be good as well.)

mwalker33, 2021-05-19:
We have also fix the change key with the same key used to authenticate.

Can you please test and report back any issues.

closed 2021-05-19 by iceman1001