RfidResearchGroup/proxmark3 #960: hf mf autopwn failed to dump with key B

Fl0-0, 2020-09-17:
**Describe the bug**
`hf mf autopwn` failed to dump with key B:

```
[=] fast dump reported back failure w KEY B
[=] Dump file is PARTIAL complete
```
`hf mf dump` works well.

**To Reproduce**
Steps to reproduce the behavior:
1. Choose a Mifare classic card with the right access bits (read data block only with key B)
2. Run `hf mf autopwn`
3. See error `fast dump reported back failure w KEY B `and the partial dumps
4. Run `hf mf dump`
5. Dump is OK

**Screenshots**
```
[usb] pm3 --> hf mf autopwn
[!] ⚠️  no known key was supplied, key recovery might fail
[+] loaded 23 keys from hardcoded default array
[=] running strategy 1
[=] Chunk: 1.2s | found 27/32 keys (23)
[=] running strategy 2
[=] Chunk: 1.2s | found 27/32 keys (23)
[+] target sector:  0 key type: A -- found valid key [ FF FF FF FF FF FF ] (used for nested / hardnested attack)
[+] target sector:  0 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  1 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  1 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  2 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  2 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  3 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  3 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  4 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  4 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  5 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  5 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  6 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  6 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  7 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  7 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  8 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  8 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  9 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  9 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector: 10 key type: A -- found valid key [ A0 A1 A2 A3 A4 A5 ]
[+] target sector: 11 key type: A -- found valid key [ A0 A1 A2 A3 A4 A5 ]
[+] target sector: 12 key type: A -- found valid key [ A0 A1 A2 A3 A4 A5 ]
[+] target sector: 13 key type: A -- found valid key [ A0 A1 A2 A3 A4 A5 ]
[+] target sector: 14 key type: A -- found valid key [ A0 A1 A2 A3 A4 A5 ]
[+] target sector: 15 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector: 15 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] Using AVX2 SIMD core.

 time    | #nonces | Activity                                                | expected to brute force
         |         |                                                         | #states         | time 
------------------------------------------------------------------------------------------------------
       0 |       0 | Start using 8 threads and AVX2 SIMD core                |                 |
       0 |       0 | Brute force benchmark: 983 million (2^29.9) keys/s      | 140737488355328 |    2d
       5 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |    2d
[#] AcquireNonces: Auth1 error
       8 |     112 | Apply bit flip properties                               |     26594650112 |   27s
[#] AcquireNonces: Auth1 error
[------ CUT -----]
[#] AcquireNonces: Auth1 error
[#] AcquireNonces: Auth1 error
      19 |    1443 | Apply Sum property. Sum(a0) = 120                       |       158699808 |    0s
      20 |    1443 | Brute force phase completed. Key found: XXXXXXXXXXXX   |               0 |    0s
[+] target sector: 10 key type: B -- found valid key [ XX XX XX XX XX XX ]
[... 112 lines cut ...]
[+] successfully read block  1 of sector 13.
[+] successfully read block  2 of sector 13.
[+] successfully read block  3 of sector 13.
[+] successfully read block  0 of sector 14.
[#] Card didn't answer to CL1 select all
[#] Can't select card
[+] successfully read block  1 of sector 14.
[+] successfully read block  2 of sector 14.
[+] successfully read block  3 of sector 14.
[+] successfully read block  0 of sector 15.
[+] successfully read block  1 of sector 15.
[+] successfully read block  2 of sector 15.
[+] successfully read block  3 of sector 15.
[+] time: 18 seconds

[+] Succeeded in dumping all blocks

[+] saved 1024 bytes to binary file hf-mf-9XXXXXXE-dump-6.bin
[+] saved 64 blocks to text file hf-mf-9XXXXXXE-dump-6.eml
[+] saved to json file hf-mf-9XXXXXXE-dump-6.json

```

**Desktop (please complete the following information):**
 - OS: Kali Linux

iceman1001, 2020-09-17:
I would think your card has a bad coupling with the device,  so you need to find a better spot where you don't get all the failed selects and auths.   Those failed ones is most likely the cause to your partial dump.

Fl0-0, 2020-09-17:
You are right for the better spot (small patch tags could be tricky), selects and auths errors are due to that. I think it is not the reason because with perfect coupling i have the same issue: the `autopwn` dump is missing key B read blocks. `hf mf dump` always works fine regarding key B reads.

```
[usb] pm3 --> hf mf autopwn
[!] ⚠️  no known key was supplied, key recovery might fail
[+] loaded 23 keys from hardcoded default array
[=] running strategy 1
[=] Chunk: 1.2s | found 27/32 keys (23)
[=] running strategy 2
[=] Chunk: 1.2s | found 27/32 keys (23)
[+] target sector:  0 key type: A -- found valid key [ FF FF FF FF FF FF ] (used for nested / hardnested attack)
[+] target sector:  0 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  1 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  1 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  2 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  2 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  3 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  3 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  4 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  4 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  5 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  5 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  6 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  6 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  7 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  7 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  8 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  8 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  9 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  9 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector: 10 key type: A -- found valid key [ A0 A1 A2 A3 A4 A5 ]
[+] target sector: 11 key type: A -- found valid key [ A0 A1 A2 A3 A4 A5 ]
[+] target sector: 12 key type: A -- found valid key [ A0 A1 A2 A3 A4 A5 ]
[+] target sector: 13 key type: A -- found valid key [ A0 A1 A2 A3 A4 A5 ]
[+] target sector: 14 key type: A -- found valid key [ A0 A1 A2 A3 A4 A5 ]
[+] target sector: 15 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector: 15 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] Using AVX2 SIMD core.

 time    | #nonces | Activity                                                | expected to brute force
         |         |                                                         | #states         | time 
------------------------------------------------------------------------------------------------------
       0 |       0 | Start using 8 threads and AVX2 SIMD core                |                 |
       0 |       0 | Brute force benchmark: 944 million (2^29.8) keys/s      | 140737488355328 |    2d
       5 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |    2d
       8 |     112 | Apply bit flip properties                               |    117135532032 |  2min
       9 |     224 | Apply bit flip properties                               |      9343000576 |   10s
      10 |     336 | Apply bit flip properties                               |      4195979264 |    4s
      11 |     447 | Apply bit flip properties                               |      2056948096 |    2s
      12 |     559 | Apply bit flip properties                               |      1709359104 |    2s
      12 |     671 | Apply bit flip properties                               |      1709359104 |    2s
      13 |     783 | Apply bit flip properties                               |      1601324928 |    2s
      13 |     893 | Apply bit flip properties                               |      1601324928 |    2s
      14 |    1002 | Apply bit flip properties                               |      1601324928 |    2s
      15 |    1112 | Apply bit flip properties                               |      1601324928 |    2s
      16 |    1223 | Apply bit flip properties                               |      1601324928 |    2s
      17 |    1334 | Apply bit flip properties                               |      1601324928 |    2s
      19 |    1443 | Apply Sum property. Sum(a0) = 120                       |       158699808 |    0s
      20 |    1443 | Brute force phase completed. Key found: XXXXXXXXXXXX   |               0 |    0s
[+] target sector: 10 key type: B -- found valid key [ XX XX XX XX XX XX ]
[+] target sector: 11 key type: B -- found valid key [ XX XX XX XX XX XX ]
[+] target sector: 12 key type: B -- found valid key [ XX XX XX XX XX XX ]
[+] target sector: 13 key type: B -- found valid key [ XX XX XX XX XX XX ]
[+] target sector: 14 key type: B -- found valid key [ XX XX XX XX XX XX ]

[+] found keys:
[+] |-----|----------------|---|----------------|---|
[+] | Sec | key A          |res| key B          |res|
[+] |-----|----------------|---|----------------|---|
[+] | 000 | ffffffffffff   | D | ffffffffffff   | D |
[... 214 lines cut ...]
[#] HF 14a config
[#] [a] Anticol override......0: No (follow standard)
[#] [b] BCC override..........0: No (follow standard)
[... cut ...]
[#]   Transfer Speed PM3 -> Client = 561152 bytes/s
[#] Various
[#]   Max stack usage.........4096 / 8480 bytes
[#]   DBGLEVEL................1 ( ERROR )
[#]   ToSendMax...............-1
[#]   ToSend BUFFERSIZE.......2308
[#]   Slow clock..............31480 Hz
[#] Installed StandAlone Mode
[#]   HF Mifare sniff/simulation - (Craig Young)
[#] Flash memory dictionary loaded
[#]   Mifare..................933 keys
[#]   T55x7...................110 keys
[#]   iClass..................7 keys
[usb] pm3 --> 
```

iceman1001, 2020-09-17:
good good,   at least one cause is gone and only the B key.
Now,  which are the access rights for that sector?   Both key A and B fails three times...

Fl0-0, 2020-09-18:
Sectors 10 to 14 have the same access rights and the same keys:
_a0a1 a2a3 a4a5_ **0f00 ff**ff _XXXX XXXX XXXX_

`0F00FF` access conditions tells us that block 0,1 and 2 of the sector could be read or write only by key B.

iceman1001, 2020-09-18:
Alright, something to work with. 
Here I set block 7 to same access rights as yours.
```
[usb] pm3 --> hf mf csetb 7 a0a1a2a3a4a50f00ffffFFFFFFFFFFFF
--block number: 7 data:A0 A1 A2 A3 A4 A5 0F 00 FF FF FF FF FF FF FF FF
[usb] pm3 --> hf mf cgetsc 1

  # | data  - sector 01 / 0x01
----+------------------------------------------------
  4 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  5 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  6 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  7 | A0 A1 A2 A3 A4 A5 0F 00 FF FF FF FF FF FF FF FF

Sector trailer decoded:
----------------------------------------------
Key A      A0A1A2A3A4A5
Key B      FFFFFFFFFFFF
Access rights
  block 4  rdB wrB
  block 5  rdB wrB
  block 6  rdB wrB
  block 7  wrAbyB rdCbyAB wrCbyB wrBbyB
UserData   0xff
```

Autopwn in action
```
[usb] pm3 --> hf mf autopwn
[#] 1 static nonce 01200145
[!] ⚠️  no known key was supplied, key recovery might fail
[+] loaded 23 keys from hardcoded default array
[=] running strategy 1
[=] Chunk: 0.7s | found 32/32 keys (23)
[+] target sector:  0 key type: A -- found valid key [ FF FF FF FF FF FF ] (used for nested / hardnested attack)
[+] target sector:  0 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  1 key type: A -- found valid key [ A0 A1 A2 A3 A4 A5 ]
[+] target sector:  1 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  2 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  2 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  3 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  3 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  4 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  4 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  5 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  5 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  6 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  6 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  7 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  7 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  8 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  8 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  9 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  9 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector: 10 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector: 10 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector: 11 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector: 11 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector: 12 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector: 12 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector: 13 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector: 13 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector: 14 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector: 14 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector: 15 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector: 15 key type: B -- found valid key [ FF FF FF FF FF FF ]

[+] found keys:
[+] |-----|----------------|---|----------------|---|
[+] | Sec | key A          |res| key B          |res|
[+] |-----|----------------|---|----------------|---|
[+] | 000 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 001 | a0a1a2a3a4a5   | D | ffffffffffff   | D |
[+] | 002 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 003 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 004 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 005 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 006 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 007 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 008 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 009 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 010 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 011 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 012 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 013 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 014 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 015 | ffffffffffff   | D | ffffffffffff   | D |
[+] |-----|----------------|---|----------------|---|
[=] ( D:Dictionary / S:darkSide / U:User / R:Reused / N:Nested / H:Hardnested / C:statiCnested / A:keyA )

[+] Generating binary key file
[+] Found keys have been dumped to hf-mf-01020304-key-2.bin--> 0xffffffffffff has been inserted for unknown keys.
[+] transferring keys to simulator memory (Cmd Error: 04 can occur)
[#] Cmd Error 04
[#] wrong response len 0 (expected 18)
[#] wrong response len 0 (expected 18)
[#] wrong response len 0 (expected 18)
[=] fast dump reported back failure w KEY A,  swapping to KEY B
[=] downloading the card content from emulator memory
[+] saved 1024 bytes to binary file hf-mf-01020304-dump-1.bin
[+] saved 64 blocks to text file hf-mf-01020304-dump-1.eml
[+] saved to json file hf-mf-01020304-dump-1.json
[=] autopwn execution time: 2 seconds
[usb] pm3 -->

```
As expected,  keyA fails and the keyB works. 

Only difference now is that your keyB was found be hardnested.   And since you filtered it out, I can't see if its used correct.
Does your "keyfile"  or the json dump file the correct keyB?

Fl0-0, 2020-09-18:
Yes the correct keys are inside partial dumps !

When i try it with a gen 1a card with the same random B key on 4 sectors it works fine but the key was found by nested not hardnested:

```
[+] Card wiped successfully
[usb] pm3 --> hf mf cwipe
 🕗 wipe block 63
[+] Card wiped successfully
[usb] pm3 --> hf mf csetb 47 a0a1a2a3a4a50f00ffffc95e34c0a15e
--block number:47 data:A0 A1 A2 A3 A4 A5 0F 00 FF FF C9 5E 34 C0 A1 5E 
[usb] pm3 --> hf mf csetb 51 a0a1a2a3a4a50f00ffffc95e34c0a15e
--block number:51 data:A0 A1 A2 A3 A4 A5 0F 00 FF FF C9 5E 34 C0 A1 5E 
[usb] pm3 --> hf mf csetb 55 a0a1a2a3a4a50f00ffffc95e34c0a15e
--block number:55 data:A0 A1 A2 A3 A4 A5 0F 00 FF FF C9 5E 34 C0 A1 5E 
[usb] pm3 --> hf mf csetb 59 a0a1a2a3a4a50f00ffffc95e34c0a15e
--block number:59 data:A0 A1 A2 A3 A4 A5 0F 00 FF FF C9 5E 34 C0 A1 5E 
[usb] pm3 --> hf mf cgetsc 14

  # | data  - sector 14 / 0x0E 
----+------------------------------------------------
 56 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 57 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 58 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 59 | A0 A1 A2 A3 A4 A5 0F 00 FF FF C9 5E 34 C0 A1 5E 

Sector trailer decoded:
----------------------------------------------
Key A      A0A1A2A3A4A5
Key B      C95E34C0A15E
Access rights
  block 56  rdB wrB
  block 57  rdB wrB
  block 58  rdB wrB
  block 59  wrAbyB rdCbyAB wrCbyB wrBbyB
UserData   0xff
----------------------------------------------
```

```
[usb] pm3 --> hf mf autopwn
[#] 1 static nonce 01200145
[!] ⚠️  no known key was supplied, key recovery might fail
[+] loaded 23 keys from hardcoded default array
[=] running strategy 1
[=] Chunk: 1.2s | found 28/32 keys (23)
[=] running strategy 2
[=] Chunk: 1.2s | found 28/32 keys (23)
[+] target sector:  0 key type: A -- found valid key [ FF FF FF FF FF FF ] (used for nested / hardnested attack)
[+] target sector:  0 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  1 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  1 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  2 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  2 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  3 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  3 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  4 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  4 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  5 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  5 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  6 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  6 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  7 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  7 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  8 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  8 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  9 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector:  9 key type: B -- found valid key [ FF FF FF FF FF FF ]
[+] target sector: 10 key type: A -- found valid key [ FF FF FF FF FF FF ]
[+] target sector: 10 key type: B -- found valid key [ FF FF FF FF FF FF ]
[... 28 lines cut ...]
[+] | 010 | ffffffffffff   | D | ffffffffffff   | D |
[+] | 011 | a0a1a2a3a4a5   | D | c95e34c0a15e   | C |
[+] | 012 | a0a1a2a3a4a5   | D | c95e34c0a15e   | R |
[+] | 013 | a0a1a2a3a4a5   | D | c95e34c0a15e   | R |
[+] | 014 | a0a1a2a3a4a5   | D | c95e34c0a15e   | R |
[+] | 015 | ffffffffffff   | D | ffffffffffff   | D |
[+] |-----|----------------|---|----------------|---|
[=] ( D:Dictionary / S:darkSide / U:User / R:Reused / N:Nested / H:Hardnested / C:statiCnested / A:keyA )

[+] Generating binary key file
[+] Found keys have been dumped to hf-mf-01020304-key-1.bin--> 0xffffffffffff has been inserted for unknown keys.
[+] transferring keys to simulator memory (Cmd Error: 04 can occur)
[#] Cmd Error 04
[#] wrong response len 0 (expected 18)
[#] wrong response len 0 (expected 18)
[#] wrong response len 0 (expected 18)
[=] fast dump reported back failure w KEY A,  swapping to KEY B
[=] downloading the card content from emulator memory
[+] saved 1024 bytes to binary file hf-mf-01020304-dump-1.bin
[+] saved 64 blocks to text file hf-mf-01020304-dump-1.eml
[+] saved to json file hf-mf-01020304-dump-1.json
[=] autopwn execution time: 182 seconds
```

Maybe it is related to hardnested found keys, i will try it all my 'hardnested' card.

iceman1001, 2020-09-18:
It seem to related to the hardnesteded recovery.

Fl0-0, 2020-09-18:
OK i've tried it with a 'real' card (a blank mifare plus)

```
[usb] pm3 --> hf search                       
 🕘  Searching for ISO14443-A tag...          
[+]  UID: 9D 33 12 EA 
[+] ATQA: 00 02
[+]  SAK: 18 [2]
[+] Possible types:
[+]    MIFARE Classic 1K / Classic 1K CL2
[+]    MIFARE Plus 2K / Plus EV1 2K
[+]    MIFARE Plus CL2 2K / Plus CL2 EV1 2K
[+]    MIFARE Plus 2K / Plus CL2 2K
[+]    MIFARE Classic 4K / Classic 4K CL2
[=] SAK incorrectly claims that card doesn't support RATS
[+]  ATS: 0C 75 77 80 02 C1 05 2F 2F 00 35 C7 60 D3 
[+]        -  TL : length is 12 bytes
[+]        -  T0 : TA1 is present, TB1 is present, TC1 is present, FSCI is 5 (FSC = 64)
[+]        - TA1 : different divisors are supported, DR: [2, 4, 8], DS: [2, 4, 8]
[+]        - TB1 : SFGI = 0 (SFGT = (not needed) 0/fc), FWI = 8 (FWT = 1048576/fc)
[+]        - TC1 : NAD is NOT supported, CID is supported
[+] Prng detection: hard
[?] Hint: try `hf mfp info`

[+] Valid ISO14443-A tag found
```
it is blank:

```
[+] found keys:
[+] |-----|----------------|---|----------------|---|
[+] | Sec | key A          |res| key B          |res|
[+] |-----|----------------|---|----------------|---|
[+] | 000 | ffffffffffff   | D | 000000000000   | D |
[+] | 001 | ffffffffffff   | D | 000000000000   | D |
[+] | 002 | ffffffffffff   | D | 000000000000   | D |
[+] | 003 | ffffffffffff   | D | 000000000000   | D |
[+] | 004 | ffffffffffff   | D | 000000000000   | D |
[+] | 005 | ffffffffffff   | D | 000000000000   | D |
[+] | 006 | ffffffffffff   | D | 000000000000   | D |
[+] | 007 | ffffffffffff   | D | 000000000000   | D |
[+] | 008 | ffffffffffff   | D | 000000000000   | D |
[+] | 009 | ffffffffffff   | D | 000000000000   | D |
[+] | 010 | ffffffffffff   | D | 000000000000   | D |
[+] | 011 | ffffffffffff   | D | 000000000000   | D |
[+] | 012 | ffffffffffff   | D | 000000000000   | D |
[+] | 013 | ffffffffffff   | D | 000000000000   | D |
[+] | 014 | ffffffffffff   | D | 000000000000   | D |
[+] | 015 | ffffffffffff   | D | 000000000000   | D |
[+] |-----|----------------|---|----------------|---|
```

```
[usb] pm3 --> hf mf wrbl 59 A FFFFFFFFFFFF a0a1a2a3a4a50f00ffffc95e34c0a15e
--block no 59, key A - FF FF FF FF FF FF 
--data: A0 A1 A2 A3 A4 A5 0F 00 FF FF C9 5E 34 C0 A1 5E 
isOk:01
```

it fails the same way:

```
[usb] pm3 --> hf mf autopwn
[!] ⚠️  no known key was supplied, key recovery might fail
[+] loaded 23 keys from hardcoded default array
[=] running strategy 1
[=] Chunk: 0.9s | found 31/32 keys (23)
[=] running strategy 2
[=] Chunk: 0.9s | found 31/32 keys (23)
[+] target sector:  0 key type: A -- found valid key [ FF FF FF FF FF FF ] (used for nested / hardnested attack)
[... 79 lines cut ...]
[+] | 014 | a0a1a2a3a4a5   | D | c95e34c0a15e   | H |
[+] | 015 | ffffffffffff   | D | 000000000000   | D |
[+] |-----|----------------|---|----------------|---|
[=] ( D:Dictionary / S:darkSide / U:User / R:Reused / N:Nested / H:Hardnested / C:statiCnested / A:keyA )

[+] Generating binary key file
[+] Found keys have been dumped to hf-mf-9D3312EA-key-3.bin--> 0xffffffffffff has been inserted for unknown keys.
[+] transferring keys to simulator memory (Cmd Error: 04 can occur)
[#] Cmd Error 04
[#] wrong response len 0 (expected 18)
[#] wrong response len 0 (expected 18)
[#] wrong response len 0 (expected 18)
[=] fast dump reported back failure w KEY A,  swapping to KEY B
[#] Cmd Error 04
[#] wrong response len 0 (expected 18)
[#] wrong response len 0 (expected 18)
[#] wrong response len 0 (expected 18)
[=] fast dump reported back failure w KEY B
[=] Dump file is PARTIAL complete
[=] downloading the card content from emulator memory
[+] saved 1024 bytes to binary file hf-mf-9D3312EA-dump-4.bin
[+] saved 64 blocks to text file hf-mf-9D3312EA-dump-4.eml
[+] saved to json file hf-mf-9D3312EA-dump-4.json
[=] autopwn execution time: 108 seconds
```

Fl0-0, 2020-09-18:
i tried the release v4.9237 and it works fine !

```
┌──(flo㉿kali)-[~/tools/NFC/RfidResearchGroup-proxmark3]
└─$ git reset --hard v4.9237                                 
HEAD is now at 833bc4d9 Release v4.9237 - Ice Coffee :coffee:
                                                                                                                                                                                                                                              
┌──(flo㉿kali)-[~/tools/NFC/RfidResearchGroup-proxmark3]
└─$ make clean && make     
```

iceman1001, 2020-09-18:
ok,  the timeout is 2 sec,  so the card might be dumping,  but  client timesout.  

Try changing these two timesout to  4000  instead and see if that solves your problem
https://github.com/RfidResearchGroup/proxmark3/blob/master/client/src/cmdhfmf.c#L925-L939

Fl0-0, 2020-09-18:
No it doesn't:

```
[+] Generating binary key file
[+] Found keys have been dumped to hf-mf-9D3312EA-key-5.bin--> 0xffffffffffff has been inserted for unknown keys.
[+] transferring keys to simulator memory (Cmd Error: 04 can occur)
[#] Cmd Error 04
[#] wrong response len 0 (expected 18)
[#] wrong response len 0 (expected 18)
[#] wrong response len 0 (expected 18)
[=] fast dump reported back failure w KEY A,  swapping to KEY B
[#] Cmd Error 04
[#] wrong response len 0 (expected 18)
[#] wrong response len 0 (expected 18)
[#] wrong response len 0 (expected 18)
[=] fast dump reported back failure w KEY B
[=] Dump file is PARTIAL complete
[=] downloading the card content from emulator memory
[+] saved 1024 bytes to binary file hf-mf-9D3312EA-dump-6.bin
[+] saved 64 blocks to text file hf-mf-9D3312EA-dump-6.eml
[+] saved to json file hf-mf-9D3312EA-dump-6.json
[=] autopwn execution time: 131 seconds
[usb] pm3 --> 

                                                                                                                                                                                                                                              
┌──(flo㉿kali)-[~/tools/NFC/RfidResearchGroup-proxmark3/client]
└─$ git diff
diff --git a/client/src/cmdhfmf.c b/client/src/cmdhfmf.c
index 6bd4c9c2..f105ab05 100644
--- a/client/src/cmdhfmf.c
+++ b/client/src/cmdhfmf.c
@@ -922,7 +922,7 @@ static int FastDumpWithEcFill(uint8_t numsectors) {
     clearCommandBuffer();
     SendCommandNG(CMD_HF_MIFARE_EML_LOAD, (uint8_t *)&payload, sizeof(payload));
 
-    bool res = WaitForResponseTimeout(CMD_HF_MIFARE_EML_LOAD, &resp, 2000);
+    bool res = WaitForResponseTimeout(CMD_HF_MIFARE_EML_LOAD, &resp, 4000);
     if (res == false) {
         PrintAndLogEx(WARNING, "Command execute timeout");
         return PM3_ETIMEOUT;
@@ -936,7 +936,7 @@ static int FastDumpWithEcFill(uint8_t numsectors) {
 
         clearCommandBuffer();
         SendCommandNG(CMD_HF_MIFARE_EML_LOAD, (uint8_t *)&payload, sizeof(payload));
-        res = WaitForResponseTimeout(CMD_HF_MIFARE_EML_LOAD, &resp, 2000);
+        res = WaitForResponseTimeout(CMD_HF_MIFARE_EML_LOAD, &resp, 4000);
         if (res == false) {
             PrintAndLogEx(WARNING, "Command execute timeout");
             return PM3_ETIMEOUT;

```

iceman1001, 2020-09-18:
Bugger,  lets enable some debug output
```
hw dbg 2
hf mf autopwn
```

Fl0-0, 2020-09-18:
It look like fast dump tries to read sector 0 !

iceman1001, 2020-09-18:
you need the emulator mem to have keys first. 
``` 
hw dbg 2
hf mf ecfill
```

And yes,  ecfill tries to read sector 0 to x,  it doesn't keep track of which sectors / blocks already read.
which I assume is the reason to the problem.

Fl0-0, 2020-09-18:
```
[usb] pm3 --> hw dbg 2
[usb] pm3 --> 
[#]   DBGLEVEL................2 ( INFO )

[usb] pm3 --> hf mf ekeyprn
[+] |-----|----------------|---|----------------|---|
[+] | Sec | key A          |res| key B          |res|
[+] |-----|----------------|---|----------------|---|
[+] | 000 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 001 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 002 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 003 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 004 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 005 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 006 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 007 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 008 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 009 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 010 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 011 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 012 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 013 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 014 | a0a1a2a3a4a5   | 1 | c95e34c0a15e   | 1 |
[+] | 015 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] |-----|----------------|---|----------------|---|
[+] ( 0:Failed / 1:Success)
[usb] pm3 --> hf mf ecfill A
--params: numSectors: 16, keyType: A

[usb] pm3 --> 
[#] Cmd Error 04
[#] Error reading sector 14 block  0
[#] wrong response len 0 (expected 18)
[#] Error reading sector 14 block  1
[#] wrong response len 0 (expected 18)
[#] Error reading sector 14 block  2
[#] wrong response len 0 (expected 18)
[#] Error reading sector 14 block  3
[#] Sector[15]. Auth nested error

[usb] pm3 --> hf mf ecfill B
--params: numSectors: 16, keyType: B

[usb] pm3 --> 
[#] Cmd Error 04
[#] Error reading sector  0 block  0
[#] wrong response len 0 (expected 18)
[#] Error reading sector  0 block  1
[#] wrong response len 0 (expected 18)
[#] Error reading sector  0 block  2
[#] wrong response len 0 (expected 18)
[#] Error reading sector  0 block  3
[#] Sector[ 1]. Auth nested error

[usb] pm3 --> hf mf eview
[=] downloading from emulator memory

[=] ----+-------------------------------------------------+-----------------
[=] blk | data                                            | ascii
[=] ----+-------------------------------------------------+-----------------
[=] 000 | 9D 33 12 EA 56 18 02 00 04 BA 24 15 BA 90 06 11 | .3..V.....$.....
[=] 001 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 002 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 003 | FF FF FF FF FF FF FF 07 80 69 00 00 00 00 00 00 | .........i......
[=] 004 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 005 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 006 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 007 | FF FF FF FF FF FF FF 07 80 69 00 00 00 00 00 00 | .........i......
[=] 008 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[... 33 lines cut ...]
[=] 042 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 043 | FF FF FF FF FF FF FF 07 80 69 00 00 00 00 00 00 | .........i......
[=] 044 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 045 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 046 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 047 | FF FF FF FF FF FF FF 07 80 69 00 00 00 00 00 00 | .........i......
[=] 048 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 049 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 050 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 051 | FF FF FF FF FF FF FF 07 80 69 00 00 00 00 00 00 | .........i......
[=] 052 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 053 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 054 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 055 | FF FF FF FF FF FF FF 07 80 69 00 00 00 00 00 00 | .........i......
[=] 056 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 057 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 058 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 059 | A0 A1 A2 A3 A4 A5 00 00 00 00 C9 5E 34 C0 A1 5E | ...........^4..^
[=] 060 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 061 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 062 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 063 | FF FF FF FF FF FF 04 00 46 8E 00 00 00 00 00 00 | ........F.......
[=] ----+-------------------------------------------------+-----------------

```

iceman1001, 2020-09-18:
pull latest, flash and test

Fl0-0, 2020-09-18:
```
     128 |    1763 | Brute force phase completed. Key found: c95e34c0a15e   |               0 |    0s
[+] target sector: 14 key type: B -- found valid key [ C9 5E 34 C0 A1 5E ]

[+] found keys:
[+] |-----|----------------|---|----------------|---|
[+] | Sec | key A          |res| key B          |res|
[+] |-----|----------------|---|----------------|---|
[+] | 000 | ffffffffffff   | D | 000000000000   | D |
[+] | 001 | ffffffffffff   | D | 000000000000   | D |
[+] | 002 | ffffffffffff   | D | 000000000000   | D |
[+] | 003 | ffffffffffff   | D | 000000000000   | D |
[+] | 004 | ffffffffffff   | D | 000000000000   | D |
[+] | 005 | ffffffffffff   | D | 000000000000   | D |
[+] | 006 | ffffffffffff   | D | 000000000000   | D |
[+] | 007 | ffffffffffff   | D | 000000000000   | D |
[+] | 008 | ffffffffffff   | D | 000000000000   | D |
[+] | 009 | ffffffffffff   | D | 000000000000   | D |
[+] | 010 | ffffffffffff   | D | 000000000000   | D |
[+] | 011 | ffffffffffff   | D | 000000000000   | D |
[+] | 012 | ffffffffffff   | D | 000000000000   | D |
[+] | 013 | ffffffffffff   | D | 000000000000   | D |
[+] | 014 | a0a1a2a3a4a5   | D | c95e34c0a15e   | H |
[+] | 015 | ffffffffffff   | D | 000000000000   | D |
[+] |-----|----------------|---|----------------|---|
[=] ( D:Dictionary / S:darkSide / U:User / R:Reused / N:Nested / H:Hardnested / C:statiCnested / A:keyA )

[+] Generating binary key file
[+] Found keys have been dumped to hf-mf-9D3312EA-key-10.bin--> 0xffffffffffff has been inserted for unknown keys.
[+] transferring keys to simulator memory (Cmd Error: 04 can occur)
[#] Cmd Error 04
[#] Error reading sector 14 block  0
[#] wrong response len 0 (expected 18)
[#] Error reading sector 14 block  1
[#] wrong response len 0 (expected 18)
[#] Error reading sector 14 block  2
[#] wrong response len 0 (expected 18)
[#] Error reading sector 14 block  3
[#] Sector[15]. Auth nested error
[#] Emulator fill sectors finished
[=] fast dump reported back failure w KEY A,  swapping to KEY B
[#] Cmd Error 04
[#] Error reading sector  0 block  0
[#] wrong response len 0 (expected 18)
[#] Error reading sector  0 block  1
[#] wrong response len 0 (expected 18)
[#] Error reading sector  0 block  2
[#] wrong response len 0 (expected 18)
[#] Error reading sector  0 block  3
[#] Sector[ 1]. Auth nested error
[#] Sector[ 2]. Auth nested error
[#] Sector[ 3]. Auth nested error
[#] Sector[ 4]. Auth nested error
[#] Sector[ 5]. Auth nested error
[#] Sector[ 6]. Auth nested error
[#] Sector[ 7]. Auth nested error
[#] Sector[ 8]. Auth nested error
[#] Sector[ 9]. Auth nested error
[#] Sector[10]. Auth nested error
[#] Sector[11]. Auth nested error
[#] Sector[12]. Auth nested error
[#] Sector[13]. Auth nested error
[#] Sector[14]. Auth nested error
[#] Sector[15]. Auth nested error
[#] Emulator fill sectors finished
[=] fast dump reported back failure w KEY B
[=] Dump file is PARTIAL complete
[=] downloading the card content from emulator memory
[+] saved 1024 bytes to binary file hf-mf-9D3312EA-dump-10.bin
[+] saved 64 blocks to text file hf-mf-9D3312EA-dump-10.eml
[... 115 lines cut ...]
[=] 041 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 042 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 043 | FF FF FF FF FF FF FF 07 80 69 00 00 00 00 00 00 | .........i......
[=] 044 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 045 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 046 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 047 | FF FF FF FF FF FF FF 07 80 69 00 00 00 00 00 00 | .........i......
[=] 048 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 049 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 050 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 051 | FF FF FF FF FF FF FF 07 80 69 00 00 00 00 00 00 | .........i......
[=] 052 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 053 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 054 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 055 | FF FF FF FF FF FF FF 07 80 69 00 00 00 00 00 00 | .........i......
[=] 056 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 057 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 058 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 059 | A0 A1 A2 A3 A4 A5 00 00 00 00 C9 5E 34 C0 A1 5E | ...........^4..^
[=] 060 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 061 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 062 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 063 | FF FF FF FF FF FF 04 00 46 8E 00 00 00 00 00 00 | ........F.......
[=] ----+-------------------------------------------------+-----------------
```

iceman1001, 2020-09-18:
Now its the nested auth,.

iceman1001, 2020-09-18:
If you comment out line 1977 and forward,   armsrc/mifarecmd.c   fct MifareECardLoad
Like this,   compile, flash, and test..

```
    for (uint8_t sectorNo = 0; sectorNo < sectorcnt; sectorNo++) {
        uint64_t ui64Key = emlGetKey(sectorNo, keytype);
        //if (sectorNo == 0) {
            if (mifare_classic_auth(pcs, cuid, FirstBlockOfSector(sectorNo), keytype, ui64Key, AUTH_FIRST)) {
                retval = PM3_EPARTIAL;
                if (DBGLEVEL > DBG_ERROR) Dbprintf("Sector[%2d]. Auth error", sectorNo);
                continue;
            }
        /*
        } else {
            if (mifare_classic_auth(pcs, cuid, FirstBlockOfSector(sectorNo), keytype, ui64Key, AUTH_NESTED)) {
                retval = PM3_EPARTIAL;
                if (DBGLEVEL > DBG_ERROR) Dbprintf("Sector[%2d]. Auth nested error", sectorNo);
                continue;
            }
        }
        */
```

Fl0-0, 2020-09-18:
```
[+] target sector: 14 key type: B -- found valid key [ C9 5E 34 C0 A1 5E ]

[+] found keys:
[+] |-----|----------------|---|----------------|---|
[+] | Sec | key A          |res| key B          |res|
[+] |-----|----------------|---|----------------|---|
[+] | 000 | ffffffffffff   | D | 000000000000   | D |
[+] | 001 | ffffffffffff   | D | 000000000000   | D |
[+] | 002 | ffffffffffff   | D | 000000000000   | D |
[+] | 003 | ffffffffffff   | D | 000000000000   | D |
[+] | 004 | ffffffffffff   | D | 000000000000   | D |
[+] | 005 | ffffffffffff   | D | 000000000000   | D |
[+] | 006 | ffffffffffff   | D | 000000000000   | D |
[+] | 007 | ffffffffffff   | D | 000000000000   | D |
[+] | 008 | ffffffffffff   | D | 000000000000   | D |
[+] | 009 | ffffffffffff   | D | 000000000000   | D |
[+] | 010 | ffffffffffff   | D | 000000000000   | D |
[+] | 011 | ffffffffffff   | D | 000000000000   | D |
[+] | 012 | ffffffffffff   | D | 000000000000   | D |
[+] | 013 | ffffffffffff   | D | 000000000000   | D |
[+] | 014 | a0a1a2a3a4a5   | D | c95e34c0a15e   | H |
[+] | 015 | ffffffffffff   | D | 000000000000   | D |
[+] |-----|----------------|---|----------------|---|
[=] ( D:Dictionary / S:darkSide / U:User / R:Reused / N:Nested / H:Hardnested / C:statiCnested / A:keyA )

[+] Generating binary key file
[+] Found keys have been dumped to hf-mf-9D3312EA-key-11.bin--> 0xffffffffffff has been inserted for unknown keys.
[+] transferring keys to simulator memory (Cmd Error: 04 can occur)
[#] Cmd Error 04
[#] Error reading sector  1 block  0
[#] wrong response len 0 (expected 18)
[#] Error reading sector  1 block  1
[#] wrong response len 0 (expected 18)
[#] Error reading sector  1 block  2
[#] wrong response len 0 (expected 18)
[#] Error reading sector  1 block  3
[#] wrong response len 0 (expected 18)
[#] Error reading sector  2 block  0
[#] wrong response len 0 (expected 18)
[#] Error reading sector  2 block  1
[#] wrong response len 0 (expected 18)
[#] Error reading sector  2 block  2
[#] wrong response len 0 (expected 18)
[#] Error reading sector  2 block  3
[#] wrong response len 0 (expected 18)
[#] Error reading sector  3 block  0
[#] wrong response len 0 (expected 18)
[#] Error reading sector  3 block  1
[#] wrong response len 0 (expected 18)
[#] Error reading sector  3 block  2
[#] wrong response len 0 (expected 18)
[#] Error reading sector  3 block  3
[#] wrong response len 0 (expected 18)
[#] Error reading sector  4 block  0
[#] wrong response len 0 (expected 18)
[#] Error reading sector  4 block  1
[#] wrong response len 0 (expected 18)
[#] Error reading sector  4 block  2
[#] wrong response len 0 (expected 18)
[#] Error reading sector  4 block  3
[#] wrong response len 0 (expected 18)
[#] Error reading sector  5 block  0
[#] wrong response len 0 (expected 18)
[#] Error reading sector  5 block  1
[#] wrong response len 0 (expected 18)
[#] Error reading sector  5 block  2
[#] wrong response len 0 (expected 18)
[#] Error reading sector  5 block  3
[#] wrong response len 0 (expected 18)
[... 194 lines cut ...]
[#] Error reading sector 14 block  0
[#] wrong response len 0 (expected 18)
[#] Error reading sector 14 block  1
[#] wrong response len 0 (expected 18)
[#] Error reading sector 14 block  2
[#] wrong response len 0 (expected 18)
[#] Error reading sector 14 block  3
[#] wrong response len 0 (expected 18)
[#] Error reading sector 15 block  0
[#] wrong response len 0 (expected 18)
[#] Error reading sector 15 block  1
[#] wrong response len 0 (expected 18)
[#] Error reading sector 15 block  2
[#] wrong response len 0 (expected 18)
[#] Error reading sector 15 block  3
[#] Emulator fill sectors finished
[=] fast dump reported back failure w KEY B
[=] Dump file is PARTIAL complete
[=] downloading the card content from emulator memory
[+] saved 1024 bytes to binary file hf-mf-9D3312EA-dump-11.bin
[+] saved 64 blocks to text file hf-mf-9D3312EA-dump-11.eml
[+] saved to json file hf-mf-9D3312EA-dump-11.json
[=] autopwn execution time: 41 seconds
[usb] pm3 --> 
```

Fl0-0, 2020-09-18:
It fails reading all sectors except 0,3 and 15 now.

```
[usb] pm3 --> hf mf ekeyprn
[+] |-----|----------------|---|----------------|---|
[+] | Sec | key A          |res| key B          |res|
[+] |-----|----------------|---|----------------|---|
[+] | 000 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 001 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 002 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 003 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 004 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 005 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 006 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 007 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 008 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 009 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 010 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 011 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 012 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 013 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] | 014 | a0a1a2a3a4a5   | 1 | c95e34c0a15e   | 1 |
[+] | 015 | ffffffffffff   | 1 | 000000000000   | 1 |
[+] |-----|----------------|---|----------------|---|
[+] ( 0:Failed / 1:Success)
[usb] pm3 --> hf mf ecfill A
--params: numSectors: 16, keyType: A

[usb] pm3 --> 
[#] Cmd Error 04
[#] Error reading sector  1 block  0
[#] wrong response len 0 (expected 18)
[#] Error reading sector  1 block  1
[#] wrong response len 0 (expected 18)
[#] Error reading sector  1 block  2
[#] wrong response len 0 (expected 18)
[#] Error reading sector  1 block  3
[#] wrong response len 0 (expected 18)
[#] Error reading sector  2 block  0
[#] wrong response len 0 (expected 18)
[#] Error reading sector  2 block  1
[#] wrong response len 0 (expected 18)
[#] Error reading sector  2 block  2
[#] wrong response len 0 (expected 18)
[#] Error reading sector  2 block  3
[#] wrong response len 0 (expected 18)
[#] Error reading sector  3 block  0
[#] wrong response len 0 (expected 18)
[#] Error reading sector  3 block  1
[#] wrong response len 0 (expected 18)
[#] Error reading sector  3 block  2
[#] wrong response len 0 (expected 18)
[#] Error reading sector  3 block  3
[#] wrong response len 0 (expected 18)
[#] Error reading sector  4 block  0
[#] wrong response len 0 (expected 18)
[#] Error reading sector  4 block  1
[#] wrong response len 0 (expected 18)
[#] Error reading sector  4 block  2
[#] wrong response len 0 (expected 18)
[#] Error reading sector  4 block  3
[#] wrong response len 0 (expected 18)
[#] Error reading sector  5 block  0
[#] wrong response len 0 (expected 18)
[#] Error reading sector  5 block  1
[#] wrong response len 0 (expected 18)
[#] Error reading sector  5 block  2
[#] wrong response len 0 (expected 18)
[#] Error reading sector  5 block  3
[#] wrong response len 0 (expected 18)
[... 263 lines cut ...]
[=] 042 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 043 | FF FF FF FF FF FF 00 00 00 00 00 00 00 00 00 00 | ................
[=] 044 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 045 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 046 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 047 | FF FF FF FF FF FF 00 00 00 00 00 00 00 00 00 00 | ................
[=] 048 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 049 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 050 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 051 | FF FF FF FF FF FF 00 00 00 00 00 00 00 00 00 00 | ................
[=] 052 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 053 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 054 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 055 | FF FF FF FF FF FF 00 00 00 00 00 00 00 00 00 00 | ................
[=] 056 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 057 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 058 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 059 | A0 A1 A2 A3 A4 A5 00 00 00 00 C9 5E 34 C0 A1 5E | ...........^4..^
[=] 060 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 061 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 062 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 | ................ 
[=] 063 | FF FF FF FF FF FF 04 00 46 8E 00 00 00 00 00 00 | ........F.......
[=] ----+-------------------------------------------------+-----------------

```

iceman1001, 2021-01-27:
There has been some fixes for hardnested,  try pulling latest and see if this issue still persist

Fl0-0, 2021-01-28:
Unfortunately, the issue still persist.

iceman1001, 2022-12-05:
Ping!
I believe we did some fixing for dump to also use key b,   have you tested the latest source?

Fl0-0, 2022-12-05:
Pong !

Still have the issue with `hf mf autopwn`:
```
[=] fast dump reported back failure w KEY B
[=] Dump file is PARTIAL complete
[=] downloading the card content from emulator memory
```

Works well with `hf mf dump` just after, with keys in memory.

iceman1001, 2022-12-05:
yeah,  so the dump command handles it with A/B,     now autopwn should do the say...

Fl0-0, 2023-07-25:
The issue is still there :(

iceman1001, 2023-07-25:
What is your output current run,  and a dump of the card you use to test?

Fl0-0, 2023-07-25:
The output is exactly the same https://github.com/RfidResearchGroup/proxmark3/issues/960#issuecomment-694783399 .

If you want the dump i can send it to you, just tell me where.

iceman1001, 2023-07-25:
you on discord?   mifare channel,   pm me,   or here.

iceman1001, 2023-07-25:
it is quite impossible you have the same output using the latest source,   "Auth nested error"  doesn't exist in the code anymore.

Fl0-0, 2023-07-27:
Issue fixed with latest.
Thanks @iceman1001 for the investigation !

closed 2023-07-27 by Fl0-0