RfidResearchGroup/proxmark3 #961: lf hitag reader needs to send its command much earlier

doegox, 2020-09-17:
`lf hitag reader 26` fails at talking to a HT2 if the HT2 is in emulation mode. We need to send the command much quicker after powering the field. 

Let's illustrate with some captures done with a second proxmark, to capture the command part as well

Here is a tag without emulation, read works fine:
[image]

But here is a tag configured to emulate a EM4100:
[image]

We see that it already started its emulation mode before the command is sent, and so the command fails.

Finally, here is an example of what should be done, here done manually with `lf cmdread d 50 z 116 o 166 e W 3000 c W11000`
[image]

doegox, 2020-09-18:
@iceman1001 with https://github.com/RfidResearchGroup/proxmark3/commit/8f2b9b9d66a74871f7d814c6b41443fe66629ccb
still too late, actually I see only very little difference:
[image]

iceman1001, 2020-09-18:
Hard to make out from your picture but how many samples before tag starts?

doegox, 2020-09-18:
Fixed 59976aa5881592c335bf125ba268f4b3718664f2

closed 2020-09-18 by doegox