RfidResearchGroup/proxmark3 #973: QT5 rabbithole: solve QWheelEvent version-related issues

doegox, 2020-09-22:
Tracking it in an issue because I feel it won't be solved in a snap...

QWheelEvent supports different methods depending on QT version.

Current mouse position:
* QT4: `x()`:OK `position().x()`:no such member
* QT5.?? (Appveyor WSL): `x()`:OK `position().x()`:no such member
* QT5.14.2 (my debian): `x()`:OK `position().x()`:OK
* QT5.15.1 (Travis OSX):  `x()`:deprecated `position().x()`:OK
* QT5.?? (Appveyor PS): `x()`:deprecated `position().x()`:OK

Wheel movement:
* QT4: `delta()`:OK `angleDelta().y()`:no such member
* QT5.?? (Appveyor WSL): `delta()`:status? `angleDelta().y()`: OK
* QT5.14.2 (my debian): `delta()`:OK `angleDelta().y()`:OK
* QT5.?? (Travis OSX):  `delta()`:deprecated `angleDelta()`:OK
* QT5.?? (Appveyor PS): `delta()`:deprecated `angleDelta()`:OK

doegox, 2020-09-22:
ok now that all versions have been collected thanks to https://github.com/RfidResearchGroup/proxmark3/commit/75dab301536cccfa4378c969867c8adc1ec806bb
a fix has been pushed https://github.com/RfidResearchGroup/proxmark3/commit/5322b0c4a18341dd2d17301ab45e12c7e0e398f7

closed 2020-09-22 by doegox

iceman1001, 2020-09-23:
This is a nice improvement @doegox   !   Well done,   making the plot window better