RfidResearchGroup/proxmark3 #986: How to use the proxmark3 client without GUI?

Arlen-LT, 2020-10-01:
While I input "proxmark3" in the command line, it prints the PM3 icon correctly but then print something as follows.
'''
qt.qpa.screen: QXcbConnection: Could not connect to display :0
Could not connect to any X display.
'''
Then the client quits. I am using the client on the WSL1 Debian on Win10.
It seems there must be a GUI. But I saw someone just using command "proxmark3" to start the client in command line. How can I run it in command line also?

Arlen-LT, 2020-10-01:
I think maybe if I install the X server like VcXsrv according to the indroduction for Win10, it could work correctly. But is there any method for using command lines instead of GUI?

iceman1001, 2020-10-01:
If you don't compile the project for a gui, ie  w/o QT support,  it should work with out a x-server.
However you don't get the plot window.  Like android termux solution.

But since you are on WSL1 why not just install the VxXsrv?   
You get a CLI regardless,   the Proxmark3 client is CLI based.

doegox, 2020-10-01:
Hmm interesting, normally the client works on pure console too but in your setup it seem to think there is a display available, then fails at connecting to it.
It brings back a suggestion I did some time ago: provide an option to force text mode...

Arlen-LT, 2020-10-01:
> If you don't compile the project for a gui, ie w/o QT support, it should work with out a x-server.
> However you don't get the plot window. Like android termux solution.
> 
> But since you are on WSL1 why not just install the VxXsrv?
> You get a CLI regardless, the Proxmark3 client is CLI based.

Because actually I use WSL1 for the first time today... my devices which have other systems now can't be used for some reasons. That means I now just can use this desktop computer whose system is win10.
So, if I make clean and then `make SKIPQT=1`, will it be able to work in the text mode?

doegox, 2020-10-01:
@Arlen-LT do you have a DISPLAY environment variable defined in WSL?
try `echo $DISPLAY`
if the client sees such variable it will try to connect to the graphics.

Otherwise yes, compiling with SKIPQT=1 will help you.

Arlen-LT, 2020-10-01:
> @Arlen-LT do you have a DISPLAY environment variable defined in WSL?
> try `echo $DISPLAY`
> if the client sees such variable it will try to connect to the graphics.
> 
> Otherwise yes, compiling with SKIPQT=1 will help you.

Thank you very much. `make SKIPQT=1` helped me.

closed 2020-10-01 by Arlen-LT

UltrashRicco, 2021-01-06:
> > @Arlen-LT do you have a DISPLAY environment variable defined in WSL?
> > try `echo $DISPLAY`
> > if the client sees such variable it will try to connect to the graphics.
> > Otherwise yes, compiling with SKIPQT=1 will help you.
> 
> Thank you very much. `make SKIPQT=1` helped me.

Thank you all and @Arlen-LT this also is necessary if you intend to use the client over SSH, for example if you have the client installed on a headless Raspberry Pi that you wish to control over SSH.