RfidResearchGroup/proxmark3 #987: RDV4 Bricked & JTAG pins are not detected

Darkleer, 2020-10-02:
Hi,

I bricked my RDV4 attempting to flash the bootloader. The "button trick" doesn't work. 
I tried to flash it through JTAG pins using Bus Pirate or JLINK edu. I followed the procedure, however it didn't worked. My RDV4 was not detected. So I tried to test my JTAG pins and to determine then thanks to JTAGulator. Finally, JTAGulator detected nothing.
Is it possible that the microprocessor needs to be erased (as described here, but on RDV3 : https://github.com/Proxmark/proxmark3/wiki/De-Bricking-Segger) in order to get flashed via JTAG. On RDV4, the ATMEL microprocessor is under the smart card reader. We can't reach it.

I hope some of you already manage this kind of problems and could help me.

iceman1001, 2020-10-02:
Have you tried to follow the instructions in https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/jtag_notes.md ?
You might have gotten the pins wrong.

Darkleer, 2020-10-05:
I did not try to execute "openocd_flash_recovery.sh" because I probably have a bootloader firmware problem.
I also though that I might have gotten the pins wrong. After multiple tries, I wanted to verify the JTAG connection with JTAGulator. JTAGulator is able to determine the JTAG pins correspondence.

iceman1001, 2020-10-05:
Just slowly pull it off,  its held down by tape and has a connector which is easy to detach and reattach.

Darkleer, 2020-10-05:
Thanks ! I removed the smartcard reader. Then, I have been able to unlock the JTAG pins. Now, the JTAG is detected by my JTAGulator. However, I still cannot connect to the proxmark via the JLink edu or Bus pirate.
For the Bus Pirate, I got the follwing Error "Buspirate did not answer correctly! Do you have correct firmware?". I have the v7.0 firmware on my BP.
My JLink edu cannot connect.
I hope this is due to my wiring because the JTAG holes/pins are small.

iceman1001, 2020-10-05:
I think that is out of the scope of the repo's purpose.  I will close this issue since its not relevant to the source code.
Feel free to join the Proxmark3 forum and/or discord server to ask questions.   Read the relevant documents available here in this repo and in the tools/jtag_openocd  scripts to find information also.

closed 2020-10-05 by iceman1001