RfidResearchGroup/proxmark3 #994: Legic.lua script crashes

dose13seitz, 2020-10-04:
I have tried to write a segment on a "wiped" legic prime card using the "as" command. The script crashes with the following message:

> Legic command? ('h' for help - 'q' for quit) (default: h ) 
 > as
/usr/local/bin/../share/proxmark3/luascripts/legic.lua:2468: attempt to index upvalue 'inTAG' (a nil value)

> [+] Finished legic.lua
          
> [usb] pm3 --> 

Thank's for looking into this,

uhei, 2020-10-05:
I'm not able to reproduce your issue with latest master.

Which version are you using? If not master (I guess so because script has been renamed to hf_legic.lua) can you give it a try with latest master?

Do you use read tag (rt) or load file (lf) for the input Tag?

What is the output of dump segments (ds) before calling as?

dose13seitz, 2020-10-06:
Hi and thank's for your reply! 

I just pulled the latest version of RDV4:

hw version:
-------------------------------------------------------------------------
 [ CLIENT ]          
  client: RRG/Iceman          
  compiled with GCC 7.4.0 OS:Linux ARCH:x86_64          

 [ PROXMARK3 RDV4 ]          
  external flash:                  present           
  smartcard reader:                present           

 [ PROXMARK3 RDV4 Extras ]          
  FPC USART for BT add-on support: absent           
          
 [ ARM ]
  bootrom: RRG/Iceman/master/30ea9103 2019-11-05 20:02:25
       os: RRG/Iceman/master/30ea9103 2019-11-05 20:02:45
  compiled with GCC 6.3.1 20170620

 [ FPGA ]
  LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
  HF image built for 2s30vq100 on 2018-09-03 at 21:40:23          

 < [ Hardware ]        
  = uC: AT91SAM7S512 Rev B          
  = Embedded Processor: ARM7TDMI          
  = Nonvolatile Program Memory Size: 512K bytes, Used: 276338 bytes (53%) Free: 247950 bytes (47%)          
  = Second Nonvolatile Program Memory Size: None          
  = Internal SRAM Size: 64K bytes          
  = Architecture Identifier: AT91SAM7Sxx Series          
  = Nonvolatile Program Memory Type: Embedded Flash Memory**
>
-------------------------------------------------------------------------

According to this version the script starts with the command "script run legic.lua" I did not find the hf_legic.lua. Am I missing something? The script itself says Version 1.0.3.

What I tried is to add a new Segment to a complete wiped Card. I just tried to add a new Segment using the "as" command. I did not us rt, lf or ds before. The problem is that I do not have the original data of the Card and I want to write data to it, using different Segments. Hope that is possible. 

Let me know in case of using the script not properly or if I should do further tests. 

Thank's

iceman1001, 2020-10-06:
Did you use the release v4.9237  or latest master?
I don't see the correct version number and your fpga images are a bit old.
Where did you pull from?

uhei, 2020-10-06:
> I just pulled the latest version of RDV4:

> According to this version the script starts with the command "script run legic.lua" I did not find the hf_legic.lua. Am I missing something? The script itself says Version 1.0.3.

OK. You're definitely not using the latest master. Current version number is 1.0.4 and script is named hf_legic.lua

> What I tried is to add a new Segment to a complete wiped Card. I just tried to add a new Segment using the "as" command. I did not us rt, lf or ds before. The problem is that I do not have the [...]

It sounds you're using it wrong. Most commands edit or modify a virtual tag. Running just `as` without reading in a tag (from a real tag or a file) gives the error you mention.

To start read in your "wiped" card with `rt`. Afterwards you can use `di` to dump the tag or run `as` to add a new segment.

iceman1001, 2020-10-06:
We should have a message telling user to do it correct... not  just crashing :)

dose13seitz, 2020-10-06:
On the one hand I am a bit embarrassed on the other hand I am glad that I found out that I obviously was on a different fork. I finally managed to get the right repository:

-----------------------------------------------------------------------------------------------
[ CLIENT ]
  client: RRG/Iceman/master/v4.9237-1572-g491adacb 2020-10-06 21:27:00
  compiled with GCC 7.5.0 OS:Linux ARCH:x86_64

 [ PROXMARK3 ]
  firmware.........................PM3RDV4
  external flash...................present
  smartcard reader.................present
  FPC USART for BT add-on..........absent

[ ARM ]
bootrom: RRG/Iceman/master/30ea9103 2019-11-05 20:02:25
os: RRG/Iceman/master/v4.9237-1572-g491adacb 2020-10-06 21:27:58
compiled with GCC 6.3.1 20170620_

-----------------------------------------------------------------------------------------------

I also learned that I am obviously using the script not in the intended way. :-( Therefore I think this "issue" can be closed. Anyway, thank's for your help so far! Probably one of you guys can tell me how to create a Segment on my wiped legic card using the hf legic write command. I started the topic here: http://www.proxmark.org/forum/viewtopic.php?id=8162

closed 2020-10-06 by dose13seitz