RfidResearchGroup/proxmark3 #996: Compilation Error Due To src/proxguiqt.cpp

NathanielJS1541, 2020-10-04:
**Describe the bug**
Compilation fails, giving the following error with `make VERBOSE=1`:
```cpp
[-] CXX src/proxguiqt.cpp
src/proxguiqt.cpp: In member function ‘void Plot::PlotGraph(int*, size_t, QRect, QRect, QPainter*, int)’:
src/proxguiqt.cpp:488:42: error: expected ‘)’ before ‘PRId64’
     sprintf(str, "max=%d  min=%d  mean=%" PRId64 "  n=%u/%zu  CursorAVal=[%d]  CursorBVal=[%d]",
            ~                             ^~~~~~~
                                          )
src/proxguiqt.cpp:488:18: error: spurious trailing ‘%’ in format [-Werror=format=]
     sprintf(str, "max=%d  min=%d  mean=%" PRId64 "  n=%u/%zu  CursorAVal=[%d]  CursorBVal=[%d]",
                  ^~~~~~~~~~~~~~~~~~~~~~~~
src/proxguiqt.cpp:488:18: error: too many arguments for format [-Werror=format-extra-args]
```

**To Reproduce**
Steps to reproduce the behavior:
1. `git clone git clone https://github.com/RfidResearchGroup/proxmark3.git`
2. `cd proxmark3`
3. `make clean && make all`
4. Compilation error at src/proxguiqt.cpp

**Desktop (please complete the following information):**
 - OS: Windows 10 Home Version 10.0.19041 Build 19041
 - Client platform: MINGW64_NT-10.0-19041-WOW64
 - compiler version: gcc.exe (Rev1, Built by MSYS2 project) 10.2.0

doegox, 2020-10-04:
@NathanielJS1541 does it help if you add `#define __STDC_FORMAT_MACROS` at the beginning of client/src/proxguiqt.cpp ?

doegox, 2020-10-04:
Or rather `#include <inttypes.h>`

NathanielJS1541, 2020-10-04:
@doegox Including `<inttypes.h>` fixes the issue - thanks for that.
`#define __STDC_FORMAT_MACRO` did not resolve the issue, and is not needed when `<inttypes.h>`  is included.

Thanks again for your time.

doegox, 2020-10-04:
ok thanks for the feedback, I'll merge the test commit

doegox, 2020-10-04:
done

closed 2020-10-04 by doegox