RfidResearchGroup/proxmark3 #1004: The "pac id tag" bug for "lf sea"

xianglin1998, 2020-10-12:
Hi guys,  recently, I found a bug when I research "PAC ID card". When use "lf sea" to search PAC ID card, arm will become slow and client will time out. After careful search, I found that the problem appeared in armsrc /em4x05.c, "find_ double_listen_windows" function takes up a lot of time to process tasks, resulting in client timeout. If I add a delay, the problem can be solved. I don't know the cause of this problem, but I happened to solve it. Can you explain it for me? By the way, fix the bug in the next commit?

pic1 (if place pac id card on rdv4):
 
[image]

pic2 (if add delay, please ignore my debugging information):

[image]
[image]

👍

xianglin1998, 2020-10-12:
We can see that if add a delay, there will be no timeout bug.

iceman1001, 2020-10-12:
The timeout in `lf search` might go away, but your suggestion breaks the lf em 4x50  commands.  We need a fix that works without breaking

xianglin1998, 2020-10-12:
Yes, we can find a better solution. It's just one of the solutions I found. Maybe what's the connection between the new solution and the add delay?

iceman1001, 2020-10-12:
maybe a longer delay in the wait call

iceman1001, 2020-12-09:
I think with the latest merge,  this bug is solved.   Would you mind verifying it?

tharexde, 2020-12-09:
I'll have look at it.

tharexde, 2020-12-09:
As iceman pointed out: SpinDelay is no solution. It interferes with the timing and an EM4x50 tag can not be read any more. 
But the function _find_double_listen_window_ should not be called at all as long as there’s no EM4x50 tag on PM3. There can be some strange sample values at the beginning of sampling, which are responsible for this behavior. 
This bug has been fixed with the latest EM4x50 PR.

iceman1001, 2020-12-09:
Thanks for verifying!

closing this

closed 2020-12-09 by iceman1001