RfidResearchGroup/proxmark3 #1006: hw tune weird values

nocomp, 2020-10-12:
hi folks,
i run a pm3 rdv4 , flashedit with lass firmware, no issues what so ever.
when i see other ppl hw tune data in other post, i can see that mine are totally exotic:
[usb] pm3 --> hw tune
[=] Measuring antenna characteristics, please wait...
 🕛   9
[=] ---------- LF Antenna ----------
[+] LF antenna: 4294845.87 V - 125.00 kHz
[+] LF antenna: 4294845.87 V - 134.83 kHz
[+] LF optimal: 4294845.87 V -  46.88 kHz
[+] Approx. Q factor (*): 749212.0 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna:  9.65 V - 13.56 MHz
[+] Approx. Q factor (*): 1.7 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.

[usb] pm3 --> 

any idea what can be wrong?

thx for your time

iceman1001, 2020-10-12:
Please add your output from
```
hw status
hw version
```
And if you got some compilation errors when you compiled and flashed fullimage.

nocomp, 2020-10-12:
hi @iceman1001 
thxx a lot for yoru reactivity.
here we go
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size.............42124
[#]   Available memory........39816
[#] Tracing
[#]   tracing ................0
[#]   traceLen ...............39812
[#]   dma8 memory.............-2112044
[#]   dma16 memory............-2112044
[#]   toSend memory...........39816
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
[#] Flash memory
[#]   Baudrate................24 MHz
[#]   Init....................OK
[#]   Memory size.............2 mbits / 256 kb
[#]   Unique ID...............0xD567A882A74AA925
[#] Smart card module (ISO 7816)
[#]   version.................v3.10
[#] LF Sampling config
[#]   [q] divisor.............95 ( 125.00 kHz )
[#]   [b] bits per sample.....8
[#]   [d] decimation..........1
[#]   [a] averaging...........Yes
[#]   [t] trigger threshold...0
[#]   [s] samples to skip.....0 
[#] LF Sampling Stack
[#]   Max stack usage.........4016 / 8480 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#]               leading zero |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#]    1 of 4 coding reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override....0 ( No ) follow standard 
[#]   [b] BCC override........0 ( No ) follow standard
[#]   [2] CL2 override........0 ( No ) follow standard
[#]   [3] CL3 override........0 ( No ) follow standard
[#]   [r] RATS override.......0 ( No q follow standard 
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed............500ms
[#]   Bytes transferred.......283648
[#]   Transfer Speed PM3 -> Client = 567296 bytes/s
[#] Various
[#]   Max stack usage.........4096 / 8480 bytes
[#]   DBGLEVEL................1 ( ERROR )
[#]   ToSendMax...............10
[#]   ToSend BUFFERSIZE.......2308
[#]   Slow clock..............31956 Hz
[#] Installed StandAlone Mode
[#]   HF - Reading Visa cards & Emulating a Visa MSD Transaction(ISO14443) - (Salvador Mendoza)
[#] Flash memory dictionary loaded

[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-1690-gdcc33d59 2020-10-12 08:46:24
  compiled with GCC 7.5.0 OS:Linux ARCH:x86_64

 [ PROXMARK3 ]
  firmware.........................PM3RDV4
  external flash...................present
  smartcard reader.................present
  FPC USART for BT add-on..........absent

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-1690-gdcc33d59 2020-10-12 08:46:47
       os: RRG/Iceman/master/v4.9237-1690-gdcc33d59 2020-10-12 08:46:59
  compiled with GCC 6.3.1 20170620

 [ FPGA ] 
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 295848 bytes (56%) Free: 228440 bytes (44%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

[usb] pm3 --> 

thxx a lot

nocomp, 2020-10-12:
hi,
just noticed i can t get the tag info, as i get from classic pm3 device. getting HID H10301 instad of something like 1cca045555
[usb] pm3 -->  lf hid read 1
[+] [H10301] - HID H10301 26-bit;  FC: 74  CN: 3880    parity: valid
[usb] pm3 -->

iceman1001, 2020-10-12:
yes,  this repo has different output for that command than the official repo.

nocomp, 2020-10-12:
hi,
so my device is not faulty, just normal behavoiur?
thank you for your time

nocomp, 2020-10-12:
what should i do in order to get this working with this repo ?
https://www.tinker.sh/badge-cloning-clone-hid-prox-with-proxmark3-rvd4/

do i need another repo for that?
best regards

iceman1001, 2020-10-12:
This is not an issue with the fw,  it seems you have more of a user issue.  Ask questions on forum  or on discord.

Regarding standalone modes, I suggest you start reading the docs,  
ref:
https://github.com/RfidResearchGroup/proxmark3/wiki/Standalone-mode
https://github.com/RfidResearchGroup/proxmark3/blob/master/armsrc/Standalone/readme.md

nocomp, 2020-10-12:
Thxx , already had a look, teying to figure out why it used to work and now since i ve flashed last version, doesnt work any more.
Am i using wrong repo? Doesnt seems to :/
Thx for your help

iceman1001, 2020-10-12:
I can tell you haven't followed instructions on for your RDV4,   not that it will change your current problems with standalone mode,  but it will weed out other potential issues.

https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/2_Configuration-and-Verification.md

closed 2020-10-12 by iceman1001