RfidResearchGroup/proxmark3 #1017: motorola clone command

will-caruana, 2020-10-17:
**Describe the bug**
Not able to write the last byte using the motorola clone command. I am able to write a0000000a00020 but not a0000000a0002021 the 21 at the end gives me an error.

**To Reproduce**
lf motorola clone a0000000a0002021

**Expected behavior**
This would write all charters to the card.

**Screenshots**
[usb] pm3 -->     lf motorola clone a0000000a0002021
Parameter error: string too long, expect max 15 chars
Parameter error: buffer overflow.

**Desktop (please complete the following information):**
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-1767-gb6fd1a86 2020-10-17 09:50:37
  compiled with GCC 9.3.0 OS:Linux ARCH:x86_64

 [ PROXMARK3 ]
  firmware.........................PM3RDV4
  external flash...................present
  smartcard reader.................present
  FPC USART for BT add-on..........present

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-1767-gb6fd1a86 2020-10-17 09:50:58
       os: RRG/Iceman/master/v4.9237-1767-gb6fd1a86 2020-10-17 09:51:10
  compiled with GCC 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 [ FPGA ] 
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 295808 bytes (56%) Free: 228480 bytes (44%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

iceman1001, 2020-10-17:
Pull latest,  should be fixed

will-caruana, 2020-10-17:
That worked thanks!

closed 2020-10-17 by will-caruana