RfidResearchGroup/proxmark3 #1020: iClass dumping gives falsely reports incompatible parameters

guberti, 2020-10-17:
**Describe the bug**
Whenever the `e`, `r`, or `n` flags are used with `hf iclass dump`, the software fails with error `"Can not use a combo of 'e', 'r', 'n'"`

**To Reproduce**
Running `hf iclass dump h` shows the help, which gives the following sample commands:
```
	hf iclass dump k 001122334455667B
	hf iclass dump k AAAAAAAAAAAAAAAA c 001122334455667B
	hf iclass dump k AAAAAAAAAAAAAAAA e
	hf iclass dump k 0
```

However, running `hf iclass dump k AAAAAAAAAAAAAAAA e` gives the output:

```
[usb] pm3 --> hf iclass dump k AAAAAAAAAAAAAAAA e
[+] Using elite algo
[-] ⛔ Can not use a combo of 'e', 'r', 'n'
Dump all memory from a iCLASS tag
...
```
and subsequently fails. The same happens when the `r` and `n` tags are used.

<details>
  <summary>Environment information</summary>
  
OS: Ubuntu 20.04
`hw version`
```
 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-1772-gabaef4de 2020-10-17 13:44:15
  compiled with GCC 9.3.0 OS:Linux ARCH:x86_64

 [ PROXMARK3 ]
  firmware.........................PM3RDV4
  external flash...................present
  smartcard reader.................present
  FPC USART for BT add-on..........absent

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-1772-gabaef4de 2020-10-17 13:44:37
       os: RRG/Iceman/master/v4.9237-1772-gabaef4de 2020-10-17 13:44:49
  compiled with GCC 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

 [ FPGA ] 
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 294704 bytes (56%) Free: 229584 bytes (44%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
```
`hw status`
```
[#] Memory
[#]   BigBuf_size.............42124
[#]   Available memory........37256
[#] Tracing
[#]   tracing ................0
[#]   traceLen ...............402
[#]   dma8 memory.............-2112044
[#]   dma16 memory............37256
[#]   toSend memory...........37768
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
[#] Flash memory
[#]   Baudrate................24 MHz
[#]   Init....................OK
[#]   Memory size.............2 mbits / 256 kb
[#]   Unique ID...............0xD5690C23DF26812A
[#] Smart card module (ISO 7816)
[#]   version.................v3.11
[#] LF Sampling config
[#]   [q] divisor.............95 ( 125.00 kHz )
[#]   [b] bits per sample.....8
[#]   [d] decimation..........1
[#]   [a] averaging...........Yes
[#]   [t] trigger threshold...0
[#]   [s] samples to skip.....0 
[#] LF Sampling Stack
[#]   Max stack usage.........4088 / 8480 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |  29 |  17 |  18 |  50 |  15 | N/A | N/A | 
[#]               leading zero |  29 |  17 |  18 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override....0 ( No ) follow standard 
[#]   [b] BCC override........0 ( No ) follow standard
[#]   [2] CL2 override........0 ( No ) follow standard
[#]   [3] CL3 override........0 ( No ) follow standard
[#]   [r] RATS override.......0 ( No q follow standard 
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed............500ms
[#]   Bytes transferred.......280064
[#]   Transfer Speed PM3 -> Client = 560128 bytes/s
[#] Various
[#]   Max stack usage.........4088 / 8480 bytes
[#]   DBGLEVEL................1 ( ERROR )
[#]   ToSendMax...............38
[#]   ToSend BUFFERSIZE.......2308
[#]   Slow clock..............31836 Hz
[#] Installed StandAlone Mode
[#]   HF - Reading Visa cards & Emulating a Visa MSD Transaction(ISO14443) - (Salvador Mendoza)
[#] Flash memory dictionary loaded
[#]   Mifare..................962 keys
[#]   T55x7...................112 keys
[#]   iClass..................7 keys
```
`data tune`
```
[=] ---------- LF Antenna ----------
[+] LF antenna: 75.51 V - 125.00 kHz
[+] LF antenna: 35.86 V - 134.83 kHz
[+] LF optimal: 75.51 V - 125.00 kHz
[+] Approx. Q factor (*): 12.4 by frequency bandwidth measurement
[+] Approx. Q factor (*): 13.2 by peak voltage measurement
[+] LF antenna is OK
[=] ---------- HF Antenna ----------
[+] HF antenna: 42.95 V - 13.56 MHz
[+] Approx. Q factor (*): 7.5 by peak voltage measurement
[+] HF antenna is OK

(*) Q factor must be measured without tag on the antenna

[+] Displaying LF tuning graph. Divisor 88 (blue) is 134.83 kHz, 95 (red) is 125.00 kHz.
```
</details>

iceman1001, 2020-10-17:
Have you pulled latest?  I thought I fixed in https://github.com/RfidResearchGroup/proxmark3/commit/8c7f1e0609c436c1aeb54d2f041eced2ccebef08

guberti, 2020-10-17:
You're right, I was running `abaef4d`. Fixed!

closed 2020-10-17 by guberti