RfidResearchGroup/proxmark3 #1024: smart card select failed

blackndoor, 2020-10-20:
Hello, I'm unable to use the smartcard reader.

```
[usb] pm3 --> smart info
[!] ⚠️  smart card select failed
```

It's a brand new proxmark3 RDV4.01. I followed the instructions to install the tools and Flash the BOOTROM & FULLIMAGE.

```
$ pm3-flash-all                                                 
[=] Session log /home/bkndr/.proxmark3/logs/log_20201020.txt
[+] loaded from JSON file /home/bkndr/.proxmark3/preferences.json
[+] About to use the following files:
[+]    /usr/local/bin/../share/proxmark3/firmware/bootrom.elf
[+]    /usr/local/bin/../share/proxmark3/firmware/fullimage.elf
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0
 🕑  59 found
[+] Entering bootloader...
[+] (Press and release the button only to abort)
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0
 🕒  48 found
[=] Available memory on this board: 512K bytes

[=] Permitted flash range: 0x00100000-0x00180000
[+] Loading ELF file /usr/local/bin/../share/proxmark3/firmware/bootrom.elf
[+] Loading usable ELF segments:
[+]    0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
[+]    1: V 0x00200000 P 0x00100200 (0x00000d40->0x00000d40) [R X] @0x298

[+] Loading ELF file /usr/local/bin/../share/proxmark3/firmware/fullimage.elf
[+] Loading usable ELF segments:
[+]    0: V 0x00102000 P 0x00102000 (0x00046730->0x00046730) [R X] @0x94
[+]    1: V 0x00200000 P 0x00148730 (0x00001804->0x00001804) [RW ] @0x467c4
[=] Note: Extending previous segment from 0x46730 to 0x47f34 bytes

[+] Flashing...
[+] Writing segments for file: /usr/local/bin/../share/proxmark3/firmware/bootrom.elf
[+]  0x00100000..0x001001ff [0x200 / 1 blocks]
. OK
[+]  0x00100200..0x00100f3f [0xd40 / 7 blocks]
....... OK

[+] Writing segments for file: /usr/local/bin/../share/proxmark3/firmware/fullimage.elf
[+]  0x00102000..0x00149f33 [0x47f34 / 576 blocks]
...................................................................
        @@@  @@@@@@@ @@@@@@@@ @@@@@@@@@@   @@@@@@  @@@  @@@
        @@! !@@      @@!      @@! @@! @@! @@!  @@@ @@!@!@@@
        !!@ !@!      @!!!:!   @!! !!@ @!@ @!@!@!@! @!@@!!@!
        !!: :!!      !!:      !!:     !!: !!:  !!! !!:  !!!
        :    :: :: : : :: :::  :      :    :   : : ::    : 
        .    .. .. . . .. ...  .      .    .   . . ..    . 
...................................................................
....................................................................... OK

[+] All done

Have a nice day!

```

Here is some other information that you might need:

```
$ pm3                                                           
[=] Session log /home/bkndr/.proxmark3/logs/log_20201020.txt
[+] loaded from JSON file /home/bkndr/.proxmark3/preferences.json
[=] Using UART port /dev/ttyACM0
[=] Communicating with PM3 over USB-CDC

  ██████╗ ███╗   ███╗█████╗ 
  ██╔══██╗████╗ ████║╚═══██╗
  ██████╔╝██╔████╔██║ ████╔╝
  ██╔═══╝ ██║╚██╔╝██║ ╚══██╗
  ██║     ██║ ╚═╝ ██║█████╔╝       Iceman ☕
  ╚═╝     ╚═╝     ╚═╝╚════╝    ❄️ bleeding edge

  https://github.com/rfidresearchgroup/proxmark3/

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman/master/v4.9237-1810-g3959b77f 2020-10-20 16:44:40
  compiled with GCC 7.5.0 OS:Linux ARCH:x86_64

 [ PROXMARK3 ]
  firmware.................. PM3RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... absent

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-1810-g3959b77f 2020-10-20 16:45:05
       os: RRG/Iceman/master/v4.9237-1810-g3959b77f 2020-10-20 16:45:19
  compiled with GCC 6.3.1 20170620

 [ FPGA ] 
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 296752 bytes (57%) Free: 227536 bytes (43%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
[usb] pm3 --> hw status
[#] Memory
[#]   BigBuf_size.............42124
[#]   Available memory........42124
[#] Tracing
[#]   tracing ................1
[#]   traceLen ...............0
[#]   dma8 memory.............-2112044
[#]   dma16 memory............-2112044
[#]   toSend memory...........-2112044
[#] Current FPGA image
[#]   mode.................... HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
[#] Flash memory
[#]   Baudrate................24 MHz
[#]   Init....................OK
[#]   Memory size.............2 mbits / 256 kb
[#]   Unique ID...............0xD5697C30975DD225
[#] Smart card module (ISO 7816)
[#]   version.................v3.11
[#] LF Sampling config
[#]   [q] divisor.............95 ( 125.00 kHz )
[#]   [b] bits per sample.....8
[#]   [d] decimation..........1
[#]   [a] averaging...........Yes
[#]   [t] trigger threshold...0
[#]   [s] samples to skip.....0 
[#] LF Sampling Stack
[#]   Max stack usage.........3952 / 8480 bytes
[#] LF T55XX config
[#]            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]
[#]            mode            |start|write|write|write| read|write|write
[#]                            | gap | gap |  0  |  1  | gap |  2  |  3
[#] ---------------------------+-----+-----+-----+-----+-----+-----+------
[#] fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A | 
[#]     long leading reference |  29 |  17 |  18 |  50 |  15 | N/A | N/A | 
[#]               leading zero |  29 |  17 |  18 |  40 |  15 | N/A | N/A | 
[#]    1 of 4 coding reference |N/A | N/A | N/A | N/A | N/A | N/A | N/A | 
[#] 
[#] HF 14a config
[#]   [a] Anticol override....0 ( No ) follow standard 
[#]   [b] BCC override........0 ( No ) follow standard
[#]   [2] CL2 override........0 ( No ) follow standard
[#]   [3] CL3 override........0 ( No ) follow standard
[#]   [r] RATS override.......0 ( No q follow standard 
[#] Transfer Speed
[#]   Sending packets to client...
[#]   Time elapsed............500ms
[#]   Bytes transferred.......276992
[#]   Transfer Speed PM3 -> Client = 553984 bytes/s
[#] Various
[#]   Max stack usage.........4096 / 8480 bytes
[#]   DBGLEVEL................1 ( ERROR )
[#]   ToSendMax...............-1
[#]   ToSend BUFFERSIZE.......2308
[#]   Slow clock..............30570 Hz
[#] Installed StandAlone Mode
[#]   HF - Reading Visa cards & Emulating a Visa MSD Transaction(ISO14443) - (Salvador Mendoza)
[#] Flash memory dictionary loaded
[#]   Mifare..................962 keys
[#]   T55x7...................112 keys
[#]   iClass..................7 keys
```

iceman1001, 2020-10-20:
Confirmed, I have noticed its not like this all time,  but is sure isn't stable.

tcprst, 2020-10-20:
Do you have any details of the cards you're trying @blackndoor?  I've tried several JCOP with contact interface, PIV, etc... and they all respond to the `smart info`

blackndoor, 2020-10-20:
I tried with a french VISA credit card, I will try with other ones. thks

tcprst, 2020-10-20:
Something else you could try is the `emv search` with the `-w` argument for using the contact interface.  I just tried with a German VISA and it finds the Visa Credit AID.   Although that same Visa does show up properly in `smart info` as well...

iceman1001, 2020-10-20:
lets see, I reverted to the high precision clock that was used before...
Not that it should matter,..
pull latest / compile / flash and test

blackndoor, 2020-10-20:
thks @tcprst, I tried with a health care card and got some info with `smart info`, also some info with `emv search -w` with the VISA.
@iceman1001 thks: `git pull, make clean && make all, sudo make install, pm3-flash-all`
I have the same result `smart card select failed`

iceman1001, 2020-10-20:
The smart card reader will have an issue with cards that uses 5v or more...
If I remember correct its only capable of 3.3v cards.    Try a different smart card in order to rid that out

tcprst, 2020-10-20:
> If I remember correct its only capable of 3.3v cards. 

Thanks @iceman1001 that explains what I'm seeing with my SLE4442 cards which are definitely 5V

iceman1001, 2020-10-24:
Pull latest and it should work again.
I missed one part to revert.

closed 2020-10-24 by iceman1001