RfidResearchGroup/proxmark3 #1026: LTO dump not working

will-caruana, 2020-10-22:
Describe the bug
lto dump keeps giving me a "Wrong response length" error.

To Reproduce
Use an LTO 1 Cartidge 
hf lto dump

Expected behavior
The card would dump

Screenshots
[usb] pm3 --> hf lto info

[+] UID: 6CC0370B90
[+] TYPE INFO: 0001
[usb] pm3 --> hf lto dump
[!] ⚠  Wrong response length (18 != 0)
[usb] pm3 --> hf lto list
[=] downloading tracelog data from device                                                                                                                                                                                                  
[+] Recorded activity (trace len = 8269 bytes)
[=] start = start of start frame end = end of frame. src = source of transfer
[=] ISO14443A - all times are in carrier periods (1/13.56MHz)

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |45                                                                       |     | REQ Standard
       2100 |       4532 | Tag |00  01                                                                   |     | 
     299520 |     301984 | Rdr |93  20                                                                   |     | SELECT
     303028 |     308852 | Tag |6c  c0  37  0b  90                                                       |     | 
     610304 |     620768 | Rdr |93  70  6c  c0  37  0b  90  4c  4e                                       |  ok | SELECT_UID-2
     621876 |     622452 | Tag |0a                                                                       |     | 
     918016 |     922784 | Rdr |30  00  02  a8                                                           |  ok | READBLOCK(0)
     923828 |     944692 | Tag |6c  c0  37  0b  90  04  00  01  01  00  00  00  00  00  35  00  30  89   |  ok | 
    1241216 |    1242528 | Rdr |80                                                                       |     | READBLOCK CONT
    1243572 |    1264436 | Tag |43  49  30  35  31  34  32  33  30  37  31  30  31  31  00  35  be  c7   |  ok | 
    1562880 |    1567648 | Rdr |30  01  8b  b9                                                           |  ok | READBLOCK(1)
    1568692 |    1589492 | Tag |05  00  00  00  00  01  00  40  00  02  00  80  0f  fe  00  c0  83  09   |  ok | 
    1889408 |    1890720 | Rdr |80                                                                       |     | READBLOCK CONT
    1891764 |    1912628 | Tag |0f  fe  00  c0  0f  fe  00  c0  0f  ff  00  c0  0a  61  ff  5b  23  67   |  ok | 
    2215552 |    2220256 | Rdr |30  02  10  8b                                                           |  ok | READBLOCK(2)
    2221364 |    2242164 | Tag |00  01  00  40  48  50  20  20  20  20  20  20  31  32  30  30  f0  0c   |  ok | 
    2543488 |    2544800 | Rdr |80                                                                       |     | READBLOCK CONT
    2545844 |    2566708 | Tag |38  31  33  38  35  34  00  01  32  30  30  38  30  32  30  37  f9  ad   |  ok | 
    2868480 |    2873184 | Rdr |30  03  99  9a                                                           |  ok | READBLOCK(3)
    2874292 |    2895092 | Tag |09  84  23  9a  1c  66  58  00  bc  00  15  59  55  31  30  33  e4  8b   |  ok | 
    3196800 |    3198112 | Rdr |80                                                                       |     | READBLOCK CONT
    3199156 |    3219956 | Tag |30  35  65  30  20  20  20  20  20  20  20  20  84  5c  0a  4a  3b  63   |  ok | 
    3521408 |    3526112 | Rdr |30  04  26  ee                                                           |  ok | READBLOCK(4)
    3527220 |    3548084 | Tag |00  02  00  40  32  30  30  38  30  31  33  30  4d  41  58  45  c9  6e   |  ok | 
    3846912 |    3848224 | Rdr |80                                                                       |     | READBLOCK CONT
    3849268 |    3870132 | Tag |4c  4c  20  20  4c  42  37  38  32  30  37  35  20  20  31  30  e5  ad   |  ok | 
    4170624 |    4175328 | Rdr |30  05  af  ff                                                           |  ok | READBLOCK(5)
    4176436 |    4197300 | Tag |30  32  2d  30  30  30  30  31  30  31  20  20  20  20  20  20  87  ad   |  ok | 
    4497536 |    4498848 | Rdr |80                                                                       |     | READBLOCK CONT
    4499892 |    4520756 | Tag |20  20  20  20  00  00  00  00  00  00  00  00  16  3e  f7  35  50  16   |  ok | 
    4820096 |    4824864 | Rdr |30  06  34  cd                                                           |  ok | READBLOCK(6)
    4825908 |    4846708 | Tag |0f  fe  01  00  0f  fe  01  00  0f  fe  01  00  0f  fe  01  00  87  c5   |  ok | 
    5149568 |    5150880 | Rdr |80                                                                       |     | READBLOCK CONT
    5151924 |    5172724 | Tag |0f  fe  01  00  0f  fe  01  00  0f  fe  01  00  0f  fe  01  00  87  c5   |  ok | 
    5472128 |    5476896 | Rdr |30  07  bd  dc                                                           |  ok | READBLOCK(7)
    5477940 |    5498740 | Tag |0f  fe  01  00  0f  fe  01  00  0f  fe  01  00  0f  fe  01  00  87  c5   |  ok | 
    5794688 |    5796000 | Rdr |80                                                                       |     | READBLOCK CONT
    5797044 |    5817908 | Tag |0f  fe  01  00  0f  fe  01  00  0f  ff  01  00  65  3d  88  c7  81  13   |  ok | 
    6118528 |    6123232 | Rdr |30  08  4a  24                                                           |  ok | READBLOCK(8)
    6124340 |    6145204 | Tag |00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  37  49   |  ok | 
    6443520 |    6444832 | Rdr |80                                                                       |     | READBLOCK CONT
    6445876 |    6466740 | Tag |00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  37  49   |  ok | 
    6768000 |    6772704 | Rdr |30  09  c3  35                                                           |  ok | READBLOCK(9)
    6773812 |    6794676 | Tag |00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  37  49   |  ok | 
    7093504 |    7094816 | Rdr |80                                                                       |     | READBLOCK CONT
[... 374 lines cut ...]

 [ PROXMARK3 ]
  firmware.................. PM3RDV4
  external flash............ present
  smartcard reader.......... present
  FPC USART for BT add-on... present

 [ ARM ]
  bootrom: RRG/Iceman/master/v4.9237-1797-g378f5fae 2020-10-19 19:50:30
       os: RRG/Iceman/master/v4.9237-1797-g378f5fae 2020-10-19 19:50:42
  compiled with GCC 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 [ FPGA ] 
  LF image built for 2s30vq100 on 2020-07-08 at 23: 8: 7
  HF image built for 2s30vq100 on 2020-07-08 at 23: 8:19
  HF FeliCa image built for 2s30vq100 on 2020-07-08 at 23: 8:30

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev A
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 295944 bytes (56%) Free: 228344 bytes (44%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

iceman1001, 2020-10-22:
Its working but I see your LTO tag is a different version that the one which we used to implement the commands with.
Yous says 0001,  mine is 0003 ,

I get the feeling your tag is smaller in size and the dump command just tries hard reading 255 blocks.
You know how much memory, blocks there is on your tag?

iceman1001, 2020-10-22:
fix  https://github.com/RfidResearchGroup/proxmark3/commit/74760c6de6194bf272810cf5f4c6ac19ba853f6f
Try pulling latest,  I adapted to the amount of blocks your tag seem to allow for reading.

So your cartridge is a LTO 1 ?  
Do you have a datasheet for it?

closed 2020-10-22 by iceman1001

will-caruana, 2020-10-22:
That worked! It is an LTO 1 and I don't have the data sheet. The software does say that my tag has 101 blocks and it reads to 100. Not sure if this is a big deal since I was still able to get a good dump. 
```
[usb] pm3 --> hf lto info

[+] UID......... B6BF370B35
[+] Type info... 0001
[+] Memory...... 101 blocks / 3232 bytes
[usb] pm3 --> hf lto dump
[+] Found LTO tag w 101 blocks / 3232 bytes memory
 🕒 ...reading block 100
[+] saved 8160 bytes to binary file hf-lto-B6BF370B35-dump.bin
[+] saved 255 blocks to text file hf-lto-B6BF370B35-dump.eml
```

Encase you wanted more data of the tag. 
```
./lto_analyzer.sh hf-lto-B6BF370B35-dump.eml
-- LTO CM Manufacturer's Information --
  LTO CM Serial Number: B6BF370B
  CM Serial Number Check Byte: 35
  CM Size: 04
  Type: 0001
  Manufacturer's Information: 010000000000350043493035313432333037313031310035
-- LTO CM Write-Inhibit --
  Last Write-Inhibited Block Number: 05
  Block 1 Protection Flag: 00
  Reserved: 0000
-- Cartridge Manufacturer's Information --
  Page Id: 0001
  Page Length: 0040
  Cartridge Manufacturer: HP
  Serial Number: 1160778053
  Cartridge Type: 0001
  Date of Manufacture: 20080207
  Tape Length: 0984
  Tape Thickness: 239A
  Empty Reel Inertia: 1C66
  Hub Radius: 5800
  Full Reel Pack Radius: BC00
  Maximum Media Speed: 1559
  License Code: 55313033
  Cartridge Manufacturer's Use: 303365302020202020202020
  CRC: 80EEC1A7
-- Media Manufacturer's Information --
  Page Id: 0002
  Page Length: 0040
  Servowriter Manufacturer: 20080201MAXELL LB784042 304-0000126
  Reserved: 0000000000000000
  CRC: BA55275A
```

iceman1001, 2020-10-22:
yes,  however it starts with reading block 0..   zero-based index

All of the output from the script would be nice to have in the `hf lto info command