RfidResearchGroup/proxmark3 #1033: ./pm3-flash-fullimage failing on 512k proxmark3

gwtp, 2020-10-25:
Successfully flashed bootrom but failed to flash image onto proxmark3 with 512k storage.

`$ ./pm3-flash-bootrom 
[=] Session log /home/foobar/.proxmark3/logs/log_20201025.txt
[+] About to use the following file:
[+]    /home/foobar/Downloads/proxmark3/proxmark3/client/../bootrom/obj/bootrom.elf
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0
 🕑  59 found
[+] Entering bootloader...
[+] (Press and release the button only to abort)
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0
 🕑  49 found
[=] Available memory on this board: 256K bytes

[=] Permitted flash range: 0x00100000-0x00140000
[+] Loading ELF file /home/foobar/Downloads/proxmark3/proxmark3/client/../bootrom/obj/bootrom.elf
[+] Loading usable ELF segments:
[+]    0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
[+]    1: V 0x00200000 P 0x00100200 (0x00000d50->0x00000d50) [R X] @0x298

[+] Flashing...
[+] Writing segments for file: /home/foobar/Downloads/proxmark3/proxmark3/client/../bootrom/obj/bootrom.elf
[+]  0x00100000..0x001001ff [0x200 / 1 blocks]
. OK
[+]  0x00100200..0x00100f4f [0xd50 / 7 blocks]
....... OK

[+] All done`

`$ ./pm3-flash-fullimage 
[=] Session log /home/foobar/.proxmark3/logs/log_20201025.txt
[+] About to use the following file:
[+]    /home/foobar/Downloads/proxmark3/proxmark3/client/../armsrc/obj/fullimage.elf
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0
 🕑  59 found
[+] Entering bootloader...
[+] (Press and release the button only to abort)
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0
 🕑  49 found
[=] Available memory on this board: 256K bytes

[=] Permitted flash range: 0x00102000-0x00140000
[+] Loading ELF file /home/foobar/Downloads/proxmark3/proxmark3/client/../armsrc/obj/fullimage.elf
[+] Loading usable ELF segments:
[+]    0: V 0x00102000 P 0x00102000 (0x0003e0f0->0x0003e0f0) [R X] @0x94
[!!] 🚨 Error: PHDR is not contained in Flash
[+] All done`

iceman1001, 2020-10-25:
If you read the output you see that your device reports having 256 kb,  I suspect you bought a easy clone and got tricked.

```
Available memory on this board: 256K bytes
```

gwtp, 2020-10-25:
Ack. Anyway to fit this onto the board by disabling some features?

iceman1001, 2020-10-25:
Yes, have a look inside:   https://github.com/RfidResearchGroup/proxmark3/blob/master/common_arm/Makefile.hal#L81-L96

You just comment out the ones you can live w/o and recompile and try flashing

closed 2020-10-25 by iceman1001