RfidResearchGroup/proxmark3 #1035: Like a bug for "hf mfu restore"

xianglin1998, 2020-10-27:
Hi guys, today i use "hf mfu dump" to read a `ul` tag, and place a `ul-gen2` tag to write by "hf mfu restore s f xxxx.bin".
Redump after restored, i have some problem found, i always lost last 2 block no write, i try to use `ul-c` and other mfu tag,  but it's a pity that this problem is occur on all mfu tag with i found...

As usual, I read the source code and found something I didn't understand.
Pic:
[image]

In the `write_special`,  the `blocks` no defind last page `from x - end`, always `from x - end - 2`.

xianglin1998, 2020-10-27:
@iceman1001 
@doegox

iceman1001, 2020-10-27:
I would say its because the two last blocks is password and pack. We don't wanna set those of mistake

xianglin1998, 2020-10-27:
OK, that is pack and pwd for ul-ev1, thanks.

closed 2020-10-27 by xianglin1998