RfidResearchGroup/proxmark3 #1036: [BUG] The restore bug for "hf 15 restore"

xianglin1998, 2020-10-29:
If we use cmd "hf 15 restore f xxx" to restore a tag, we can get a wrong data write result,

[image]

Because data offset not increment of block size, only data address + 1.

[image]

xianglin1998, 2020-10-29:
@iceman1001 This may be a bug, because it broke my imagination of block.

xianglin1998, 2020-10-29:
If it is confirmed to be a bug, then I will new a PR to fix it.

iceman1001, 2020-10-29:
Are you talking about `hf 15 restore`   or are you talking about `hf 15 write` 
restore calls write underneath.   Messy, messy.

restore:
it seems to be dependent of block size.    Which should default to 4,   or user-specified to N.

write:
.....

I am missing output in your pictures,  I don't see your restore command among others.
Can you run it again with the debug on turned on?
```
data setd 2
hf 15 restore
```

iceman1001, 2020-10-29:
Good catch!  I confirmed it.  Fix pushed.

closed 2020-10-29 by iceman1001

xianglin1998, 2020-10-29:
HaHaHa, Thanks PR, I will continue to find bugs, which is one of my hobbies.

iceman1001, 2020-10-29:
Find bugs is good.   Refactor bad stuff is even better !

commit 0e4f5f48 (2020-10-29): fix #1036 ,  wrong offset used. Thanks @xianglin1998