RfidResearchGroup/proxmark3 #1050: Proxmark3 won't appear when trying to flash.

Syax242, 2020-11-07:
I've been trying to get my proxmark3 easy working with my computer on windows 10 and every other step leading up to flash seems to have worked just fine. Device manager is detecting that is plugged in and working on COM4 but when I try to flash it the shell gets stuck on waiting for the proxmark3 to appear. I haven't been able to find a fix for this problem thus far so any help would be greatly appreciated.

iceman1001, 2020-11-07:
Which commands are you using?  Did you follow the install-windows guide ?

Syax242, 2020-11-08:
> 
> 
> Which commands are you using? Did you follow the install-windows guide ?

I followed the install guide video up to the flash. The flash command shown in the video didn't work so I've been trying ./pm3-flash-all which at least gets to the part where it's waiting for the proxmark to appear. I've noticed when I install the proper driver from the file that it also says in parentheses that it is an old driver though, not sure if that'd have an impact on anything. I think my proxmark3 easy is also the chinese version so that could also be prudent.

iceman1001, 2020-11-08:
yeah, those videos are over a year old,  maybe not 100% but should be close.  The doc where you found the video link is however up-to-day. When you started this issue, you got an template to fill in some extra info,   which is quite handy...   So that we don't have to ask you the same questions.  So I can only recommend that you re-read the relevant docs found here in the repo and make sure you follow the instructions.

closed 2020-11-08 by iceman1001