RfidResearchGroup/proxmark3 #1052: Positive response on invalid mfdes commands

ikarus23, 2020-11-08:
Just to keep track of the little things: When I do a wrong `mfdes` command, there is the positive response "Successfully created aid.", which is not true.

```
[usb] pm3 --> hf mfdes createaid -a 000001 -k 0f -l ff
[!] ⚠️     Can't create aid -> Value of the parameter(s) invalid
[+] Successfully created aid.
[usb] pm3 --> hf mfdes createaid -a 123456 -f 1111 -k 0E -l 2E -n Test
[!] ⚠️     Can't create aid -> Out of Eeprom, insufficient NV-Memory to complete command
[+] Successfully created aid.
```

iceman1001, 2020-11-08:
good catch,
I pushed a fix

ikarus23, 2020-11-08:
From report to fix in under 7 minutes. Damn!

closed 2020-11-08 by ikarus23

iceman1001, 2020-11-08:
...simple fixes...  why wait?  ;)