RfidResearchGroup/proxmark3 #1053: DESFire: can't delete application (AID)

ikarus23, 2020-11-08:
I have a (demo) tag that only has one application (with no files) and the AID 0x000001. Everything uses default keys. However, for creating/deleting app the is no key required.

The bugs:
* I can't delete the application (even after authentication, which should not be required).
* The pm3 shows `CMK required for create/delete : YES` but this is not true (checked with NXP's TagInfo app).

Maybe the bugs are connected.

```
[usb] pm3 --> hf mfdes info

[=] --- Tag Information ---------------------------
[=] -------------------------------------------------------------
[+]               UID: 04 64 0E E2 EA 1D 80 
[+]      Batch number: BA 95 D9 4D 10 
[+]   Production date: week 40 / 2009

[=] --- Hardware Information
[=]      Vendor Id: NXP Semiconductors Germany
[=]           Type: 0x01
[=]        Subtype: 0x01
[=]        Version: 1.0 (DESFire EV1)
[=]   Storage size: 0x16 (2048 bytes)
[=]       Protocol: 0x05 (ISO 14443-2, 14443-3)

[=] --- Software Information
[=]      Vendor Id: NXP Semiconductors Germany
[=]           Type: 0x01
[=]        Subtype: 0x01
[=]        Version: 1.3
[=]   Storage size: 0x16 (2048 bytes)
[=]       Protocol: 0x05 (ISO 14443-3, 14443-4)

[=] --- Card capabilities
[=] 	1.3 - DESFire Ev1 MF3ICD21/41/81, Support extended APDU commands, EAL4+
[+]    Number of Masterkeys                  : 1
[+]    [0x08] Configuration changeable       : YES
[+]    [0x04] CMK required for create/delete : YES
[+]    [0x02] Directory list access with CMK : YES
[+]    [0x01] CMK is changeable              : YES
[+]    Operation of PICC master key          : (3)DES
[+]    PICC Master key Version               : 0 (0x00)
[=]    ----------------------------------------------------------
[+]    [0x0A] Authenticate      : YES
[+]    [0x1A] Authenticate ISO  : YES
[+]    [0xAA] Authenticate AES  : NO
[=] -------------------------------------------------------------

[=] --- Free memory
[+]    Available free memory on card         : 2208 bytes
[=] -------------------------------------------------------------

[usb] pm3 --> hf mfdes enum

[=] -- MIFARE DESFire Enumerate applications --------------------
[=] -------------------------------------------------------------
[+]  Tag report 1 application 

[+] --- AMK - Application Master Key settings
[+]   AID : 000001
[=]   DF AID Function 000001     : (unknown)
[+]   AID Key settings           : 0x0f
[+]   Max key number and type    : 1, (3)DES
[=] -------------------------------------------------------------
[+]   Changekey Access rights
[+]   -- AMK authentication is necessary to change any key (default)
[+]    [0x08] Configuration changeable       : YES
[+]    [0x04] AMK required for create/delete : NO
[+]    [0x02] Directory list access with AMK : NO
[+]    [0x01] AMK is changeable              : YES
[=] -------------------------------------------------------------
[=]   Application keys
[+]    Key [0]  Version : 0 (0x00)
[+]  Tag report 0 files
[=] -------------------------------------------------------------

[usb] pm3 --> hf mfdes deleteaid -a 000001
[!] ⚠️     Can't delete aid -> Current authentication status does not allow the requested command

[usb] pm3 --> hf mfdes auth -m 2 -t 2 -a 000000 -n 0
[+]   Key        : 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[+]   SESSION    : DF 35 BD 37 67 79 B6 6B DF 35 BD 37 9A 2F 72 66 
[=] -------------------------------------------------------------
[=] -------------------------------------------------------------

[usb] pm3 --> hf mfdes deleteaid -a 000001
[!] ⚠️     Can't delete aid -> Current authentication status does not allow the requested command
```

tcprst, 2020-11-08:
As a workaround, I've used `hf mfdes formatpicc` on @mwalker33's suggestion

iceman1001, 2020-11-08:
just for the sake of it,   try  aid  010000  instead..  maybe the byte order is wrong in the client parsing

ikarus23, 2020-11-08:
I already checked that, It did not work ;)

ikarus23, 2020-11-08:
Ok, I've learned something important! Even if the PICC master key is not required for app deletion, there must be an authentication. So deleting is always:
1. Authenticate with the PICC master key or the application master key.
2. Delete application.

Regardless of this, it still does not work on the PM3. The displayed `CMK required for create/delete` is wrong and even if you authenticate before deletion it will not work.

iceman1001, 2020-11-19:
... so now its a bit different.  You will need to do 

1. Auth (field is on)  (and select a application/AID)
2. delete/create (afterwards field is off)

The new style will most likely mean that the other commands need to be adapted.  In order to fit a auth first.

Pull it and test create/delete.
If you wanna see if the rf field is up,  the add  "data setd 1"  client debug flag on.

ikarus23, 2020-11-20:
Great stuff! Not it works.
```
[usb] pm3 --> hf mfdes auth -m 1 -t 1 -a 000001 -n 0
[+]   Key        : 00 00 00 00 00 00 00 00 
[+]   SESSION    : BA 51 6E 7D BD 34 C1 3F 
[=] -------------------------------------------------------------
[usb] pm3 --> hf mfdes deleteaid -a 000001
[+] Successfully deleted aid.
[usb] pm3 --> hf mfdes enum

[=] -- MIFARE DESFire Enumerate applications --------------------
[=] -------------------------------------------------------------
[+]  Tag report 0 applications
[=] -------------------------------------------------------------
```

closed 2020-11-20 by ikarus23